pcb2gcode/pcb2gcode #469: Outline: board perimeter is cut before internal cutouts

codehero, 2020-07-13:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):

v 2.1.0

What did you try (**include command-line arguments**):

Generating outline from edge cuts file edge.txt

pcb2gcode \
	--metricoutput=1 \
	--metric=1 \
 --drill=holes.drl \
 --min-milldrill-hole-diameter=0 \
 --milldrill \
 --milldrill-diameter=0.79 \
 --zmilldrill=-1.7 \
 --zdrill=-1.7 \
 --drill-feed=300 \
 --drill-speed=24000 \
 --nog91-1 \
 --nog81 \
 --nog64 \
 --zcut=-1.7 \
 --zsafe=10 \
 --outline=holes.txt \
 --cutter-diameter=1.27 \
 --cut-feed=300 \
 --cut-speed=24000 \
 --cut-infeed=0.25 \
 --cut-vertfeed=150 \
 --zchange=35

What happened:

Perimeter of PCB is cut before internal cutouts

What did you expect to happen:

Internal cutouts should be cut BEFORE perimeter of PCB.

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

[edge.txt](https://github.com/pcb2gcode/pcb2gcode/files/4911942/edge.txt)
[holes.txt](https://github.com/pcb2gcode/pcb2gcode/files/4911946/holes.txt)

eyal0, 2020-07-13:
Why is the order in which cutouts are done important?  Right now they are done in the order that minimizes the overall time to make the board.

codehero, 2020-07-13:
Obviously you can't cut any inside parts if the outside part is completed first.
When you remove perimeter of the board and leave only small bridges, the board itself is allowed to deform without any of the hold down stresses.
Thus, would prefer to cut the inside of the board first, then the perimeter.

eyal0, 2020-07-13:
I understand.  So you're saying that even with bridges, you would still prefer increased sturdiness so it would be better to cut the inside before the outside.  Is that right?

I could sort the cuts so that the one with the least length happened first.  For example, if your board is a rectangle and the cut-out is a circle inside the rectangle, the circle has a smaller perimeter than the rectangle so the circle would be completed before the rectangle.

Would that fix your problem?

codehero, 2020-07-13:
Yes precisely. Some copper clad on ebay is even as thin as 0.005". FR1 is much more flexible than FR4.

Your heuristic would work for the current board I am using, but in theory smaller area does NOT necessitate smaller perimeter.

I would think you still sort by cutting time optimization, but just use the line sweep concept to find the outermost perimeter and tag it for cutting first.

eyal0, 2020-07-14:
I was just trying to think of the easiest thing that would work.  I'd rather make a simple solution that works and later upgrade to a more complex one when someone finds the need, rather than try to anticipate a future need that may never arise, you know?

So for now I'll just sort them edge cut length, which is a pretty good proxy for doing the inside before the outside.  Shouldn't be too hard to do.

codehero, 2020-07-14:
I don't understand your logic, but you can solve it how you want. Line sweep is literally just sorting by the least x coordinates of each path, which has no cases that break. Edge cut has cases that break. That is my final suggestion and thank you for looking into the issue.

PR #471 (2020-07-15): Sort outline cuts from shortest to longest instead of using TSP
This fixes #469
closed 2020-07-15 by eyal0