pcb2gcode/pcb2gcode #470: allow the caller to assign tool identifiers

c-, 2020-07-14:
I just picked up a used mill with an ATC and I've been reviewing my workflow to see what I'd need to do to make it work properly (no rush, I'm going to have to iterate through a couple of controllers before I get to that point).

The main problem I'm seeing is that pcb2gcode generates tool numbers in the gcode according to the order it sees them. i.e.:

    mill-diameters=0.5mm,0.3mm

would generate g-code with T1=0.5mm, T2=0.3mm, and flipping the order would reverse the mapping. This... semi random assignment might be a bit of a management problem for an ATC, particularly because the drill tool numbers in the next operation would also be indexed from 1.

I think the least complicated solution would be to allow the user to map a tool identifier to a given diameter using a syntax something like:

    mill-diameters=T2:0.5mm,T4:0.3mm

This would generate:

      T2
      M6
      ...
      T4
      M6

to handle the tool changes, and the job of maintaining the tool mappings remains entirely the problem of the user.

Of course, that also raises the issue of varying speed and feeds according to tools, but at least being able to globally uniquely identify the tool would make something like a post-processor easier to implement.

eyal0, 2020-07-15:
First up, congrats on the new equipment!

I don't want to require a post-processor.  Let's see if we can do this right!

So, one way is like you said, to use colons.  `T2:0.5mm,T4:0.3mm`.  But, like you said, the speeds and feeds are different for those.  I guess that we could use even more colons!  `T2:0.5mm:123mm/s:456mm/s`.  You can see that it's quickly getting pretty ugly!  Nicer might be, like, `T2:diameter=0.5mm:feed=123mm/s:speed=456mm/s`  Having them tagged with the purpose makes it easier to see what's going on, though the text is getting annoyingly long!

Another idea would be to separate the tool definitions from the usage.  Like this:
```
tools=T2:0.5mm,T4:0.3mm,T6:0.1mm
mills=T2,T4
drills=T6
```

That seems pretty flexible.  A nice part about that is if you want to use different tools, you don't have to change as much stuff.  In fact, you could put those tools into the `millproject` file and then just put the `--mills=T2,T4` on the command-line, for example.

I don't think that any of it would be too hard to do, it's just a matter of making it backward compatible so that we don't break stuff that already works and also think about what might be needed in the future.

c-, 2020-07-15:
> First up, congrats on the new equipment!

Oddly enough, I was really looking for a small metal lathe, but when I saw the ridiculous price I wasn't passing it up. [The hardware is a gorgeous example of German overengineering](https://www.bungard.de/de/produkte/fraesbohrplotter/ccd-atc), but the software is just... nope.

Having the tool definitions separate from the usage would work well.

> tools=T2:0.5mm,T4:0.3mm,T6:0.1mm

Can pcb2gcode support repeated instances of an option? i.e.

    tool=T2:diameter=0.5mm:feed=123mm/s:speed=456mm/s
    tool=T4:diameter=0.3mm:feed=321mm/s:speed=654mm/s

... might be easier for a user to manage. I guess:

    tool2=diameter=0.5mm:feed=123mm/s:speed=456mm/s
    tool4=diameter=0.3mm:feed=321mm/s:speed=654mm/s

would also work, but it's not very Unixy.

eyal0, 2020-07-15:
Yes, I thought about the latter, too, but it's not very clean using the boost program options library, which is what pcb2gcode is using.

pcb2gcode can already handle some options because specified twice.  For example, mill-diameters can be done on multiple lines or separated by commas.  Either way it all becomes one big list.

First comes the name, mandatory because you need to use it to specify in the `mill-diameters` or `available-drills` option which tool you want by name.  And then a bunch of colon separated keys and values, like diameter, feed, speed, etc.  For drills I also have a tolerance setting that I need to deal with.

For both drills and mills, I'll be able to keep supporting the old way and the new way because the new way will always start with a "T" and the old way always starts with a digit.  And I'll try to spit out an error message when the user screws it up.

Maybe a warning if a tool goes unused?  Not sure about that one, I think that the warning could get annoying and would distract from actual warnings, like traces that can't be drilled.

Pretty machine you got there.  It'll be sweet if you're able to just press go and get all the different drill bits working right.  I assume that the machine does tool offset measurements for you.

c-, 2020-07-16:
> For drills I also have a tolerance setting that I need to deal with.

There may be some value in other flags, like "this tool is okay for mill-drill". Or if something like the v-bit angle could be specified and pcb2gcode does the depth-extra-width calculation automagically. I'm sure there's plenty of spin-off tickets sitting under this idea. On the other hand, there's also a lot of opportunity for feature creep. I'd be perfectly happy with just the basics.

> Maybe a warning if a tool goes unused?

I don't think so, unless maybe it's triggered by a --pedantic option. In practice I would see a user having a "standard" tool list that they pick and choose a subset of tools from for each project. I'd find this useful *now* even for manual tool changes, just in terms of having a common tool inventory and being able to memory the tool numbers. It would be more usable if there was some sort of includefile mechanism to pull that list in, but there's that feature creep again. Obviously, it'd be an error if a tool was referenced which doesn't exist, or is inappropriate for the task (i.e. if v-bit angles were configurable, then including a v-bit in the drills list is an obvious error).

> It'll be sweet if you're able to just press go and get all the different drill bits working right.

That's the goal. But even if I can only get it doing manual tool changes in the short term, the fast rapids, tool-free pneumatic chuck, and TLO is the next best thing.

> I assume that the machine does tool offset measurements for you.

The probe is at the start of the tool holders. Although it's documented as a broken/out-of-spec drill detector (so they probe the tool after picking it up and before putting it away) and they prefer to depend on the depth ring to set a consistent 21mm tool length. I think it's a necessary approach if you're using the floating head (yup, it also includes the floating head) for isolation milling since you can't adjust it easily if the tool length changes between bits.

friedrich-goetze, 2021-02-07:
I had the same issue with the toolnumbers a lot and used to fix it with my own rather crude postprocessor script.
I like the idea with the tool table, which is easy to copy between projects. It would be nice to specify feeds and speeds for different operations like
`tool=T2:diameter=0.5mm:mill-feed=123mm/s:mill-speed=4560:drill-feed=234mm/s:drill-speed=567:cut-feed=111mm/s:cut-speed=333`

It might be a bit cumbersome, but you have to specify all these values anyway. I prefer to have them all at one place. Also, you can make sure,  a that sth like a drill is not selected for cuts, because you'll see that the cut-values are missing.

eyal0, 2021-02-21:
Now that #512 is done I'll look at this.

I'm thinking about the design.  We would like to separate the definition of tools from their usage.  The notation above looks good to me though I would like for the tool name to also be a field.  So it would be like:

```
tool=id=T2:diameter=0.5mm...
```

Notice how it says "id=" in there instead of just assuming that the first field is an ID.  I'll probably have both an "id" field and a "name" field.  If you only set one, the other will be the same.  And if you want, you can set both.  It's the name that will be in the usage.  Say you have one machine for milling and one for drilling.  And T2 is the bit that you want to use for both of them.  You need a way to associate the tool with the task but you can't have two "T2"s.

All tools will be specified the same way.  After pcb2gcode reads the whole spec on a tool, then it can figure out if it's a drill or a milling bit or whatever.  And then it'll notice if something is wrong, like you specified the overlap for a drill bit.  For any tool that has a missing specifier, we'll take it from the usual settings which will behave like defaults.

hustav, 2023-07-23:
I will second on this feature. 

Alternatively, could we list the tools' diameters and their use (drill, mill, cut) and use the milling/drilling/cutting settings as they are?