pcb2gcode/pcb2gcode #472: Bridges in the worse possible location

c-, 2020-08-08:
I'm guessing this is a side-effect of #409, but I don't think that's where the bridges on a simple rectangle (I didn't even bother rounding the corners) belong:

[image]

[Edge.Cuts-back-1.2mm.nc.txt](https://github.com/pcb2gcode/pcb2gcode/files/5046131/Edge.Cuts-back-1.2mm.nc.txt)
[millproject.txt](https://github.com/pcb2gcode/pcb2gcode/files/5046132/millproject.txt)
[Edge.Cuts.gbr.txt](https://github.com/pcb2gcode/pcb2gcode/files/5046133/Edge.Cuts.gbr.txt)

eyal0, 2020-08-10:
Yes, it's from #409 .  Now that it's possible to make bridges that span multiple segments, and because the bridges try to place themselves where they are furthest away from one another, the corners were selected here.  The corners are actual round because the bit has to curve around the corner of the PCB.

Is it really bad to have bridges at the corners?  Why?

Can you think of a better heuristic for deciding where the bridges should go?  I'd like to keep it automated but if it can be made smarter, that's great, too!

c-, 2020-08-10:
> Is it really bad to have bridges at the corners? Why?

It's awkward to get a blade in to cut a bridge at a corner versus a side (you have to cut with the tip rather than a meatier part of the blade) and snapping off the bridge is more likely to cause failure at a corner than in the center of a long edge. I also find it's a bit cleaner to smooth a bridge when it's in the middle of a long flat section.

> Can you think of a better heuristic for deciding where the bridges should go?

First bridge at the middle of the longest segment? The rest of the algorithm should be fine. There would obviously be cases where it might still put a bridge in the "wrong" place (i.e. four bridges on a triangle), but at least the user can change the number of bridges to land them somewhere else.

eyal0, 2020-08-12:
Hmm...  The idea of the "first" bridge is weird because I don't place them in any order.  I just place them all in random locations and then move them around trying to find the best spacing.

Maybe what I could do is only place them on segments that are long enough like before?  And then, if I find that the number of bridges exceeds the number of segments that are long enough, only then will I include all the short segments, too.

This behavior would fix the original bug while maintaining the original behavior for most people.  It might not act as well as I'd like on oddly shaped PCBs but for rectangular and round, which are surely the most common, I think that it would work fine.

c-, 2020-08-12:
> Maybe what I could do is only place them on segments that are long enough like before?

I think it would be better (and fix my problem), but there are obvious pathological cases... a rectangular board with chamfered corners, or a very long, narrow rectangular board with two bridges.

Hm. It's tough. I assumed there was some sort of strategy/weighing in selecting bridge segments, at least initially. If it's purely random that's harder.

closed 2020-10-19 by eyal0

c-, 2020-10-20:
That seems to do the trick, and still handles the messy cases like we find in #337.

commit 016a291f (2020-10-19): Only use small parts of the edge as bridges if necessary

This fixes #472