pcb2gcode/pcb2gcode #473: Feature request: add option to specify spindle speed with M3/M4

Patronics, 2020-08-27:
Could you add an option to specify the spindle's speed using the form of (for an example speed of 15000 RPM) `M3 S15000` instead of `G00 S15000`?

This isn't really an issue with pcb2gcode, but with some other software (specifically [CNC-GCode-Controller](https://github.com/im-pro-at/cncgcodecontroller)), but unfortunately CNC-GCode-Controller seems to be unmaintained at this point, and it should be relatively easy to workaround with pcb2gcode.

eyal0, 2020-10-18:
Are you sure that CNC-GCode-Controller works like that, with `M3 S15000`?  I looked through their code and I didn't see that...

Where do you see that it should be that way?

Patronics, 2021-01-27:
Sorry for the very late reply, I haven't used my CNC machine again until recently, and thought I had responded previously. The issue with CNC-GCode-Controller is [described in more detail here](https://github.com/im-pro-at/cncgcodecontroller/issues/41), basically it is concluding that because there is no movement involved other than the speed change (which it apparently doesn't understand) it removes the line completely. The marlin firmware I'm using allows M3 with the speed as an alternate way to start the spindle at a specific speed, ([documented here](https://marlinfw.org/docs/gcode/M003.html)), which is not automatically removed by CNC-GCode-Controller, as the G0 equivalent is.