pcb2gcode/pcb2gcode #474: spinup/down time needs to be dumber

c-, 2020-08-29:
--spinup-time and --spindown-time are outputting "G04 Pxxx" as a value in ms, with a decimal point. So an argument like --spinup-time=20s comes out in g-code as "G04 P20000.00000"

This doesn't work with all controllers. And it *can't* work with all controllers, because apparently dwell time units are a standards fail.

[LinuxCNC expects dwell units to be seconds](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g4).

GRBL uses seconds as well (see the mc_dwell() function in motion_control.c).

[TinyG also expects seconds](https://github.com/synthetos/TinyG/wiki/Gcode-Support).

[Marlin uses P being ms, and S being seconds](https://marlinfw.org/docs/gcode/G004.html).

ms can be used with some other controllers, but [you need to use X or U as the param code, not P](https://www.intuwiz.com/articles/g04.html).

But I also get the impression that some machines will interpret a P value with a decimal as seconds, and a P value without as decimal as milliseconds.

Hence the title of the issue. Because of the amount of variability, it may be safest if pcb2gcode just took the raw g-code parameter as the argument and passed it directly through to output without trying to interpret them as a time value. i.e

    --spinup-time="P0.5" --spindown-time="X500"

c-, 2020-08-29:
For backwards compatibility purposes, interpreting a numeric/time value as ms and something prefixed with a character as a pass-through would also work okay.

eyal0, 2020-08-29:
I like your idea and also the backward compatiblity. Sounds fine.

This is a similar problem to #473 but in that one, it's not just the parameter that isn't good, it's the whole line. So maybe it make sense, for the sake of consistency, to have the alternative to a number be the whole line, including the G04.

`spinup-time="G04 X500"`

c-, 2020-08-29:
Yes, that would also suit controllers which might have a custom "wait for spindle to reach full speed" g-code (macro or built-in).

c-, 2020-09-05:
Just a note on a workaround... if you use a unitless --spinup-time=20, you get P20.00000 in the output. So as long as P is the correct parameter for your particular controller it passes through. Maybe the fix is to not read the documentation or code and then overthink the problem... ;)

eyal0, 2020-09-10:
It is?  I didn't remember that.  I think that when you put a unitless number into pcb2gcode, it defaults to milliseconds, so that's probably what's happening.

How frustrating that there is no agreement on g-code parameters!  I don't know what a good solution for this would be, except to do it like you have it.  The only other way that I can think of would be to add yet another flag that indicates if your printer is `G04 P` in seconds or milliseconds, which would be a clumsy solution.

Just typing it in unitless seems like an okay solution!

c-, 2020-09-10:
> Just typing it in unitless seems like an okay solution!

For the controllers that take P<value> as a param, yeah, it solves that immediate problem.

A controller that uses U or X is still out of luck. I don't know how likely someone is to point such a machine at pcb2gcode output.

There might be some value in generalizing the entire concept into a pre/post gcode parameter for spindle start and stop, similar to --pre-milling-gcode/--post-milling-gcode... ? That's also where things like coolant control might be a better fit than always automatically including it in the output.