pcb2gcode/pcb2gcode #475: spinup/spindown dwells are being used where they aren't needed, multiple times

c-, 2020-09-03:
pcb2gcode doesn't keep track of whether the spindle is on/off, which means when spinup/down is used there are pauses in places like at the beginning of a job, prior to the first tool change, when the spindle was never started.

One of this things that makes this extra noticeable when using a long dwell is that M5's are being generated repeatedly. This is what I get in my milling output between at a tool change:

    G00 Z10.000000 ( retract )

    M5 ( Spindle off. )
    G04 P20.000000

    G00 Z10.00000 (Retract to tool change height)
    T1
    M5      (Spindle stop.)
    G04 P20.00000 (Wait for spindle to stop)
    (MSG, Change tool bit to mill diameter 0.38000mm)

Drill tool changes are fine:

    G00 Z10.00000 (Retract)
    T5
    M5      (Spindle stop.)
    G04 P20.00000
    (MSG, Change tool bit to drill size 1 mm)
    M6      (Tool change.)
    M0      (Temporary machine stop.)
    M3      (Spindle on clockwise.)
    G0 Z0.50000
    G04 P20.00000

eyal0, 2020-10-18:
Okay, I have a fix but I don't know if I got them all!  Looks like after each tool we do M5 and then delay.  And before each tool, we also do it!  So we're getting it twice, at the end of one tool and then once again because a new tool is about to begin!

I've got a fix that will make it so that we do it before each tool but not at the end of each tool, only at the end of the final tool.

As far as I can tell, it's only a problem for the M5, not for the M3 to start the spindle.  Is that right?  If so then I think that my fix is sufficient.  It's fix #485 .  Let me know if that looks right to you.  Thanks!

c-, 2020-10-18:
That looks pretty good, thanks.

It was definitely the end-of-job/end-of-file "Spindle Off" that was being the biggest annoyance.

It would be nice to also bypass the dwell after the *first* M5 in the file, since the spindle shouldn't be spinning, but it's just a single extra dwell at a non-critical time, so... probably not worth an extra *if*.

PR #485 (2020-10-19): Only output gcode end on for the last tool
This fixes #475
commit 4a8db526 (2020-10-19): Merge pull request #485 from eyal0/gcode_end_only_once

Only output gcode end after the last tool

This fixes #475
commit 803d67c3 (2020-10-19): Only output gcode end after the last tool

This fixes #475
closed 2020-10-19 by eyal0