pcb2gcode/pcb2gcode #477: Bug: --mill-infeed causes run across copper areas

hannesweisbach, 2020-09-25:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):

```
pcb2gcode --version
2.1.0
Git commit:
Boost: 106600
Gerbv: 2.7.0
Geos: Not installed
```

What did you try (**include command-line arguments**):

I ran `pcb2gcode`, without command line parameters. All my configuration is in the attached `millproject` file.

What happenned:

pcb2gcode created a gcode file which contains a move across the board (including copper areas) at full milling depth. In the case of this minimal example milling depth should have been -0.06mm, zsafe=1mm and mill-infeed=0.03 (2 passes). From the attached `front.ngc`:

```
G01 X45.28711 Y-13.33500
( Mill infeed pass 2/2 )
G01 Z-0.06000
G04 P0 ( dwell for no time -- G64 should not smooth over this point )
G01 F80.00000
G01 X36.83000 Y-20.32000
```
I.e. first lower the mill bit and then move to the beginning of the mill path.
This appears to be a general issue when `mill-infeed` is used and milling paths are not a closed loop.

What did you expect to happen:

Travel moves at zsafe, i.e. raise to zsafe, travel to X36.83000 Y-20.32000, lower to z = -0.06, continue milling.
Or more general: If the end of the milling path is not the starting point for a new pass, raise the bit to zsafe, move to the begin, lower to the new depth, continue following the path.

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

Github does not let me upload the files directly, so attached them zipped 
[test.zip](https://github.com/pcb2gcode/pcb2gcode/files/5282160/test.zip) and in this gist for your online viewing pleasure: https://gist.github.com/hannesweisbach/ae4dfeb4be195786e8fa5b6f9eee28ac

eyal0, 2020-10-06:
I'll take a look.  It's been a while since I worked on this project and the CI is in disrepair.  It needs to work first before any new commits.

hannesweisbach, 2020-10-07:
Sure, no worries.

eyal0, 2020-10-11:
Now that I've taken a look, I can explain what's happening here.  pcb2gcode compares the time to retract, move, and lower versus just moving across.  If it's faster to just mill across, that's what it does.  This will usually save time.  It shouldn't ever cut across an existing trace, however.

So if you're seeing it cut across a trace then that needs to be fixed.  I'm trying to figure out if that's the case.

If you're just don't like the look of mill running across the board and you want it to retract then you can adjust some options to tell pcb2gcode that g0 is actually much faster, in which case it should choose it more:

```
  --g0-vertical-speed arg (=0.0211667 m s^-1)
                                        speed of vertical G0 movements, for use
                                        in path-finding
  --g0-horizontal-speed arg (=0.0423333 m s^-1)
                                        speed of horizontal G0 movements, for 
                                        use in path-finding
```

That line *does* seem to cut across the trace, though.  I can't tell exactly.  If that's the case then we need to use the covered_by and such like you noticed.

hannesweisbach, 2020-10-11:
> Now that I've taken a look, I can explain what's happening here. pcb2gcode compares the time to retract, move, and lower versus just moving across. If it's faster to just mill across, that's what it does. This will usually save time. It shouldn't ever cut across an existing trace, however.

I know that feature and it *does* mill across copper areas. See the attached example files. I can also attach (larger) real-world project files & photos where this happened.

I don't really care how non-copper areas look ;)

eyal0, 2020-10-11:
Ah, I see what's going on now.  I didn't understand before.  It's for the case when you want to mill your trace in multiple passes and the start and end aren't the same.  Now I get it!

I have pushed a new change.  I also added your test.zip file into the examples directory so that it will be part of future integration tests, to prevent this feature from breaking in the future.

I modified your code to run only when the start and end are different and only when it's not the first pass.  Let me know how it looks.

hannesweisbach, 2020-10-11:
Yes, I mill multiple passes (230V circuit)! Sorry, I didn't know that was important  :(

> I modified your code to run only when the start and end are different and only when it's not the first pass. Let me know how it looks.

Umm … I already programmed it that way … ?

eyal0, 2020-10-11:
> Umm … I already programmed it that way … ?

Your code seemed to retract before ach pass, including the first one.  And it ran even when there was no need to lift because the code was a loop.  Also there was a bug where X and Y were confused.  Maybe you made the change locally but didn't upload it?

Here's what was uploaded: https://github.com/hannesweisbach/pcb2gcode/commit/295c9177ffc00229bc4ab75375457470b6f3630a

hannesweisbach, 2020-10-12:
Ha, you're right, I never pushed newer changes 🤦🏼‍♂️

eyal0, 2020-10-12:
No problem.  Is yours similar to mine?  Are you satisfied with my changes?

hannesweisbach, 2020-10-12:
They are the same except for two  differences:

(1) Your's takes care of the cutter path. I didn't know the same issue exists there.
(2) I used `boost::geometry::equals` (instead of `!=`) to compare `front()` and `back()`.

I was hoping  `equals()` wouldnt compare the doubles for point_d2 with `!=`, but instead with an epsilon. I'm  not sure, if that is an issue that can happen in the case of pcb2gcode, but I learned  that comparing doubles with `==` or `!=` is rarely the thing you want.

Probably one could also take into account `--tolerance` and/or `--optimize` for the comparison (and don't lift if beginning and end are close enough together), but maybe that is handled elsewhere in the code properly and the ends of the milling path are at this point in the code the same or they aren't. I'm not that familiar with the pcb2gcode code base.

But your changes look good to me so far.

PS: I pushed one of my changes: 89aa713, but later overwrote it with an old change. Looks like I f***** up rebase and/or `git push --force` … doesn't matter now.

eyal0, 2020-10-12:
bg::equals is the same as !=.  In `bg_operators.hpp` I put in a bunch of operator overloads to make it easier to use boost geometry.

I don't know if the cutter will show the actual problem because this only happens when the path isn't a loop and usually cutter paths are a loop because of the `--fill-outline` option, which is default true because that's how most CAD software works, I think.  So it's having no effect.  I might add an integration test to intentionally trigger this so that the coverage is improved.

Boost geometry has a concept of a linestring, which is a series of points, and a ring, which is a series of points where the first and last must be the same.  Boost makes sure that the rings maintain that closed loop so there's no fear that they go from identical to just nearly the same.  The only algorithm that might change points to make them almost but not exactly the same could be the optimize function, which can trim a linestring to make it approximate the original but with fewer points.  And for those, the first and last point are never modified so it should still be okay.

So it should be fine to compare for equality.  And anyway, worst case, it'll just take a little longer with a retract and plunge, no big deal.

hannesweisbach, 2020-10-12:
> I don't know if the cutter will show the actual problem because this only happens when the path isn't a loop and usually cutter paths are a loop because of the `--fill-outline` option, which is [...]

I thought about it on my way to work and the only thing where is could be needed is if you mill a slot that is wider that your cutter (and probably non-uniform in width).

> Boost geometry has a concept of a linestring, which is a series of points, and a ring, which is a series of points where the first and last must be the same. Boost makes sure that the rings maintain [...]

Ah, ok. Good to know :)

commit 48ec8f49 (2020-10-17): Merge pull request #478 from hannesweisbach/patch-2

Lift mill between steps to zsafe if path is not a closed loop

This fixes #477
closed 2020-10-17 by eyal0