pcb2gcode/pcb2gcode #483: Cut isolation slots drawn in Edge Cuts layer

hannesweisbach, 2020-10-17:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):

```
 pcb2gcode --version
2.1.0
Git commit: v2.1.0-61-g6738e21-dirty
Boost: 107300
Gerbv: 2.7.0
Geos: Not installed
```

What did you try (**include command-line arguments**):

No commandline arguments, see `millproject` below.

I'm trying to mill isolation slots drawn in the edge cuts layer of KiCad. #361 makes it sound a single line without a loop should be interpreted by pcb2gcode as a slot.

I tried the following, simple test example: 

[image]

- a closed line as actual outline,
- a single unclosed line (that does not end up in the gcode file), and
- a closed line that modeles a slot.

What happenned:

The generated gcode:

[image]

shows tool paths for both closed loops, but no tool path for the open loop slot.

What did you expect to happen:

I expected to that a tool path for the open loop line is generated that will result in a slot in  the PCB.

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

[Test.zip](https://github.com/pcb2gcode/pcb2gcode/files/5396076/Test.zip) that includes gerber file, millproject and generated outline.ngc.

`millproject`:
```
front=../Test-F_Cu.gbr
outline=../Test-Edge_Cuts.gbr

vectorial=true
metric=true
metricoutput=true

zsafe=1
zchange=15

zcut=-1.8
cut-feed=90
cut-vertfeed=30
cut-speed=30000
cut-infeed=0.2
cutter-diameter=0.8

zwork=-0.054
mill-feed=100
mill-vertfeed=30
mill-speed=30000
mill-infeed=0.018
mill-diameters=1.51mm,0.25mm
#default : optimise=0.000254
optimise=0.005
#offset=0.2
extra-passes=10

bridges=2
zbridges=-1.2
bridgesnum=4

nog81=true
zdrill=-0.03
drill-feed=30
drill-speed=30000
drills-available=0.1
```

#361 makes it sound like that open loop lines drawn in the "Edge cuts"-layer of KiCad result in tool paths for isolation slots. Or am I wrong/What am I doing wrong?

eyal0, 2020-10-17:
Wow, you are right!  It seems like one of the examples in the examples directory did the right thing after #361 but a different example is now doing the wrong thing!  How bizarre.  I will need to investigate.

If you look at the svg files here: https://github.com/pcb2gcode/pcb2gcode/pull/361/commits/daf54b7917d9bef0ac54b7d8cc45b00dd2197b50?short_path=3dd4a54#diff-3dd4a54ebf477799d12fb537923187b6aec9287c06cd1fb90d5746dcf68faac2

(You can compare them visually if you click on the "Display the rich diff" icon.)

You can see that the "overlap" example is doing a good job but the "broken" example has reverted to *worse* behavior.  This should not have happened and I should have noticed!

I'll take a look.

eyal0, 2020-10-17:
Okay, I think that I have an idea what it going wrong.  If you look at the .svg files created when you ran pcb2gcode, you can see that it finds the slot in `outp0_original_outline.svg` and in `processed_outline.svg`.  But not in `traced_outline.svg`.

Your cutter has a diameter of 0.8mm (it says in the millproject file).  I opened your gerber file in gerbv and when I click on the slot, it says that it was drawn with a tool that is 0.05mm.  That's too small for your cutter so it didn't cut!

[image]

When you specify a line that isn't a loop, it's treated as if it's an outline of the slot that you're trying to cut.  So if you want to cut a slot that is, say, 1mm wide, you need to draw a 1mm thick line.  Or draw like an oval shape that is a closed loop that is 1mm in thickness.  In that case, the thickness of the line itself is ignored.

Does that make sense?

Sorry that you're having so much trouble with the software!  Thanks for your patience.  I hope that you're having some success despite these setbacks!

hannesweisbach, 2020-10-18:
Thanks for you quick reply.

> Okay, I think that I have an idea what it going wrong. If you look at the .svg files created when you ran pcb2gcode, you can see that it finds the slot in `outp0_original_outline.svg` and in `processed_outline.svg`. But not in `traced_outline.svg`.

Yes, I noticed that. It led me to conclude my design was actually corrected because the isolation slot was apparently recognized by pcb2gcode.

> Your cutter has a diameter of 0.8mm (it says in the millproject file). I opened your gerber file in gerbv and when I click on the slot, it says that it was drawn with a tool that is 0.05mm. That's too small for your cutter so it didn't cut!

Yes, I tried with a line width  of 0.8mm and I didn't get a toolpath either. After your comment I tried again, because I thought I did something wrong but I didn't get a toolpath either. This time I tried also with a line width of **0.801mm** and I *did* get a tool path (although an oval, not a straight line - is it supposed to be an oval?)

I thought there was somewhere in the code something like `cutter_diameter > shape_width` that should have been `cutter_diameter >= shape_width`, but I wasn't able to find a code location like that. Maybe that's not even the case and the discrepancy is due to rounding.

> When you specify a line that isn't a loop, it's treated as if it's an outline of the slot that you're trying to cut. So if you want to cut a slot that is, say, 1mm wide, you need to draw a 1mm thick [...]

Yes, that does make sense.

I tried to confirm with the following gerber containing one straight line on the left that has a line width of 1mm and a rectangular cutout of 2x13mm drawn with a line thickness of 0.05mm:

[image]

With 0.8mm defined as tool diameter of the cutter I get a tool path from `X4.09999` to `X3.89997, i.e. 0.20002mm wide. with a 0.8mm wide tool that results in a 1.00002mm wide slot. A similar issue occurs on the rectangluar cutout which extends from `X11.39998` to `X12.59997` i.e. is 1.19999mm wide (+0.8mm tool results in a 1.99999mm wide cutout)

Not that my machine is that precise, what I'm getting at is that this might indeed be a rounding issue. Hm, do you get doubles directly from libgerbv? Maybe it is best to do calculations on ints? Or is this not possible because boost::geometry wants doubles?

> Sorry that you're having so much trouble with the software! Thanks for your patience. I hope that you're having some success despite these setbacks!

No worries :) Now that I know what to do it makes no difference to me. Having a thick line, or rather a line as wide as the slot I want, makes actually things easier for me in the design. I was just confused, because I tried 0.8mm line width with a 0.8mm  cutter and it didn't work. Making the line wider as the cutter is not a problem.

PS: What is your plan with GEOS? I have only little understanding what GEOS actually is, but I think you want to use it as replacement  for boost::geometry? Is this the case? Or does GEOS include functionality that is not included in boost::geometry?

eyal0, 2020-10-18:
You asked a lot of questions and the answer to all of them is that it has to do with doubles. 😁

To make a path we take a trace, inflate it by half the width of the mill, and then mill along the outline. If you have two Mills and the second one is smaller for touch ups in tight corners then we do the same procedure but subtract from the new trace the first trace inflated by half the width of that first bit. That's so we don't remill places that already got milled.

Boost geometry depends on all shapes being properly specified. That means, for example, that the hole inside of a shape mustn't extend beyond the outline of a shape. Or if you add two shapes together and they overlap, they should become one shape.

It all works fine unless you have shapes that just barely overlap or very close to overlap but don't. When an inflate operation works on those, it can sometimes output illegal shapes or leave slivers of area where there shouldn't be. This can really screw things up sometimes!

To mitigate the problem I add or subtract a tolerance number from some of the inflate or deflate operations. That way things which just barely overlap and might leave a sliver now overlap well and create good shapes. The other thing is that I noticed that the GEOS library seems to work better in some really pathological cases so if it exists, it is used.

So what you're seeing is not a rounding error but actually a small tolerance that is being added in to *solve* rounding errors. Most people don't notice because it's so small.

If your bit is the same width as the slot, when we expand the hole, it becomes a sliver with zero area. A shape with zero area has no perimeter so there's nothing to trace anyway.

I could try to rework it to get rid of all this munging but it's really delicate and when it fails, it's spectacular. So I just leave it. If you want, set tolerance to zero and see what happens. It might work!

As for the shape that is cut, it's rounded because you drew a line with a round aperture. You could draw with a square aperture and get a rectangle instead. Remember that a round milling bit won't be able to tuck into a corner all the way so you'll get a rounded corner even if the shape is rectangular. Cut fix that unless you broach it or something!

Not a perfect explanation but I hope that clears it up a little.

eyal0, 2020-10-20:
Re-open if you have more questions, thanks!

closed 2020-10-20 by eyal0

hannesweisbach, 2020-10-26:
Thanks for your explanation! I saw the tolerance variable in the source code but I didn't know what it was for.

Anyway, I always feel understanding how the tool works helps me to better utilizing it. Sounds like I will be using GEOS in the future too :)