pcb2gcode/pcb2gcode #488: Windows Build?

g4eml, 2020-12-01:
Has anyone managed to build Windows binaries for the latest versions of pcb2gcode and pcb2gcodeGUI? 

I have tried for several hours today to build my own by following the instructions here. However I had no success, I could not build gerbv and so could get no further.  The gerbv make command is failing with several 'Multiple definition' errors from the linker and I have not been able to resolve them.

eyal0, 2020-12-15:
This is related to #453

bjanders, 2020-12-16:
Related to this, README.md should remove the text "Windows prebuilt binaries (with all the required DLLs) are available in the release page." since there apparently aren't any prebuild binaries. I didn't spend a lot of time noticing this, so not a big issue, but kind of frustrating looking all over the place thinking you are doing something wrong.

eyal0, 2021-01-07:
This is fixed by b88a2fbd and 011336db and 6d15f726

closed 2021-01-07 by eyal0