pcb2gcode/pcb2gcode #489: Flip all Layers

jniclas, 2020-12-03:
Hey,
first of all, Great work! A simple, fast and refreshing Software to generate toolpaths.
Only thing I am missing is the ability to flip all layers incl. the holes for through hole boards. 
Is there already an option which i missed?

best regards,
Niclas

jniclas, 2020-12-03:
Found the option for choosing the side for cutout and drilling. So my problem is solved: I create backside tracks which can be flipped by the software and cutout and drilling can be done from the back too.
You could extend the documentation to point this out?!

best regards, 
Niclas

eyal0, 2020-12-03:
There's this: https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Milling#--front-vs---back

The Wiki is open to edits so if you have a suggestion, I think that you can just type it in.  I'm not sure how to write it so that it would be more clear.

I'm glad that it worked out for you in the end!  I hope that you're happy with the results!