pcb2gcode/pcb2gcode #490: Custom autoleveling references unset variables

Uks2, 2020-12-05:
pcb2gcode version:
2.1.0
Git commit: v2.1.0
Boost: 107200
Gerbv: 2.7.0
Geos: Not installed

What did you try (**include command-line arguments**):
```
pcb2gcode --front=./F_Cu.gbr --output-dir=./GCode --metric=true --metricoutput=true --mirror-axis=0.0000 --nog64=true --optimise=0.0000 --tile-x=1 --tile-y=1 --tolerance=0.0100 --zchange=10.0000 --zchange-absolute=false --zero-start=true --zsafe=5.0000 --extra-passes=0 --mill-feed=600 --mill-speed=10000 --offset=0.1000 --voronoi=false --zwork=-0.0500 --al-back=false --al-front=true --al-probecode=G31 --al-probefeed=100 --al-probevar=2002 --al-setzzero="G92 Z0"  --al-x=15.0000 --al-y=15.0000 --software=custom
```

What happened:
The output G code measures 12 points across the design, storing the results in variables `#500` - `#511`. The G code for milling the far right side of the PCB (in line with the last column of measurement points) references the unset variables `#512` - `#515` (what would be the next column of measurements) causing the G Code interpreter to report an error. The values in these variables are multiplied by zero when calculating the actual z coordinate of the tool, so they don't actually need to be measured.

What did you expect to happen:
Ideally, use a different equation to calculate the tool z position when directly between two measured points. A quick and dirty fix would be to set variables `#512` - `#515` (or equivalent for different designs) to some dummy value at the top of the output, knowing that they'll be multiplied by zero and wont affect the tool-path.

[Gerber](https://gist.githubusercontent.com/Uks2/8aeed682ddf0d042e9cc8e28140a95d0/raw/3990b7e669a06b2735b7d5d49ea4fbabeaff52b3/F_Cu.gbr)
[GCode](https://gist.githubusercontent.com/Uks2/8aeed682ddf0d042e9cc8e28140a95d0/raw/3990b7e669a06b2735b7d5d49ea4fbabeaff52b3/front.ngc)

eyal0, 2020-12-05:
Ah good!  I have never used autolevelling and the person that wrote the code wrote it a long time ago!  In fact, I have no idea if it works.  I'm happy to have someone that can tell me how it should look.  If you can suggest how the gcode output should actually look, I will do it.

By the way, is the only a problem of accessing too high a column or is it also a problem of access too low a column.

Another solution is to make the grid of sampled points a little bigger than the milled area so that it is never exactly lining up like you saw.

Uks2, 2020-12-06:
Accessing to low a column isn't a problem because if it's on the line, the algorithm will try to select points above and to the left. Incidently, I think the problem also exists when accessing too high a row, but trying to get a point above the top row gets you the bottom of the next row so the interpreter wont report a problem.

Looking at the code, it wasn't too complicated so I've tweaked the interpolation function to check if it's in line with any of the columns or rows of measured points and sent you a pull request (#491).

Like you said, making the grid of sampling points a bit bigger would also work and might be more resilient to floating point errors too.

eyal0, 2020-12-14:
Done, right?

closed 2020-12-14 by eyal0