pcb2gcode/pcb2gcode #511: Rendering edge cuts from KiCAD 5.x fails with "terminate called after throwing an instance of 'boost::numeric::positive_overflow'"

proffalken, 2021-01-16:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):

```
2.2.1
Git commit: latest
Boost: 107500
Gerbv: 2.6A
Geos: 3.8.1
```

What did you try (**include command-line arguments**):

I've exported my plots from KiCAD 5.x making sure that "gerber2" is switched off, however when I run the command, it causes a boost overflow and the gcode is not rendered.

pcb2gcode --back=D1MiniGSR/D1MiniGSR-B_Cu.gbr --drill=D1MiniGSR/D1MiniGSR.drl --front=D1MiniGSR/D1MiniGSR-F_Cu.gbr --outline=D1MiniGSR/D1MiniGSR-Edge_Cuts.gbr --metric=true --metricoutput=true --mirror-axis=0.0000 --nog64=false --optimise=0.0000 --tile-x=1 --tile-y=1 --tolerance=0.0100 --zchange=10.0000 --zchange-absolute=false --zero-start=true --zsafe=5.0000 --extra-passes=0 --mill-feed=600 --mill-speed=10000 --offset=0.1000 --voronoi=false --zwork=-0.0500 --drill-feed=1000 --drill-side=auto --drill-speed=10000 --milldrill=false --nog81=false --nog91-1=false --onedrill=false --zdrill=-2.5000 --bridges=0.5000 --bridgesnum=2 --cut-feed=600 --cut-infeed=10.0000 --cut-side=auto --cut-speed=10000 --cut-vertfeed=60 --cutter-diameter=3.0000 --fill-outline=true --zbridges=-0.6000 --zcut=-2.5000

What happened:
```
Importing front side... DONE.
Importing back side... DONE.
Importing outline... 
** (process:243716): WARNING **: 08:15:55.581: Most likely found a RS-274D file...trying to open anyways

DONE.
Processing input files... terminate called after throwing an instance of 'boost::numeric::positive_overflow'
  what():  bad numeric conversion: positive overflow
Aborted (core dumped)
```

What did you expect to happen:

SVG/GCode to be rendered

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

eyal0, 2021-01-16:
I tried using the same settings as you but with the example gerber files in the testing directory.  However, I didn't not have the same problem as you.  So it must be something specific to your gerber files.  I'm not sure how I could fix it without those files.

Are you able to attach those files?  If you want, you can try to run pcb2gcode without front or without back or without drill and try to determine which is the problematic file and then send me just that one.  That way if you want to keep your design a secret, I will only have one piece.  Up to you what you prefer.

After you send it, I can try it out myself and see how it goes.

Also, are you using windows or linux or macos?

eyal0, 2021-01-16:
I usually use Boost 1.66 but I also tried with boost 1.75 like you had and I didn't see the problem.  It must be something special in your gerber files.

You can also try to run pcb2gcode on the examples in the `testing/gerbv_example` directory.  Just enter one of the directories that has a `millproject` file in it and run `pcb2gcode`.  pcb2gcode will get all the parameters from the millproject and should run correct and make output files.

proffalken, 2021-01-17:
Hey @eyal0, sorry, completely forgot about the file upload, I've added the files I'm working with to the original text.

I'm also pretty sure it's an issue with the way KiCad is exporting the files, I just can't see how to turn off whatever it is that's causing this!

eyal0, 2021-01-18:
Ah, it's failing for me, too!  I'll take a look.

I don't know of any existing issues with exports from KiCad so I definitely want to understand this one.  You think that it's related to gerber2?  I was under the impression that gerber2 was just some extra comments in gerber files.  BTW, by gerber 2, you mean Gerber X2?

eyal0, 2021-01-18:
Looks like the svg_writer, which makes the debug svg outputs so that you can look at them, is trying to make an infinite sized svg.  That's weird!  It shouldn't do that!  svg_writer.cpp line 15.

Okay, I see an issue: Looks like your Edge cuts file is blank.  Like, it's got no lines in it.  So when pcb2gcode tried to figure out the size of the debug image that it was trying to make, it made infinity.  Because no outline means infinite size.  That's why the error was an overflow.

To see that it's the error, open up the Edge Cuts file and see that it's got nothing in there.  You can view it with gerbv or an online gerber viewer.

pcb2gcode ought to be friendlier about this.  I will make it so that when an input is empty, it doesn't fail.

eyal0, 2021-01-18:
BTW, if you remove the `--outline=D1MiniGSR/D1MiniGSR-Edge_Cuts.gbr` from the command that you ran, you'll find that the output is reasonable, I hope!

eyal0, 2021-01-18:
Last thing: I would like to make a test case out of your files so that I can ensure that this bug doesn't happen again.  Would it be okay if I use the files that you attached as part of the test code?

proffalken, 2021-01-18:
Hey, yeah, that's fine, this is just me playing around, I'll have a look at KiCAD and see why it's exporting a dud edges file, thanks for tracking that down!

proffalken, 2021-01-18:
@eyal0 - I went back into KiCAD and drew an outline on a different layer, it now renders correctly, so thanks for your insight into that!
It turns out that the layers in the template don't actually output anything when you `plot` them from Kicad, so you need to draw a polygon around your components to get the right shape.
Closing this as it's me holding KiCAD wrong, rather than anything pcb2gcode is doing.

commit a0f7eecc (2021-01-21): Ignore empty outline files

Only use the outline if it isn't empty.  Test that the min_corner is
below and to the left of the max_corner.

This fixes #511
closed 2021-01-18 by proffalken