pcb2gcode/pcb2gcode #512: backtracking is ignoring paths from larger bits

c-, 2021-01-16:
I noticed that when running a job with multiple cutter widths, the last few cleanup passes have a *lot* of z-lifts... far more than I'd expect with my backtrack setting. It seems that the smaller bit paths just won't consider passing through areas cleared by larger bits?

Patho(il)logical test case attached (a SOT-23-6 footprint milled with 0.8mm and 0.3mm bits).

This highlights the final toolpath for a 0.3mm bit once the 0.8 is finished:

[image]

With a 10000mm/s backtrack, I don't think that last bit should ever need to leave the copper.

It's not entirely clear to me whether the root of the problem is that the backtrack call just isn't being fed the paths from the larger bits, or if the engine doesn't consider backtracking to paths that are a cutter radius away, or something deeper/weirder?

[backtrack-multibit.zip](https://github.com/pcb2gcode/pcb2gcode/files/5824677/backtrack-multibit.zip)

eyal0, 2021-01-16:
You might be right about the milling being unwilling to go through a path that already exists.  However, the feature that you want is not called backtracking, it's called path-finding.

I didn't go into an explanation of those in the wiki but I should have.  I will do that.

Backtracking is for when you might be able to speed up a path by milling the same thing twice.  For example, say you want to mill this path:

[image]

If you try to draw that on a piece of paper without lifting your pen, you'll find that the only way to do it is to backtrack over a path.

The backtrack feature asks: How much am I willing to backtrack to avoid lifting my mill?

What you want is path-finding, which when pcb2gcode tries to connect traces together if it can do it without going through a trace.  The default for path-finding is 1, which means that it gives up after the first dead-end.  The higher that you make the number, the longer that it will take to run pcb2gcode.  But you can go pretty higher before you'll notice any slow down.

Try running `pcb2gcode --path-finding-limit 100000` and see that it will combine two milling paths.  So already that's better!

But still, those could have been all one path.  I will try to figure out what's going wrong there.  I think that it's like you suspect: The second mill bit is trying to avoid going where the first one went.  We do want to avoid because we don't want to re-mill area for nothing.  But later on when we're doing path finding, we don't need to avoid.  I'll take a look and see if we can do better.

For now, let me know if the path made with path-finding-limit=100000 looks acceptable to you.  If we were to connect the whole thing into one big oval, would that be better?

Hope that helps.

c-, 2021-01-17:
I'm actually starting the backside of a job right now, I'll see if that makes any difference.

c-, 2021-01-17:
Okay, I'll have to go back through this when I'm sitting down, but that behaved a lot more like what I was expecting (similar situation as that test case, but with an SOIC footprint).

eyal0, 2021-01-17:
Path finding is slow. It works like trying to solve a maze but because there are so many corners in a pcb, including all the little sections that make up a rounded corners, it takes a while.

On top of that, you need to figure out in what order to do path finding. If you've got 10 milling segments then you could potentially connect any end of those ten to any end of the other 9. And you have to do them in the right order.

So my plan so far has been to do path finding only among the milling paths that were created for the same trace. So like in your case, a single milling path is created around each of the 6 traces. And then they get trimmed because the first milling bit took case or some of it so the milling path for the second bit is just the tight spots. And those are disjoint.

The middle left one and the middle right one each generate two lines. It's those lines that are connected by path finding. But path finding didn't connect left to right because those a different traces.

I can try to add that path finding and see how it affects results and performance. I'll take a look at it

c-, 2021-01-17:
> We do want to avoid because we don't want to re-mill area for nothing.

But backtracking introduced the idea that it's only "for nothing" if it doesn't get us to the next path faster than a lift/rapid operation.

I guess backtracking as you've implemented can be thought of as a specific case of the more general optimization strategy of saving time by passing through *areas* that were already cleared... with backtracking, the area being passed through happens to be an exact path that was already covered.

> If we were to connect the whole thing into one big oval, would that be better?

Better, yes. Worth the hassle... that's a tougher call. My gut feeling is that some kind of performance hit is hiding behind the question of why the smaller bit won't cross the center, but speeding that up more than bumping up --path-finding-limit is likely getting past the point of diminishing returns. Unless it's something really obvious.

eyal0, 2021-01-20:
My explanation of backtracking wasn't perfect, because I sort of forgot how it works.

backtracking is asking the question: How much extra milling are you willing to do in order to save 1 second?  Your milling bit gets more worn whenever you do some milling so there's a trade-off.  Milling an extra centimeter to save 10 minutes, totally worth it.  Milling a 10 inch path to save 1 second... eh, not so great.  Because the trade-off is in distance versus time, the units are mm/s.

Now there's two ways that we can use that number:

1. We can do path finding to find a way to connect two different milling paths without lifting the tool.  You just need to find a way to connect the start of one path with the end of another without crossing through a trace.
2. We can just look at the paths that we've already milled and re-mill them.  This is useful for paths like the one that I drew above.

The second one is one that we always search for and we do a really good search for it because it's a super quick calculation.  If you don't like it, you just set backtracking 0.  And if all you care about is time, you set it to infinity, which is the default anyway.

The first one is much harder.  The search that needs to be done to connect two traces is computationally very expensive.  And if you have to do it for a whole lot of traces, it takes a really long time.  It's also not obvious which traces need to attach to which ones so you have to do them in the right order to get great results.  pcb2gcode solves this by only doing path finding for milling paths that were made for the same trace.  So if one trace generated two milling paths, like in your example the middle left and middle right ones, then path finding might put them together.  But if the milling path is not from the same trace then it won't try.

Mostly this works out pretty well because milling paths from different traces might be pretty far apart.  But sometimes it can miss some obvious connections, like you see in your example.

I've thought about maybe adding a final step where we try to connect paths from different traces.  It's not too hard to do but it might slow things down a lot for big PCBs.  I'll look into it.

c-, 2021-01-21:
> Mostly this works out pretty well because milling paths from different traces might be pretty far apart.

I'd estimate that 80% of the milling with smaller bits is cleaning up completely disjoint areas (usually thermal reliefs) where there's no alternative to a lift.

But when the smaller bit is used to route between the pins of footprints too small for the larger bits is where joining everything is really handy and while there's a definitely speed improvement just from cranking up --path-finding-limit (it also acts as a bit of a "spring pass" around the pads, which does help clean up  any burrs), it feels like it could try harder. The design that triggered this ticket involves about 50 pins worth of those cleanups, and the lift-move for every single one of those pins was driving me nuts.

eyal0, 2021-01-22:
I made a branch that does more path finding and the results are good... but it runs very slowly on large projects.  Especially if there is a ground plane!  Really big projects with a ground plane will have a lot of milling that is separate and can't be connected because connecting them would chop through the ground plane.  So it does a long fruitless search.

But you can try it out and let me know how you like it.  Here's what it did with your example:

[image]

It kind of snaked it's way around because it only tries to connect endpoints.  But it did something.  If you want to try it, you need to check out the `final_path_finder` branch and then build and compile in the normal way.

I'll keep thinking about how to make it run faster.

c-, 2021-01-22:
I'll check it when I have a chance. Not sure I'll be able to run any real tests until this cold snap passes and my workshop is comfortable again.

c-, 2021-01-24:
Had a go at the final_path_finder branch... I think it might be a bust.

You're correct about the "very slowly"; on one of my larger (90x100mm) but not excessively dense designs, it takes a little over twice as long (the back side is 320s vs 150s) and the bCNC statistics suggest that it shaves about 15 seconds off the overall runtime.

More concerning is this path where it does something very ... interesting.

[image]
[image]

The blue highlights all the cuts after a tool change from a 0.8mm cutter to a 0.3mm v-bit (actual width  approx 0.35mm at that depth) and it's entirely possible the path is staying within the bounds of the cleared area, but I have a bad feeling. This pattern showed up in a number of places:

[image]
[image]

c-, 2021-01-24:
I should add, those funky paths doesn't show up in the processed SVG's:

[image]

They should be in here, where the T joins, and there's nothing.

[image]

But I don't think I'm misinterpreting that there's no lift in there, so maybe the algorithm lost track of something that should be a lift-rapid and is just doing a rapid, or... I dunno?

eyal0, 2021-01-24:
> Had a go at the final_path_finder branch... I think it might be a bust.

Thanks for trying it!  Sorry that it's not very good yet.

> You're correct about the "very slowly"; on one of my larger (90x100mm) but not excessively dense designs, it takes a little over twice as long (the back side is 320s vs 150s) and the bCNC statistics suggest that it shaves about 15 seconds off the overall runtime.

bCNC's numbers are not totally to be trusted.  I'm not sure what bCNC thinks is the speed for retract and plunge.  It might be way off.  I looked into it once and found it had some mistakes but maybe they are since fixed.

bCNC usually arranges your milling into blocks:

[image]

Each block is one retract and plunge so if you count up the number of blocks before and after, that should give you an idea of how much up-down motion was eliminated.  And if you know the speed of those then you can get an estimate of how much time is saved.  Maybe that estimate will be better than the one that bCNC is providing?

> The blue highlights all the cuts after a tool change from a 0.8mm cutter to a 0.3mm v-bit (actual width approx 0.35mm at that depth) and it's entirely possible the path is staying within the bounds of the cleared area, but I have a bad feeling. This pattern showed up in a number of places:

The code is probably making a safe path that doesn't cross through any traces but yes, it's ugly.  The reason is that it will try to connect milling paths at the endpoints instead of at the closest points.  For example, say we want to mill two circles that are side by side:

[image]

We'd like them to look like a barbell:

[image]

But because the original loops start at the top of the first ring and the bottom of the second, it will actually make this:

[image]

It's weird-looking.  I need to modify the code to just connect anywhere instead of at the endpoints.  I'll see what I can do.

The slowness is because we're trying to do that path finding between every pair of milling paths.  So if you have 50 different little paths to mill, that's 50*50=2500 paths to examine!  It takes a while.  Maybe one day I'll figure out how to do the multi-threading and use all the CPUs in the computer to do it.  That would speed things up a bit.

The reason that it doesn't show up in the SVG file is my fault: I made the svg before adding these path connectors.  I can fix that, no problem.

So that you can understand what's going on, you PCB trace probably looks like the black shape below:

[image]

Because the large bit wasn't able to tuck into that corner circled in green, a small bit had to do it.  Same for some corner in the red circle and the yellow circle.  `pcb2gcode` figured out that it would be faster to mill between them rather than lift and plunge again, so it made that line.

For yellow, I guess that it didn't find a path.  Either yellow was too far away or the path-finding limit was too small so it gave up before finding that path.

It's weird that the path went from green, to red, and then back to green!  I would expect that it would just go one way!  I think that I know what went wrong there and also why it didn't go to yellow.  I think that it would be a quick fix and improve the paths a lot.  If I could get it to plunge at green and mill green, then red, then yellow, and then come up, would that be a good milling path?

If you send me more images of what pcb2gcode did versus what you expected, I can try to tune it better!  I'll also work on making the SVG file more useful.

c-, 2021-01-24:
> Each block is one retract and plunge so if you count up the number of blocks before and after

Hm. A bit of grepping indicates it's saving around 30 z lifts in total after the tool change (248 0.3mm blocks before, 220 after). This particular board has a *lot* of thermal reliefs that simply won't be a candidate for any sort of path finding (and I'm sure that's not helping the run speed) so finding that 30 in the remainder isn't too bad of an improvement if it weren't for how expensive it is to run the calculations (it also makes me wonder if a better performance improvement might come from not doing extra cleanup passes on thermal vias with smaller bits?)

You're probably right that bCNC's stats are a bit optimistic (usually a failure to model acceleration properly, which is going to be a dominant factor in short moves).

> The code is probably making a safe path that doesn't cross through any traces but yes, it's ugly.
> It's weird-looking.

That's exactly what made it jump out at me. In principle I don't have a problem with the path *if* it's staying in milled areas... after all, the entire premise of this issue is "why is pcb2gcode lifting the bit when it can just cut through those already cleared spaces", but that was a bit more aggressive and direct than maybe I was expecting.

> The reason that it doesn't show up in the SVG file is my fault: I made the svg before adding these path connectors. I can fix that, no problem.

Ah, that'll do it. Yeah, being able to check these artifacts in the SVG would make it a lot less concerning.

> If I could get it to plunge at green and mill green, then red, then yellow, and then come up, would that be a good milling path?

Definitely.

> If you send me more images of what pcb2gcode did versus what you expected

I'll see what I can dig up. Although I tried running it against some other designs and this is the only board that shows these kinds of artifacts, or really any noticeable difference.

That being said, it's obvious that it works. For example:

[paths.zip](https://github.com/pcb2gcode/pcb2gcode/files/5863193/paths.zip)

The before and after is significant:

[image]

[image]

So it works well in small areas and it's hard to imagine how that particular section could be optimised much further. I don't suppose pcb2gcode has some sort of internal concept of path "locality" that could be used to restrict these sorts of searches to "nearby" path segments?

eyal0, 2021-01-25:
> Ah, that'll do it. Yeah, being able to check these artifacts in the SVG would make it a lot less concerning.

Done.  The latest version will make another svg file with "final" in the name that shows all the milling that will be done.  It won't have the same coloring as the other svgs but it's still very usable.
 
> So it works well in small areas and it's hard to imagine how that particular section could be optimised much further. I don't suppose pcb2gcode has some sort of internal concept of path "locality" that could be used to restrict these sorts of searches to "nearby" path segments?

pcb2gcode is using locality.  All the possible ways to connect up those little milling paths are sorted from shortest to longest and then processed in order.  Because of a mistake that I made in the implementation, it will sometimes connect two paths together twice, at both ends.  It shouldn't!  I'll try to get a fix for that one soon.

As for making it run much faster, it might require a bigger rewrite.  I think that I'll first focus on making it work well, even if slow, and then see what kind of speed improvements can be done.  I have some ideas but it'll be a pretty big rewrite.  (Should be fun to do, though.)

eyal0, 2021-01-25:
Okay, I've made a new one.  Try it out and let me know how it works, please.  It'll still be just as slow but it should be smarter about what things get connected together.  Unfortunately I don't have a good example gerber on which to test it out.  If you've got one that you're willing to share, I'd appreciate it.

c-, 2021-01-25:
Here's the gerber that's been giving me problems.

[bunGrbl-Mega.zip](https://github.com/pcb2gcode/pcb2gcode/files/5867197/bunGrbl-Mega.zip)

(just a GRBL controller).

> It'll still be just as slow

Nope. Somehow, your new version runs way faster... down to 200s from 320s.

It still shows that cut across the top of the via (top left quadrant, just below-right of the SOIC-8), which the SVG indicates is staying in the milled areas.

[image]

And it's also doing some much better pathfinding in areas it was previously leaving disconnected:

[image]

*Definitely* worth an extra 50 seconds, IMO.

The thing that jumps out the most, I think, is thermals on square pads:

[image]

I assume it would join them it it had a more direct path, like:

[image]

eyal0, 2021-01-25:
> Here's the gerber that's been giving me problems.

Thanks.
 
> Nope. Somehow, your new version runs way faster... down to 200s from 320s.

Huh, go figure.  I'm using a different way to determine which milling paths should be connected now and which shouldn't so I guess that made a difference.

> It still shows that cut across the top of the via (top left quadrant, just below-right of the SOIC-8), which the SVG indicates is staying in the milled areas.

Some of those paths are worrying me:

[image]

I'm not convinced that the milling path isn't cutting into the trace.  It looks too close.  I should be able to measure it directly on the SVG files.

> The thing that jumps out the most, I think, is thermals on square pads:
> 
> [image]

Yeah, that's weird, and I suspect it might be related to the clipped corners that I'm seeing above.  The path finding algorithm is trying not to cross the outline of a trace and maybe it got the wrong idea about whether that milling is *inside* or *outside* the trace in the first place.  I'll take a look.  It should have figured that out because it's an easy connection there.

c-, 2021-01-25:
> I'm not convinced that the milling path isn't cutting into the trace. It looks too close. I should be able to measure it directly on the SVG files.

Huh. Didn't notice that, and now I can't unsee it. Interesting.

Ah, you're right, it is cutting into the trace, and I bet if you measure it it's almost dead on 0.007mm? 

I was using tolerance=0.07mm, which is normally not even noticeable, but here we are. If I drop that to 0.025mm, we get a much more reasonable:

[image]

But the calculation time *explodes* to 380s.

On one hand, it's cool that it's smart enough to cut the line as close as it can get away with. On the other hand, maybe for this application it doesn't really need to cut it so close to the edge that tolerance should be factored in.

eyal0, 2021-01-25:
Yes, so it's the tolerance.  I just did a test and saw the same.  The tolerance is a fudge factor.  Because there are a lot of floating point operations, it's possible to end up in a weird situation were mathematically a milling line shouldn't cross a trace but the calculation shows that it does.  Or vice-versa.  So all traces are shrunk by the tolerance before path finding.

The reason that it ran slower is because it had to make a lot more points to bend around that curve instead of going straight across.  Those extra points take more time.  I'll try to figure out where that extra time is and see what can be done.

eyal0, 2021-01-30:
Here's my latest attempt: https://github.com/eyal0/pcb2gcode/archive/ultra_find_path.zip

It should be even faster than the other one.  I haven't yet checked that the output is perfect but I eyeballed it and it looks pretty good.

c-, 2021-01-30:
On the larger board, about 50-60% faster than the current master branch.

I'm not seeing a *huge* difference in the output. It's still cutting into traces by up to the tolerance (although maybe a hair less aggressively?), and it still has that problem of "seeing around corners" on those square thermal pads.

On the other hand, I have another (smaller) board where there's a *lot* of regression. Maybe a consequence of using three different milling bits? I'll try to break it down into a usable test case to show the behaviour.

c-, 2021-01-31:
This pins it down:

[ESP32-finalpath.zip](https://github.com/pcb2gcode/pcb2gcode/files/5898982/ESP32-finalpath.zip)

Here's what the stock pcb2gcode was doing:

[image]

Cranking the path-finding-limit meant that the cleanup around the ESP32 pads is reduced to just a few tool lifts, which is really nice.

This is what the ultra_find_path is giving me:

[image]

Which amounts to a regression to what I had before cranking up path-finding-limit. It doesn't seem to be a consequence of having three bits because it still won't link up those small areas if I remove the 0.5.

eyal0, 2021-01-31:
It's working fine for me, I think that it just depends how far you crank it up.  Surprisingly, cranking up the path finding limit higher actually made it run faster!  Here's what I'm seeing:

| branch | path-finding-limit | run time (seconds) | tolerance (inch) | grep retract \| wc |
|--------|--------------------|--------------------|------------------|--------------------|
| master | 1                  | 4.76s              | 0.07             | 137                |
| master | 10000              | 6.55s              | 0.07             | 137                |
| master | 100000             | 7.81s              | 0.07             | 101                |
| master | 200000             | 6.2s               | 0.07             | 95                 |
| master | 300000             | 6.03s              | 0.07             | 94                 |
| master | 400000             | 6.03s              | 0.07             | 94                 |
| final  | 1                  | 7.78               | 0.07             | 134                |
| final  | 10000              | 13.34              | 0.07             | 134                |
| final  | 100000             | 10.44              | 0.07             | 99                 |
| final  | 200000             | 7.49               | 0.07             | 94                 |
| final  | 300000             | 7.44               | 0.07             | 93                 |
| final  | 500000             | 7.4                | 0.07             | 93                 |
| final  | 1000000            | 7.43               | 0.07             | 93                 |
| ultra  | 1                  | 3.84               | 0.07             | 134                |
| ultra  | 10000              | 6.27               | 0.07             | 132                |
| ultra  | 100000             | 20.65              | 0.07             | 127                |
| ultra  | 200000             | 9.4                | 0.07             | 94                 |
| ultra  | 300000             | 9.47               | 0.07             | 94                 |
| ultra  | 500000             | 9.64               | 0.07             | 94                 |
| master | 100000             | 12.73              | 0.0004           | 54                 |
| master | 200000             | 12.59              | 0.0004           | 49                 |
| ultra  | 200000             | 15.51              | 0.0004           | 41                 |
| ultra  | 100000             | 26.95              | 0.0004           | 63                 |
| master | 300000             | 10.22              | 0.0004           | 46                 |
| master | 400000             | 9.5                | 0.0004           | 45                 |

Setting the tolerance lower seems to make a difference, which is nice because it addition to giving a better output, you'll also mill into your traces less.  I didn't expect that.

I also have example gerber files where ultra did better than master.  It depends on how far apart the traces are.

I took the bunGrbl-Mega-B_Cu.gbr file that you sent and deleted a bunch of regions from it to make it a smaller example.

[bunGrbl-Mega-B_Cu.gbr.gz](https://github.com/pcb2gcode/pcb2gcode/files/5899177/bunGrbl-Mega-B_Cu.gbr.gz)

With the `ultra` version I'm getting 10.53seconds and 132 retracts.  With master, it's a shorter run time of 7.63 seconds but 194 retracts.  So in that case, it was really worth the extra runtime.

The difference is in the gerber.  Here's what the ESP one looks like:

[image]

The pads are already close to each other so even master can join them.  And master is faster at that.

[image]

Here the traces are far apart so there are potential connection in the milling that master is unable to make.  `ultra` is slower but it's able to do connections that for master aren't even possible.  That's because `ultra` tries to connect all the milling paths of the entire board.  Processing the whole board all at once is slower than processing it per trace but it allow whole-board connections.

I guess that it's back to the drawing board for me!  I think that the path finding is already really fast, it's just the processing of 1 whole-board vs many per-trace that is making the difference.  If I can get the former as fast as the latter, then it should be both good and fast.

We'll see.

c-, 2021-01-31:
> It's working fine for me, I think that it just depends how far you crank it up.

Wow. You're right, the new path finding seems to prefer much higher path-finding-limit values. That regression goes away around 400000, and indeed the resulting path is better than master; there's only really a small smattering of places where I think it maybe missed some opportunities to join paths.

The relationship to a tighter tolerance is non-intuitive. Maybe that constrains the size of the search space somehow by giving it less "wiggle room", but that's a hard one to wrap my head around.

Trying to decrease tolerance gets faster in simpler cases, but with the full bunGrbl-Mega board not so much.... even path-finding-limit=5000000 doesn't offset the expense of decreasing tolerance to 0.0025, and the resulting output is probably counter-productive (at the point that it's crisscrossing way too often through already cleared areas). At tolerance=0.007 (mm) and path-finding-limit=700000 the overall results are reasonable, IMO... it's capturing the "around the corner" cases in the thermals and the only place where it obviously seems to be "missing" something is that one pad on the SOT-23-6, but still runs faster than master. So that's pretty good. However, increasing path-finding-limit=1000000 starts to slow down again.

It seems like path-finding-limit, etc needs to be vary with the complexity of the board and other settings like tolerance, which suggests that it might make sense to try to specify optimization in terms of some sort of "effort" unit, ala "gzip -1" cs "gzip -9"? That is, assuming it's possible to automagically work out a relationship between effort, time, and all those variables.

eyal0, 2021-02-18:
I have a few new things to report:

* Lots of performance improvements were added to the master branch.  You might find a 50% speed-up.  The release notes explain the changes: https://github.com/pcb2gcode/pcb2gcode/releases/tag/v2.2.3
* I made a new branch called `segment_tree` that uses new path-finding code that is much faster than the old one.  You might find a further speed-up.  It will also do pathfinding between traces so you'll find fewer tool retracts.  As usual, you need to set the options `backtrack`, `path-finding-limit`, `g0-vertical-speed`, and `g0-horizontal-speed`.
* In order to make it easier to compare the milling, there are the `*_final_*` files, which have all the milling paths in them along with all the traces so that you can confirm that the path is good.
* I have created a new tool for comparing images because I couldn't find a good one online.  It's for visually comparing two SVG files and it can pan and zoom.  I'm finding it useful for comparing the output of different versions of pcb2gcode.  It requires a lot of python imports that you might not have and I haven't yet figured out the complete list. But it works one you get it installed. 
 Let me know if you try it. https://github.com/eyal0/image-diff-gui

On my stripped-down version of your grbl controller, the code at master took 6 seconds and the segment_tree code took 3 seconds.  New on the left, old on the right:

[image]

That's with `path-finding-limit=1`, the default. With `path-finding-limit=100000`, the runtime is 13 seconds.  Here's the diff between 1 and 100000.  Again, new on the left:

[image]

In some places it's doing better:

[image]

See how, now that the path-finding-limit is higher, it can make those round connections instead of having to zig-zag with dtraight lines?  I mean, way better:

[image]

Sometimes worse and I don't yet know why:

[image]

But retracts are saved.  116 retract at path-finding-limit 100000 vs 137 with path finding limit 1.  And with master it's 195 retracts!  so the new version is definitely doing better and with increased path-finding-limit, you can squeeze out a little more.

Let me know if you're able to try anything out.  Also, I recommend the image-diff-gui tool for inspecting the output.

c-, 2021-02-18:
I'll give it another spin. Our cold snap is finally coming to an end and I might even be able to cut some boards too.

closed 2021-02-19 by eyal0

c-, 2021-02-24:
I was able to run a pile of tests on several boards  over the weekend, both with master and `segment_tree`. Conclusions...

It's definitely faster. Can't complain about the speed. And `segment_tree` on top of that is smoking.

I'm having a bit of trouble comparing master directly to `segment_tree` with the same settings, but I agree that it seems to hug existing paths better, which IMO is preferable. Overall, either of them are an improvement, with `segment_tree` having a clear edge.

But... I found one case which neither version seems to be able to figure out. It's a board that uses three bits (0.8, 0.5 v, and 0.2 v), and the 0.5 bit is unable to use the areas already cleared by the 0.8. But the 0.2 has no trouble. Both master and `segment_tree` show the same sort of problem:

[image]

This is a test case built I've distilled from the larger project.

[ESP32-footprint.zip](https://github.com/pcb2gcode/pcb2gcode/files/6032424/ESP32-footprint.zip)

Interestingly enough, on my "real" board, it still does the same thing, *but* the 0.5mm bit is clearing a path between the footprint and the ground pour:

[image]

It's like those tiny points cleared by the 0.8mm bit act as some sort of blocker... ?

eyal0, 2021-02-24:
It's due to the tolerance.  The tolerance is meant to be a very small number, used for rounding.  If two traces are within tolerance of one another then it's like they are connected.  For path-finding, if the bit tries to squeeze between traces and it has less than tolerance of room to make it, it will fail.  So it looks like your bit is just barely squeezing through.  If I comment out the tolerance line, I get better results.  Run time increases to around 30 seconds.  Still not bad.

There's a weird artifact around the chip:

[image]

Zoomed in:

[image]

Even more zoom:

[image]

There are three milling lines separating the pad's trace from the ground plane under the chip.  One of the lines is milling near the ground plane, the other is milling around the pads.  They are really close to each other because the 0.81mm is not quite wide enough to do them both in one pass!  So it's needs two passes.

That middle pass is unfortunate.  It got done because of the isolation requirement.  The `millproject` file is asking for 1mm isolation and the first pass did just 0.81.  Each pass is processed individually so while doing the ground plane, it's unknown that the milling around the other pad would already give enough isolation.  Because we're only looking at one pad, we don't know that the rest of the milling will do the rest.

That extra pass is falling on the voronoi dividing line.  They are visible as faint colors here:

[image]

Maybe it would be possible to get pcb2gcode to not do that middle line if it's known that the other one will cover it.  I'll look into it.

c-, 2021-02-24:
> It's due to the tolerance. The tolerance is meant to be a very small number, used for rounding.

D'oh! I had a feeling it was something in my options, but forgot about tolerance. I found that increasing tolerance gave a bit of a performance bump with the older code. It doesn't seem to be doing that with the newer algorithms, so I should probably just stick with the default.

Thank you, that looks way, way better.

I'll need to rerun some tests, but `segment_tree` definitely still has an edge.

> Maybe it would be possible to get pcb2gcode to not do that middle line if it's known that the other one will cover it. I'll look into it.

On one hand, I'm not too bothered by the occasional extra pass since it means the bit is still doing work, even if it's just a spring pass (which actually does seem to help with cleaning up burrs and other artifacts of the process). On the other hand, that's more the job of `milling-overlap`.

eyal0, 2021-02-27:
I did find a fix for it.  There's almost no slowdown on the code and it decreases the milling, often a lot!  I've opened #555 to track it.

commit 2b8d10fc (2021-02-19): Merge pull request #540 from eyal0/segment_tree

This fixes #512