pcb2gcode/pcb2gcode #514: pcb2gcode 2.2.0 build failure

chenrui333, 2021-01-17:
👋 trying to build the latest release, but run into some build issue. The error log is as below:

```
configure: error: Package requirements (glibmm-2.4 >= 2.8) were not met:

No package 'glibmm-2.4' found
```

looks like after `autogen`, the `configure` could not find the glibmm installation.

Appreciate some help on figuring out the build failure.

---

relates to https://github.com/Homebrew/homebrew-core/pull/68428

eyal0, 2021-01-18:
For macos, right?

```
        brew install boost@1.75 gtkmm moreutils gettext librsvg autoconf automake pkg-config
```

You can see how it compiles successfully here: https://github.com/pcb2gcode/pcb2gcode/runs/1717716487?check_suite_focus=true  It's possible that the CI environment on github comes preloaded with some other brew install stuff so I'm not sure if that's a complete list.

The steps to compile come from the CI file which is here: https://github.com/pcb2gcode/pcb2gcode/blob/master/.github/workflows/ci.yml

If you search that file for mentions of `macos`, you'll find many of the steps that are used.  The CI file is pretty readable.

Thanks for doing this!  If there's a way for me to make it easier in the future, let me know!  For example, I have a compiled release on the releases page: https://github.com/pcb2gcode/pcb2gcode/releases/tag/v2.2.2  Is that useful or could I make it more useful?

Also note that I use a custom version of gerbv compilation because the default one has bugs and gerbv official development has ended.  The default gerbv might work okay for some cases but my version is better.

chenrui333, 2021-01-18:
yeah, I saw that CI, that is why I feel a bit weird.

Another interesting to note is the tarball size difference between the three os (macos one is significantly smaller).
[image]

> Also note that I use a custom version of gerbv compilation because the default one has bugs and gerbv official development has ended. The default gerbv might work okay for some cases but my version is better.

This is interesting, I am using the homebrew version of `gerbv`

chenrui333, 2021-01-18:
Also, I am not sure whether the `autogen` step would mess up the `configure` step. (not quite family with that `PKG_CHECK_MODULES` Macro)

eyal0, 2021-01-18:
@chenrui333 It might be that the macos tarball is smaller because libboost is dynamically linked for macos but statically linked for the others?  I'm not sure.

So did you try the `brew install` command that I wrote you?  It didn't help?  Maybe there is some other dependency that is needed for the macos build that is already pre-installed on the github CI runners?

One thing that you can try is to fork this repo and add a `brew list` line to the CI so that it will display everything that is installed.  And then compare that list to the list on your computer.

I don't know too much about autogen.  In the CI I use autoreconf.  I think that autoreconf replaces autogen.

PKG_CHECK_MODULES checks if a library is available.  It does this by making a small program and trying to compile it.  Sometimes it tries a few different combinations.

I think that your best bet is to make a fork and PR with `brew list` and compare that list to your own list.  Maybe something in there will be glibmm?

chenrui333, 2021-01-18:
looks like there is a new release? I will give 2.2.2 a try.

chenrui333, 2021-01-18:
> I don't know too much about autogen. In the CI I use autoreconf. I think that autoreconf replaces autogen.

interesting, sounds like i should use autoconf.

eyal0, 2021-01-18:
I use autoreconf, not autoconf. I don't know what the difference is.

Now that it's easier for me to make releases, because they are automatic, I
will release them more frequently. I try to use semantic versioning.

eyal0, 2021-02-19:
Were you able to sort this out?  I hope so!

Re-open if you need more help.

closed 2021-02-19 by eyal0