pcb2gcode/pcb2gcode #537: Allow for positive / zero zwork while milling

rtreffer, 2021-02-08:
**pcb2gcode version** (run pcb2gcode --version to see this):
```
2.2.2
Git commit: latest
Boost: 107100
Gerbv: 2.7.0
Geos: Not installed
```
(build from this repo)

**What I tried:**

I tried to mill solder mask, as explained e.g. in the [DIY Professional Double Sided PCB](https://www.instructables.com/DIY-Professional-Double-Sided-PCB/) tutorial.

Applying / curing solder mask, then milling the solder mask away sounded promising.

**Example command:**
```
pcb2gcode \
  --metric \
  --metricoutput \
  --front ./soic20-F_Mask.gbr \
  --output-dir ./gcode \
  --zsafe 1 \
  --zchange 10 \
  --zchange-absolute \
  --nog64 \
  --invert-gerbers \
  --mill-feed 20 \
  --mill-vertfeed 20 \
  --mill-speed 11000  \
  --mill-diameters 0.2 \
  --mill-infeed 0.03 \
  --zwork 0.1 \
  --front-output mask1.ngc
```

**What happenned:**

No actual paths were outputted. The engraving head moved from pad to pad without milling.

**What did you expect to happen:**

Milling above the Z=0 point to remove the added solder mask while reusing the height map of bCNC.

**Gerber file**

[soic20-F_Mask.gbr.txt](https://github.com/pcb2gcode/pcb2gcode/files/5944389/soic20-F_Mask.gbr.txt)

eyal0, 2021-02-08:
Weird, when I try to run it, it makes a file that it 100+Megabytes in size!  It's explained in the warning, where zwork is set to be higher than 0 (you have it at 0.1).  So it's trying to start at height 0 and mill downward 0.03mm at a time until it reaches 0.1.  But because 0.1 is above 0, it's never going to get there!  I have no idea how come yours actually terminated and didn't create an infinite file!

Anyway, I did get it working with different options.  

```
pcb2gcode \
  --metric \
  --metricoutput \
  --front ./soic20-F_Mask.gbr \
  --zsafe 1 \
  --zchange 10 \
  --zchange-absolute \
  --nog64 \
  --invert-gerbers \
  --mill-feed 20 \
  --mill-vertfeed 20 \
  --mill-speed 11000  \
  --mill-diameters 0.2 \
  --zwork 0.1 --mill-infeed=-0.03mm --extra-passes=10 --ignore-warnings
```

It mills at height 0, then higher and higher until it reaches height 0.1.  I understand that this is the reverse of the direction that you'd probably want!

[image]

If you remove the mill-infeed part then it will mill only at height 0.1mm.

Maybe if you describe your goal to me, I can see what needs adjusting.  Is it your goal to mill away the paint from heights 0 through 0.1mm?  Or do you just want to mill once, at height 0.1?  Or something else?

rtreffer, 2021-02-08:
I tried to mill at `-0.001` to overcome it, and this is what I got that way (finished about an hour ago)
[image]

This is with a 60°/0.1mm end. I did a few rounds with increasing mill diameters to cut out all the paint.
A quick test of the pads is positive, they are conductive and connected.. So I guess this is the result I want, except that the milling height of `-0.001` makes me a bit nervous.

I'll try drilling tomorrow. This is a simple soic20 / dip20 board that I wanted to try - mostly because I was curious if I could reach the required precision for soic20 at home.

rtreffer, 2021-02-08:
A single round of removing the paint would be enough. It is not a hard material, so I am not worried about the bit, my only concern is not damaging the copper and not readjusting the height (== using one height map before and after applying the solder mask)

eyal0, 2021-02-08:
Okay, so it sounds like you should set zwork to be higher than 0 but make sure that mill-infeed is not set, so that it'll do just one pass at the height that you want.

You don't need to do it with increasing mill diameters.  Just use the extra-passes option and it will work its way inward, like I show in my image above.  See how it does all the inner loops:

[image]

The distance between each pass is set by the milling-overlap, which can be either a percentage of the milling bit width or a distance.  The default is 50%, so if your bit is specified as 0.1mm wide, each extra pass will move inward by 0.05m.

Best of luck!  If you think of other features that you need, let me know.

closed 2021-02-11 by eyal0