pcb2gcode/pcb2gcode #545: Remove all copper that isn't in gerber file

synthead, 2021-02-14:
> Please fill this out:
> 
> pcb2gcode version (run pcb2gcode --version to see this):

```
$ pcb2gcode --version
2.2.2
Git commit: latest
Boost: 107500
Gerbv: 2.7A
Geos: Not installed
```

What did you try (**include command-line arguments**):

```
pcb2gcode \
  --metric \
  --metricoutput \
  --zero-start \
  --zsafe 0.5 \
  --zchange 0.5 \
  --front *-F_Cu.gbr \
    --zwork -0.03 \
    --mill-feed 200 \
    --mill-speed 1000 \
    --offset 0.06 \
  --drill *.drl \
    --zdrill -3 \
    --drill-feed 200 \
    --drill-speed 1000 \
  --outline *-Edge_Cuts.gbr \
     --zcut -3 \
     --cut-feed 200 \
     --cut-speed 50 \
     --cutter-diameter 0.5 \
     --cut-infeed 3 \
     --bridges 1 \
     --zbridges -0.08 \
  --output-dir gcode
```

What happenned:

Copper that does not exist in gerber file is left on the PCB.

What did you expect to happen:

Copper that does not exist in gerber file is cut from the PCB.

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

I have a gerber file that looks like this:
[image]

The keepout area for the copper fill is important because the SD card slot that will live on the PCB has exposed copper pins:
[image]

However, the resulting gcode from the command above leaves the copper on the PCB:
[image]

How can I mill away the keepout area?

Here is the gerber file:
[649-10067847-001RLF-breakout-F_Cu.gbr.zip](https://github.com/pcb2gcode/pcb2gcode/files/5977039/649-10067847-001RLF-breakout-F_Cu.gbr.zip)

synthead, 2021-02-14:
Seems related to https://github.com/pcb2gcode/pcb2gcode/issues/391.

eyal0, 2021-02-14:
Yup, it's related.  Have you tried the isolation-width option or the extra-passes option?  Maybe those will be good enough?

I can't think of a reliable way to indicate which parts of the PCB should entirely cleared away versus those that just need isolation.

eyal0, 2021-03-01:
I just thought, by the way: If you have regions that you want to mill away entirely, you can put them on another layer and then use the `invert-gerbers` option to mill away a shape.

So you'll want to run pcb2gcode twice: Once with your gerber and then once again with your shape that you want to mill away and with the `invert-gerbers` option on.  You should make sure to either use the `zero-start` option or have Edge-Cuts specified (or both), so that the two runs of pcb2gocde will create milling paths with the same alignment.

I hope that makes sense!

sybrenstuvel, 2023-07-21:
> I just thought, by the way: If you have regions that you want to mill away entirely, you can put them on another layer and then use the `invert-gerbers` option to mill away a shape.

This works fine for me to remove, say, copper in a rectangular area. However, for situations like these, it's getting a lot trickier:

[image]

Here the 230V traces need more clearance than the 12V ones.

eyal0, 2023-07-21:
Another way, I forgot to mention, you can specify multiple passes.  It will cause it to mill further and further from the trace.  This will eventually clear out the whole thing.  I don't recall if you tried this...

https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Milling#--isolation-width-prefer-this-instead-of-the-old---extra-passes

sybrenstuvel, 2023-07-21:
The problem is that the low voltage traces need 0.4mm clearance, and the high voltage ones need 2mm. How would I specify which trace gets which clearance?

eyal0, 2023-07-21:
If those low-voltage traces get too much clearance, is that a problem?  The milling will take longer but otherwise, is there any affect on the board?

Also, if you're anyway doing a ground fill then the extra milling will end where the ground pour starts so it limits itself.

sybrenstuvel, 2023-07-22:
Oooooh that works well! This is with `--isolation-width 2.0`:

[image]

Is there any reason to not use `--isolation-width 1000` for all projects?

eyal0, 2023-07-22:
Some users don't have a ground pour so it would be unnecessarily too much milling.