pcb2gcode/pcb2gcode #546: building pcb2gcode requires GUI libraries

codehero, 2021-02-15:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this): git tag v2.2.3

What did you try (**include command-line arguments**): ./configure

What happened:  No package 'gdkmm-2.4' found

What did you expect to happen: I did not expect a commandline only tool to require gtk or other gui libraries. I am trying to run pcb2gcode on my headless rpi and do not want to install extra dependencies

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

eyal0, 2021-02-17:
I understand.  gdk is unfortunately used by gerbv, which is the library that pcb2gcode uses for parsing gerber files.  I'm not sure if it is possible to compile gerbv without the graphics but I will try.

eyal0, 2021-02-19:
I don't see how to do it.  Looks like you'll need to install those libraries.  Is it so bad?

An RPi might be kind of slow at processing PCBs by the way.  It would probably be better to run this on a computer and then transfer the gcode over to an RPi if that's your CNC controller.

closed 2021-02-19 by eyal0

codehero, 2021-02-19:
Thank you for looking.