pcb2gcode/pcb2gcode #552: unnecessary bridges

c-, 2021-02-21:
I created a simple environmental sensor PCB, and I decided to get fancy and added isolation slots around the temperature sensor in my edge cuts.

pcb2gcode added bridges to this cutout.

[image]

I probably should have caught that before I milled the board. Fortunately, isolation is overkill for this application and it's not a big deal if I leave them in.

I'm thinking that maybe bridges aren't really needed on a path that isn't a closed loop?

[bridge-to-nowhere.zip](https://github.com/pcb2gcode/pcb2gcode/files/6018812/bridge-to-nowhere.zip)

eyal0, 2021-02-22:
That would be easily fixed, sure.  Is it the right thing to do?  What if the connection is really tiny?

By the way, can you tell me how you drew that edge cut in the middle?  Did you draw it as a closed loop (left) or as a thick path (right)?

[image]

c-, 2021-02-22:
I tried both. Both work about the same; you have to make the slot wider than the cutter width, and the difference is where it leaves the bridges. If you make the slot just a hair wider than the mill you'll mostly remove the bridges, but then your edge cuts all use that mill, which may not be desirable.

So there are tricks to mitigate it, but it just seems odd to have a bridge land in a space that gets milled away. And the cleanup isn't much fun.

For really tiny connections, bridges might still be useful. Although I'm not sure how thin such a feature could be and still support traces and have enough mechanical strength to be usable.

The gray area is where you have slivers of cutoff material left, like a 3mm slot milled with a 1mm cutter; maybe bridges are useful there? But a 3mm milldrill hole would have that same problem, and bridges aren't needed there either.

c-, 2021-02-22:
One approach to this problem might be an `internalbridgesnum` option to allow the user to specify a different number (including zero) of bridges on internal features. It's not ideal since it fails on a board that mixes different kinds of internal features (i.e. both large cutouts and slots) so a more automatic approach would still be better, but it works for the most common case.