pcb2gcode/pcb2gcode #554: Try to eliminate the milling on the voronoi line when there is enough overlap already.

eyal0, 2021-02-24:
It's due to the tolerance.  The tolerance is meant to be a very small number, used for rounding.  If two traces are within tolerance of one another then it's like they are connected.  For path-finding, if the bit tries to squeeze between traces and it has less than tolerance of room to make it, it will fail.  So it looks like your bit is just barely squeezing through.  If I comment out the tolerance line, I get better results.  Run time increases to around 30 seconds.  Still not bad.

There's a weird artifact around the chip:

[image]

Zoomed in:

[image]

Even more zoom:

[image]

There are three milling lines separating the pad's trace from the ground plane under the chip.  One of the lines is milling near the ground plane, the other is milling around the pads.  They are really close to each other because the 0.81mm is not quite wide enough to do them both in one pass!  So it's needs two passes.

That middle pass is unfortunate.  It got done because of the isolation requirement.  The `millproject` file is asking for 1mm isolation and the first pass did just 0.81.  Each pass is processed individually so while doing the ground plane, it's unknown that the milling around the other pad would already give enough isolation.  Because we're only looking at one pad, we don't know that the rest of the milling will do the rest.

That extra pass is falling on the voronoi dividing line.  They are visible as faint colors here:

[image]

Maybe it would be possible to get pcb2gcode to not do that middle line if it's known that the other one will cover it.  I'll look into it.

_Originally posted by @eyal0 in https://github.com/pcb2gcode/pcb2gcode/issues/512#issuecomment-784693842_

c-, 2021-02-27:
#555 is... well, ridiculous.

I think the only potential issue I can see is that the more fragile smaller bits *may* end up doing harder milling rather than just tidying up after the more durable larger bits. That's hard to check... maybe there'd be value to adding per-bit statistics to pcb2gcode?

I'm running a couple boards off this morning, so let's see how it performs on actual copper.

c-, 2021-02-27:
> I think the only potential issue I can see is that the more fragile smaller bits may end up doing harder milling rather than just tidying up after the more durable larger bits.

It does, but I'm not going to complain. The net effect seems to be more "outlining" of the small footprints you need the small bits on, rather than just milling slots to separate pads, which makes for much, *much* cleaner results; my SOT-23-6's have never looked quite this nice:

[image]

eyal0, 2021-02-27:
Harder milling?  What do you mean?

I should put a nicer explanation of what the PR does.  Here's goes:

Say we want to mill around these traces:

[image]

And let's assume that our milling bit is not very wide and we want lots of isolation, so we're going to do extra passes, either by setting the `extra-passes` option or by setting the `isolation-width` option.  First, we draw a voronoi diagram that divides the PCB into regions such that each region contains all the points closest to a single trace.  Like this:

[image]

One region per trace.  If you look at the lines that divide the regions, they are the dividing lines that are equidistant between traces.

When we do extra passes, no matter how many passes we do, there's no point exiting the voronoi region because the extra passes of the neighboring trace will anyway reach there.  Because if out extra pass is taking us far enough from the trace to reach the voronoi boundary then surely the extra pass of the neighbor will **also** reach that voronoi boundary.

The default overlap for milling is 50% (configurable).  That means that each pass will be 50% of the width of the tool further out from the trace.  So if the tool diameter is 2mm, the first pass will be at 1mm out.  The next pass will be at 2mm out, which is 1mm plus half the tool diameter.  And the next one at 3mm, etc.

In the case where we get so far out that we reach the voronoi boundary, we'll have a pass that will be **on** the boundary.  And it will also be on the boundary for the neighbor.  pcb2gcode already knows about repeated milling and it will eliminate it.

But seeing as the user only requested a 50% overlap, do we really need to go all the way to the voronoi boundary?  No, because we could be a distance away from the voronoi boundary and still have overlap with the milling from the neighbor.

So rather than go all the way to the boundary, we go to within 25% of the boundary.  The neighboring trace's milling will also go to within 25% of the boundary and so they will be 50% apart, just like the user asked for.

The result in practice looks like this:

[image]

You can see before on the left and after on the right.  The upper trace is the off-white color and the lower traces are the tan and dark green.  There are 3 milling lines between them on the old picture: There's the black line that is closest to the off-white trace, there is a black line that goes around the tan trace, and the one around the dark green.  On the left, there is also a third line, which between the other ones and that's along the voronoi edge.

On the right, that voronoi milling line is missing because pcb2gcode can see that the overlap between the milling of the shapes was sufficient so the voronoi milling line was skipped.

That's the plan.  I still see some bugs in there around the spike so it's not yet ready to merge but I think that it should work well and save a lot of milling time.

c-, 2021-02-27:
> Harder milling? What do you mean?

The smaller bit is removing more material, so it's working harder than it was when it was just eating at the areas the larger bit absolutely couldn't reach. But based on your explanation, that's unlikely, and in fact there should be less actual "work" being done by all bits.

I wonder if what I'm actually seeing is a side-effect of all those extra paths being removed; it's effectively acting as a multiplier for `--path-finding-limit` and I'm just getting more backtracking and/or path joining, etc with identical settings?

eyal0, 2021-02-28:
Don't know.  I guess take a design, analyze it both ways, and then load it into bCNC.  bCNC can report the total milling distance.  For the samples in the `gerbv_examples` directory, I think that they all got smaller and some drastically so!

c-, 2021-02-28:
bCNC doesn't do a per-bit breakdown, so it won't help identify whether there's been a shift of work from the larger to the smaller bit.

So I hacked up a perl script ... the rapid totals are a little off from bCNC, probably due to a different starting position and my script only counting after the first tool change.

[pathstats.zip](https://github.com/pcb2gcode/pcb2gcode/files/6057062/pathstats.zip)

For my bunGrbl-Mega board, master gives:
```
bunGrbl-Mega-B_Cu-back-0.8mm+0.3mm.nc
   0.80000    retract:  140 milling: 9054.64mm rapids: 810.81mm
   0.35000    retract:  172 milling: 2606.44mm rapids: 872.97mm
    totals    retract:  312 milling: 11661.08mm rapids: 1683.78mm
```

and `voronoi_shrunk`:

```
bunGrbl-Mega-B_Cu-back-0.8mm+0.3mm.nc
   0.80000    retract:  147 milling: 7761.87mm rapids: 867.78mm
   0.35000    retract:  172 milling: 2559.46mm rapids: 928.85mm
    totals    retract:  319 milling: 10321.33mm rapids: 1796.63mm
```

For the ESP-32 test case the respective numbers are:

```
front.ngc
   0.81000    retract:    4 milling: 570.00mm rapids: 32.33mm
   0.56000    retract:   21 milling: 254.25mm rapids: 151.83mm
   0.26000    retract:   18 milling: 464.12mm rapids: 91.36mm
    totals    retract:   43 milling: 1288.36mm rapids: 275.52mm

front.ngc
   0.81000    retract:    3 milling: 455.57mm rapids: 39.42mm
   0.56000    retract:   19 milling: 201.25mm rapids: 126.18mm
   0.26000    retract:   19 milling: 372.38mm rapids: 127.93mm
    totals    retract:   41 milling: 1029.20mm rapids: 293.53mm
```

Most other boards I tested are in line with this.

I have one design which goes against the grain and shows a tiny increase in the amount of milling done by the smaller bit, but whether that's "work" or backtracking/path joining isn't something that can be shown by coarse statistics like this. That same board shows an increase in the number of Z lifts.

To summarize, aside from that one case, `voronoi_shrunk` offers a considerable drop in the amount of milling work being done *particularly* for the large clearing bit, there's a small increase in the amount of rapids, and a small decrease in the number of Z lifts, and there's really nothing to support my belief that the smaller bit might be doing more or "cleaner" work.

eyal0, 2021-02-28:
Looks good to me.  Obviously we want a decrease across the board (haha!) but the priorities are definitely to decrease milling time and decrease the number of retractions.  Those are the slowest part.  The distance of rapid motion doesn't matter too much.  After all, it's *rapid*.

The rapid motion number should also be taken with a grain of salt.  The time to do a rapid move depends on the *angle*.  A CNC will do a rapid move as fast as it can, of course, but it can do diagonal moves faster than side-to-side ones.  For example, moving 14.1 inches along a diagonal will let it max out in both the x and y direction, which each move 10 inches.  But moving 14.1 inches along the X axis will go only as fast as the X axis can go.  Basically, the speed of the Y-axis motor is unused.

pcb2gcode know about this and so it considers, for example, a motion that goes x+5 and y+5 to be faster than x+6 and y+0.  And if you try it on your machine, I think that you'll find the former to actually be faster than the latter, even though the former is technically a further distance.  (This is called a Chebyshev distance and you can find mention of Chebyshev in the code.)

bCNC **doesn't** know about this so it can estimate time spent doing rapid moves wrong.  So it's possible that even the rapid moves have gotten faster!  But again, time spent on rapid moves is probably not a big deal.  It's the milling and retract that really matter.

c-, 2021-02-28:
>The distance of rapid motion doesn't matter too much. After all, it's rapid.

Yes, a 1% increase in rapid distance being traded off for a 10-20% decrease in milling distance and 5% of retractions is certainly a net win. In theory fewer long rapids (i.e. total rapids / total retractions) means the machine spends more time at top rapid speed rather than on the acceleration curve, so it actually should be a bit faster.

The ideal measure of efficiency would be a per-bit calculation of the area of material removed over milling distance, but that's a whole other rabbit hole.

commit b26f3356 (2021-03-01): Merge pull request #555 from eyal0/voronoi_shrunk

perf: Use shrunken voronoi to decrease milling

This fixes #554
closed 2021-03-01 by eyal0