pcb2gcode/pcb2gcode #566: Invalid sizes in SVG files

LordsWars, 2021-03-14:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):
**pcb2gcodeGUI v1.3.2-1**
What did you try (**include command-line arguments**):
**I work with a laser to create printed circuit boards.**
What happenned:
**Files * .svg when read in other programs are about 2 times larger than the original.**
What did you expect to happen:
**opening the svg file in the Snapmakerjs 2.6.0 program for further processing and creating the G-code specifically for the laser with turning the M3-M5 spindle on and off**
Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

g-code file from Snapmakerjs output
[original_drill.svg._20210314095136682.zip](https://github.com/pcb2gcode/pcb2gcode/files/6136109/original_drill.svg._20210314095136682.zip)
all the files I worked with
[gbr.zip](https://github.com/pcb2gcode/pcb2gcode/files/6136110/gbr.zip)
and this is the test result board with a screen and a sample output to the cutter :)
[image]

eyal0, 2021-03-14:
Can you also send the millproject file or the command line that you used?

eyal0, 2021-03-14:
Ah, I see that you're using the GUI.  I don't support that, it's really old code and it's not nearly as good at the new code.
Is it possible for you to use the CLI version?  It's not much harder to use than the GUI, it has more features, and it works on Mac, on Linux, and on Windows.  You can download it from the releases page: https://github.com/pcb2gcode/pcb2gcode/releases

LordsWars, 2021-03-14:
it is sad that you do not support the graphical view, well, what do you have to understand in the command line, because I have not yet found the fastest way to convert files specifically for laser cutting of printed circuit boards ... it's just that I'm now converting a 3D printer into a multi system with a quick replacement of the device head (milling cutter - laser - 3d printer)

eyal0, 2021-03-14:
The reason that I don't maintain the GUI is because it's too much work and too hard to keep it synchronized with the code. Everytime that I add a feature, I would need to expand the GUI. I'd someone else.wants.to develop the GUI, I would be willing to help! But I don't have the time.

Can you tell me more about how you are converting the files to laser printer inputs? Do you use the gcode file or the SVG file as input to the printer? Is your goal to laser around the traces or on them?

BTW, how do you use a laser for making a CNC? Does the laser remove copper?

There are many, many examples of millproject files in the examples directory of this project. If you copy one of them into your directory with the gerber's and modify the file names, I hope that it will work well for you.

eyal0, 2021-03-14:
Also the wiki pages have explanations and pictures of all the features. Click wiki at the top.

LordsWars, 2021-03-14:
how i make the pcb:
I prepare the textolite (I clean it from dirt), then dry it in the oven at 80 degrees for 30 minutes. then I stick a self-adhesive film on the textolite, preferably a dark color, but the white one burns the same :) Then I prepare the GBR and DRL files and process them in CopperCAM, then transfer them to FlatCAM and export them to SVG after loading Snapmakerjs where I process them specifically for the laser and at the output I get the G-code Well, at the end I load Repetier-Host for printing :) that's how it worked through the jungle. what I don't like about the L-code after CopperCAM processing is the constant operation of the laser at full power - idle movements with the laser on is not very good, and so the program completely suits me ...

LordsWars, 2021-03-14:
well, the etching itself - a tablespoon of homemade salt, a tablespoon of edible citric acid and 200 ml of hydrogen peroxide (there is always in the pharmacy), mix, heat up to 30 degrees, then throw the board into the liquid and wait 15 minutes ... take out the board and remove the protective film - at we have an almost finished board in our hands, it remains to be tinned

LordsWars, 2021-03-14:
Is your goal to apply a laser around or on the tracks?
ideally, the DRL file should burn the entire surface inside (with a drill diameter of 0.8mm, the entire surface of 0.8mm should be burned out) and the GBR file - the edges of the tracks are required, but this is only for low-current devices, and for high-voltage devices, the entire distance between the tracks must be cleared ... but this is ideal and of course it is unattainable :)

eyal0, 2021-03-14:
I don't understand it entirely, I think that I would need pictures.

So you use the laser to etch the textolite that you have attached to the copper. And then you peel it and use chemical etching?

Well, pcb2gcode can convert Gerber files to gcode, and I think that gcode should be useful for the laser printer, right? You might need to modify some.od the gcode so that it will turn on and turn off the laser at the right times. Maybe some of the pcb2gcode options will help you or maybe you need to modify the gcode a little. If it's a useful change I could add it to the program.

I suggest that you pick a millproject file, run the latest pcb2gcode, and see how different the gcode is from what you need it to be.

LordsWars, 2021-03-14:
again :)
we apply the film to the textolite.
we cut the film with a laser.
we etch a printed circuit board with a film (the laser burned the film to copper and these places are etched, and the rest is covered with a film and is not etched).
remove the protective film.

eyal0, 2021-03-14:
> Is your goal to apply a laser around or on the tracks?
> ideally, the DRL file should burn the entire surface inside (with a drill diameter of 0.8mm, the entire surface of 0.8mm should be burned out)

Usually there are two ways to CNC a hole in a PCB: either use a drill and poke a hole or use a mill and make small circles that will cut out a hole.

For you, I suggest the second option. It should draw little circles that will make holes. So that is the option called milldrill.

>the entire distance between the tracks must be cleared ... but this is ideal and of course it is unattainable :)

For this you should look at the isolation-width option. As the voltage gets higher, you can request more isolation so that tracks will have better separation.

If you are concerned about how much lasering and you want to save time, you can try the voronoi option which will split along the center path between traces instead of surrounding traces. It's faster but maybe you won't find the output attractive.

Like I said, try different options and look at the gcode and tell me what's missing and I will help you adjust.

eyal0, 2021-03-14:
I understand the process now. The laser is burning away the film which exposes the copper below to etchant chemical.

LordsWars, 2021-03-14:
Yes sir :)

LordsWars, 2021-03-14:
hmm ... the wiki introduces system hangs :) a lot of things and not a single example :)
why not --DIR pointer to the folder where all the files are stored and where new files will be created?
there is an implicit link to output files: "You can also specify the output directory with --output-dir; if not specified, output files will be created in the current directory."

hmm ... I re-read the wiki ten times ... everything seems to be clear, but the feeling that something is missing - there is still :) also a clumsy translation from English into Russian ...

I understand correctly that in the pcb2gcode-2.3.0 folder where the executable file is located - create the millproject file. then we fill it with what is needed and run and run the executable file and get what is required depending on the settings?

LordsWars, 2021-03-14:
[image]
I don’t understand why it doesn’t work

internals of the millproject file:
non-existant-option=true
front c:\pcb2gcode-windows-2.3.0\pcb2gcode-2.3.0\gbr\Top.gbr
zwork=0           	#Z-координата, на которой происходит гравировка
zsafe=10 			#Z-координата для перемещений между шагами гравировки
mill-feed=300 unit/minute #скорость подачи, при которой происходит гравировка (горизонтальная скорость)
mill-speed=10000 rpm 		#скорость вращения шпинделя при гравировке (об / мин)
optimise=true
dpi=3000
svg=true
metric=true
metricoutput=true
zero-start=true
mill-diameters=0.15
milling-overlap=0.05
pre-milling-gcode=M3 S10000
post-milling-gcode=M5 S100
front-output=front.nc

eyal0, 2021-03-14:
You forgot to put `=` after the word `front`.

Maybel it is having problems with the Cyrillic filenames? I have never use anything but ASCII filenames.

Also, you chose a millproject file that is intentionally broken for testing the behavior of unknown options. Don't pick that one! Remove the non-existent-option please.

LordsWars, 2021-03-14:
[image]

uploaded the resulting file outp0_original_front.svg

LordsWars, 2021-03-14:
waited 20 minutes for export to the front.nc file ... this does not happen with my computer - i3, 16gb of RAM, SSD 500mbit. Everything hangs memory all loaded SSD works like a victim ...
internals of the millproject file:
front = c:\pcb2gcode-2.3.0\gbr\Top.gbr
zwork=0           	#Z-координата, на которой происходит гравировка
zsafe=10 			#Z-координата для перемещений между шагами гравировки
mill-feed=300  #скорость подачи, при которой происходит гравировка (горизонтальная скорость)
mill-speed=10000 rpm 		#скорость вращения шпинделя при гравировке (об / мин)
optimise=1
mill-diameters=0.15
zchange=0
vectorial=1
metric=true
metricoutput=true
zero-start=true
mill-diameters=0.15
milling-overlap=0.05
pre-milling-gcode=M3 S10000
post-milling-gcode=M5 S100
front-output=front.nc

LordsWars, 2021-03-14:
As a result, I got an SVG file - the size is not the same at all. in height somewhere 52.7mm in diptrace and on the screen from file 125mm
[image]

The g-code is somehow strange and small ... when viewed by the program, only idle movements are visible while the size seems to be saved correctly
[image]
here are all the generated files
[result_files.zip](https://github.com/pcb2gcode/pcb2gcode/files/6137057/result_files.zip)
I have a feeling that I have not configured something :) by the way, when I turned off the optimization, everything is generated quickly

eyal0, 2021-03-14:
I will try.  I am using the millproject file from multivibrator example but I have turned off voronoi.

```
front=Top.gbr
drill=Through.drl
back=Bottom.gbr

mill-feed=360 in/min
mill-speed=200rotations/second
zwork=-0.04inch
drill-feed=30
drill-side=auto
drill-speed=12000
milldrill=true
onedrill=true
zdrill=-1.6mm
cut-feed=100
cut-infeed=0.19mm
cut-side=auto
cut-speed=10000
cutter-diameter=1.5mm
milldrill-diameter=0.8mm
zbridges=-0.6000
zcut=-0.13mm
fill-outline=true
zsafe=0.08
zchange=1.0
mill-feed-direction=any
```
The runtime for pcb2gcode is just half a second:

```
└──> time ~/private/pcb2gcode/pcb2gcode 
Importing front side... DONE.
Importing back side... DONE.
Importing outline... not specified.
Processing input files... DONE.
Exporting back... DONE. (Height: 1.92913in Width: 1.99094in)
Exporting front... DONE. (Height: 1.92913in Width: 1.99094in)
Importing drill... DONE.
Exporting milldrill... Warning: 14 holes were bigger than the milling tool.
Exporting drill... DONE. The board should be drilled from the FRONT side.
END.
~/private/pcb2gcode/pcb2gcode  0.42s user 0.01s system 99% cpu 0.434 total
```

Here's is the result of the front of the board, viewed using the bCNC program:

[image]

The ruler says that is is about 50mm in each direction.

Here it is in inkscape viewer for the SVG.  I put the ruler on top.  It also shows 50 mm:

[image]

Here it is in gerbv viewer.  It also shows 55 mm on each side:

[image]

The top of the SVG files has this header:

```
"http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg width="191.131" height="185.197" viewBox="0 0 3981.89 3858.27" version="1.1"
xmlns="http://www.w3.org/2000/svg"
xmlns:xlink="http://www.w3.org/1999/xlink">
```

It says width 191.131.  Standard is 96 pixels per inch, which is 3.78 pixels per mm.  191.131 divided by 3.78 is also around 50mm.  So it seems like all the sizes are correct.  Maybe the snapmakerjs is not working well?

eyal0, 2021-03-14:
I think that I will modify the generation of the SVG files to output units instead of assuming 96 pixels per inch, just in case.

Can you tell me: If you change the SVG at the top to show width and height in mm with `mm`, will it work better?

LordsWars, 2021-03-15:
I tried to run the program at work and in response, check the version of the program x32 or x64 .... at home it is Win 64 and at work 32 :( so checks only in the evening after work .... at work there is a CNC machine created by me personally with dimensions fields 320x200 and today I built a laser into it :) it remains to connect and change the controller, which began to freeze after executing 300-500 instructions, I don’t want to understand what’s the matter, because it’s a Chinese clone CNCUSBController and it will periodically turn on broken program always with problems :)
[image]

LordsWars, 2021-03-15:
yeah checked:
g-code almost corresponds to the required one, you need to cut test shawls :)
SVG has only become larger :) the main thing is that the g-code has become adequate ... now I'll run it for verification :)

LordsWars, 2021-03-15:
view in the program 
[image]
photo after execution of g-code
[image]

LordsWars, 2021-03-15:
figured out the dimensions in the g-code :) by adding
metricoutput = true
metric = true
got a normal code.
after printing, I noticed that the first movement of G00 occurs with the laser on, spoiling the picture.
look further

LordsWars, 2021-03-15:
internals of the millproject file:
`front=Top.gbr
drill=Through.drl
back=Bottom.gbr
outline=Границыплаты.gbr

metricoutput=true
metric=true
mill-feed=100 mm/min
mill-speed=200rotations/second
zwork=-0.01mm
drill-feed=30
drill-side=auto
drill-speed=12000
milldrill=true
onedrill=true
zdrill=-0.6mm
cut-feed=100
cut-infeed=0.19mm
cut-side=auto
cut-speed=10000
cutter-diameter=0.3mm
milldrill-diameter=0.3mm
zbridges=-0.6000
zcut=-0.13mm
fill-outline=true
zsafe=0.00
zchange=15.0
mill-feed-direction=any
pre-milling-gcode=M3 S10000
post-milling-gcode=M5 S100
output-dir=c:\pcb2gcode-2.3.0\gbr\
front-output=front.nc
drill-output=drill.nc
outline-output=outline.nc`

the resulting photo result:
[image]

the width of the tracks is something very small ... or the thickness of the laser is very large ... tomorrow at work, if I have time, I will write a code to calibrate the thickness of the laser, we will look for how much the laser needs to be raised in order to get a 0.15mm beam thickness. By the way, zwork only works in the negative range from -0.01, and with a positive value, it gives an error that is not very correct for a laser

eyal0, 2021-03-15:
The milldrill-diameter is only the diameter for making holes.  You need to set the `mill-diameters` option.  Use `mill-diameters=0.3mm` to specify 0.3mm laser beam thickness, for example.

`zwork` is usually negative because the CNC is drilling downward.  If that is not good for you, you can use a positive `zwork`.  It will cause an error but you can ignore errors by using `ignore-warnings=true`.

LordsWars, 2021-03-17:
photo of the original
[image]

photo what happened 
[image]
line width 1mm = 0.8mm
line width 0.5mm = 0.3mm
ring diameter 2mm = 2.4mm
mill-diameters = 0.1mm real laser beam thickness 0.15mm
internals of the millproject file:
```
front=up.gbr#Top
drill=Through.drl
back=Bottom.gbr
outline=Границыплаты.gbr

ignore-warnings=true
metricoutput=true
metric=true
mill-feed=100 mm/min
mill-speed=200rotations/second

zwork=5.01mm

drill-feed=100
drill-side=auto
drill-speed=10000
milldrill=true
onedrill=true
zdrill=-0.6mm
cut-feed=100
cut-infeed=0.19mm
cut-side=auto
cut-speed=10000
cutter-diameter=0.3mm
milldrill-diameter=0.2mm

mill-diameters=0.1mm

zbridges=-0.6000
zcut=-0.13mm
fill-outline=true

zsafe=5.00

zchange=15.0
mill-feed-direction=any
pre-milling-gcode=M3 S10000
post-milling-gcode=M5 S100
output-dir=c:\pcb2gcode-2.3.0\gbr\
front-output=front.nc
drill-output=drill.nc
outline-output=outline.nc
```

eyal0, 2021-03-17:
I don't have the files that you used so I can't test it.  Can you send:

up.gbr
Through.drl
Bottom.gbr
Границыплаты.gbr

Also, it will help me if you rename them to use ASCII because I cannot read Cyrillic text.

LordsWars, 2021-03-18:
I have a question:
Is it possible to set 0 in X and Y for drilling and milling (laser)?
The bottom line is that when creating a 2-sided board, we mill 2-4 control holes, then mill or burn out, BUT the relative position of 0 for drilling and laser is different, while the 0 of the machine may differ from the place where the board is attached. If you know the shifts by coordinates, is it possible to set them when creating the G-code? In large machines it is possible to set several (up to 3) houses, if there are such functions as enable?

eyal0, 2021-03-18:
https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Alignment-for-two-sided-PCB-and-offsets

You can set offsets. There is also an option for zero start that will choose if you want to use the zero that is in your input file or if you want to use a different zero.

For multi sided board you can choose the mirror axis. That is the line for flipping the board over.