pcb2gcode/pcb2gcode #571: zwork doesent seem to lift between most pads when --mill-feed exeeds 1102 mm/min

Efudd-byte, 2021-04-08:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):
2.3.0
Git commit: latest-dirty
Boost: 107400
Gerbv: 2.7.0
Geos: Not installed

What did you try (**include command-line arguments**):
Uh, using the pcb2gcodeGUI frontend:
pcb2gcode --output-dir=/home/kljhgv/CNC/PROJECTS --metric=true --metricoutput=true --mirror-axis=0.0000 --nog64=false --optimise=0.0000 --tile-x=1 --tile-y=1 --tolerance=0.0020 --zchange=20.0000 --zchange-absolute=false --zero-start=true --zsafe=0.5000

What happenned:
getting isolation cuts where travel should be, Also if --tolerance set too high, then NO lift of Z occurs at all...

What did you expect to happen:
--mill-feed behaves fine up to 1102 mm/min and --tolerance MAX works fine at that setting.
Would like to get ideal top engraving speed 3200-5000 mm/min 
since paneling in X and Y will take quite a long time @ 1100 mm/min...  

I use a laseraxe 2417 engraver (grbl1.1f) for pcb manufacture and use a 
pin scribe (loaded and spindle off) for high speed engraving.
Toolchain is kicad, pcb2gcode, candle2...

I was looking through git source i built, looking through main surface_vectorial and options*.*pp files
But I am not that skilled at C++ to find where the problem was.

TIA!

[image]
[the-files.zip](https://github.com/pcb2gcode/pcb2gcode/files/6275809/the-files.zip)

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

eyal0, 2021-04-08:
I'll need to run your code to investigate but at first glance it sounds like pcb2gcode is doing an optimization.

pcb2gcode has two ways to get from one milling path to another:

1. Lift the tool, quick move to the new spot, and lower.
2. Just mill from one spot to the other, going around any traces that might be in the way.

It will do whichever is fastest by default. If you want to change that behavior, you can modify some of the optimization flags that are listed in the wiki.

Let me know if that is enough help.

eyal0, 2021-04-08:
Ah, the wiki is a little out of date. Sorry about that! Please just read the output from `pcb2gcode --help`. It's in the optimizations section.

eyal0, 2021-04-08:
https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Optimizations#--backtrack

Looks like you should try with `--backtrack=0`.  Let me know if that makes a difference.

Efudd-byte, 2021-04-08:
Thanks for the speedy answer.
ran GUI output from command-line --tolerance=0.0000 and it gobbled up all my memory.  
good thing OOM daemon was installed :)
Warning: Deviation from commanded toolpath set to 0 (tolerance=0). No smooth milling is most likely!
...
pcb2gcode --front=/home/user/CNC/PROJECTS/example.grb --output-dir=/home/user/CNC/PROJECTS --postamble=/home/user/CNC/CONFIG/POSTAMBLE.ngc --metric=true --metricoutput=true --mirror-axis=0.0000 --nog64=false --optimise=0.0000 --tile-x=1 --tile-y=1 --zchange=20.0000 --zchange-absolute=false --zero-start=true --zsafe=0.5000 --extra-passes=0 --mill-feed=3200 --mill-speed=900 --offset=0.0500 --voronoi=false --zwork=-0.0010 --tolerance=0.0001 --backtrack=0

still getting zwork dragging on preview in candle.
in what file / location of the source will i find turning off option #2 of you reply? 

> I'll need to run your code to investigate but at first glance it sounds like pcb2gcode is doing an optimization.
>
> pcb2gcode has two ways to get from one milling path to another:
>
>     1. Lift the [...]

Efudd-byte, 2021-04-08:
looking in backtrack.cpp and .hpp files for clues, looks a bit complex :)

Efudd-byte, 2021-04-08:
Sorry, about that didnt mean to close, arthritis... :(

Efudd-byte, 2021-04-08:
Well i have tried --backtrack=0 , 1, 2 , -1, 0.0 and this seems to have no effect what so ever.
Any other ideas?

eyal0, 2021-04-08:
Ah, I see.  Also set `--path-finding-limit=0`

Before:

[image]

After:

[image]

Let me know how that works.

Efudd-byte, 2021-04-08:
Ah, you solved it!
Thanks, now i can rip through panalized boards with a scribe at full speed.
This is way better than visolate btw :).
Closing...

closed 2021-04-08 by Efudd-byte

Efudd-byte, 2021-04-08:
Well, not exactly from visolate, but is it possible when giving  --mill-speed=0 not to include gerber command M3 in the ngc output file?
On my inexpensive engraver any M3 command starts at minimal rpm, and since i have a stylus there, no sense in turning it.
( i just usually delete M3 before running the job anyway)

eyal0, 2021-04-08:
I think that there is already an outstanding issue for people that want more control over the M3.  #473 I think is related.

Currently you are just removing all instances of M3 in the file?

Efudd-byte, 2021-04-08:
Yeah, just the one instance, not a big deal, i could just sed out the gerber command in the output file. :)

eyal0, 2021-04-09:
Maybe it's best with sed.  There are already a lot of options in pcb2gcode and adding more of them for every type of CNC is making the list of options very long!

PR #573 (2021-04-09): fix: Use backtracking limit in path_finding
When `--backtrack` is not infinity (the default), then take it into
account not just when using the backtrack feature but also when doing
path finding.

This fixes #571
commit 1a7969bf (2021-04-09): fix: Use backtracking limit in path_finding

When `--backtrack` is not infinity (the default), then take it into
account not just when using the backtrack feature but also when doing
path finding.

This fixes #571