pcb2gcode/pcb2gcode #576: Feature Request : No flipping of --back option ( mill next to previous front mill job )

Efudd-byte, 2021-04-09:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):
2.3.0
Git commit: latest-dirty
Boost: 107400
Gerbv: 2.7.0
Geos: Not installed

What happenned:  <- happened :)
--back does - x boardsize offset for 2 sided boards ( not a bug:)
But, I normally use phenolic resin single sided stock of various thickness, and don't have to mount and align the stock twice.
It seems too hard to find "Double-Sided"  FR1, FR2 or CEM :( 
Also Non-glass type stock is Way Easier on drills and flat end-mills for cutting... 

Now i Could just as well load --front=back.gbr , or do --mirror-axis=SizeOfBoard+5mm ...

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

eyal0, 2021-04-09:
I don't understand what has gone wrong.  Can you explain?

You're doing a one-sided board.  So usually I would either specify only `--back` or only `--front`.  Usually for one-sided boards the copper is on the back.  Is the problem that the location of (0,0) when milling isn't working out?

You might look at the `--zero-start=true` option.  Your CNC probably can zero itself to whichever x,y you want.  I use bCNC, for example, so I press the button that draws the outline of the board and make sure that it traces over the top of the copper area.  If it doesn't then I job the CNC head over to the right spot and then zero it there.

Needing to zero a CNC is pretty common because the location of zero in the gerber file might have no relation to the location of zero on your CNC, which might be the bottom corner or the top corner or whatever.

Efudd-byte, 2021-04-10:
I am using pcb2gcodeGUI to generate files and have seen that after i set new home point,
the job wants to mill from bottom left instead of the bottom right. 

I left out the command-line in my posting,  sorry for the confusion.

 pcb2gcode --back=/home/user/CNC/PROJECTS/nano-v3/NanoV3.3/NanoV3.3/NanoV3.3-Bottom.gbr --drill=/home/user/CNC/PROJECTS/nano-v3/NanoV3.3/NanoV3.3/NanoV3.3.drl --front=/home/user/CNC/PROJECTS/nano-v3/NanoV3.3/NanoV3.3/NanoV3.3-Top.gbr --outline=/home/user/CNC/PROJECTS/nano-v3/NanoV3.3/NanoV3.3/NanoV3.3-Edge_Cuts.gbr --output-dir=/home/user/CNC/PROJECTS/nano-v3 --postamble=/home/user/CNC/CONFIG/POSTAMBLE.ngc --metric=true --metricoutput=true --nog64=false --optimise=0.0000 --tile-x=1 --tile-y=1 --tolerance=0.0020 --zchange=20.0000 --zchange-absolute=false --zero-start=true --zsafe=0.2500 --extra-passes=0 --mill-feed=3000 --mill-speed=0 --offset=0.0000 --voronoi=false --zwork=-0.0010 --drill-feed=200 --drill-side=auto --drill-speed=800 --milldrill=false --nog81=true --nog91-1=true --onedrill=true --zdrill=-1.0000 --bridges=0.2500 --bridgesnum=2 --cut-feed=300 --cut-infeed=0.5000 --cut-side=auto --cut-speed=1000 --cut-vertfeed=60 --cutter-diameter=1.0000 --fill-outline=true --zbridges=-0.6000 --zcut=-2.0000 --mirror-axis=0

I am doing a 2 layer board, using single sided stock , after --front=top.gbr  gets "milled" I home to new area 
+x 2 mm from the right of the first mill, I COULD then I run second job as --front=bottom.gbr, to 
Then i glue both "boards" together to make one, saving me the trouble of pulling up the 
I have a single sided board tack-glued to the spoil-board, only want to have no mill to the left done on --back 
It would be nice to have --back offset to the left in this case.

The GUI cnc job expects front and back to be in the same area.  
The thing is, The GUI ( not your fault ) is a bit limited for my requirements and I should stop using it for my method of manufacture.
( or until i run out of SS stock lol. )
I am just going to use your program since the cnc is dialed in now, --front=back.gbr, the simplest option lol.

Again another idea is for me to home --front left-most on copper blank, then --back  would have plenty of room 
to the right of the first job.
Thanks for your timely response.
Closing.

closed 2021-04-10 by Efudd-byte