pcb2gcode/pcb2gcode #577: Converting some easyeda gerbers causing "Internal error in gerberimporter" and crash

skotopes, 2021-04-10:
## Description of bug:

Converting easyeda generated gerbers causes pcb2gcode to crash.
Exception caused by`new_mls[0].front() != new_mls[0].back()` in simplify_cutins

## What did you try (include command-line arguments):

0.  Using https://myau.su/0w0/Gerber_PCB_2021-04-10.zip
1. `pcb2gcode --output-dir Gerber_PCB_2021-04-10/gcode --back Gerber_PCB_2021-04-10/Gerber_BottomLayer.GBL --outline Gerber_PCB_2021-04-10/Gerber_BoardOutline.GKO --drill Gerber_PCB_2021-04-10/Gerber_Drill_PTH.DRL --vectorial --voronoi --metric --metricoutput --onedrill --min-milldrill-hole-diameter 0 --nog64 --bridges 1 --zwork -0.04 --zsafe 1 --zchange 15 --zdrill -2 --zcut -1.8 --drill-feed 100 --drill-speed 10000 --mill-feed 100 --mill-speed 10000 --cutter-diameter 1 --cut-feed 100 --cut-infeed 0.6 --cut-speed 10000 --g0-vertical-speed 2000 --g0-horizontal-speed 2000 --tile-x 1 --tile-y 1 --ignore-warnings 1`

## What happened:

Application crash on uncaught exception

```
Importing front side... not specified.
Importing back side... DONE.
Importing outline... DONE.
Processing input files... Internal error in gerberimporter
POLYGON((0.265 0.42,0.266 0.35,0.37 0.35,0.35 0.41802))
MULTILINESTRING((0.265 0.42,0.266 0.35,0.37 0.35,0.35 0.418019))
```

## What did you expect to happen:

complete successfully

## pcb2gcode version (run pcb2gcode --version to see this):

```
2.3.1
Git commit: latest-42-gb3f55dd
Boost: 107500
Gerbv: 2.7.0
Geos: Not installed
```

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**  You might need to zip them.  If you don't provide a millproject file, make sure that you copy-paste the command-line that you used.

eyal0, 2021-04-10:
EasyEDA is making invalid contours.  According to [the spec](https://www.bosco.co.za/DownloadDocs/The_Gerber_Format_Specification.pdf) section 4.6.1, contours must be closed.  EasyEDA is making open contours.  Maybe it assumes that they will be automatically closed by the tools?

Anyway, it easy enough to fix this in pcb2gcode.  I already have some working code and I'll get a patch done soon.

```
─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
modified: gerberimporter.cpp
─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
@ gerberimporter.cpp:819 @ pair<multi_polygon_type_fp, map<coordinate_type_fp, multi_linestring_type_fp>> G
      } else if (currentNet->aperture_state == GERBV_APERTURE_STATE_OFF) {
        if (contour) {
          bg::append(region, stop);
          if (region.front() != region.back()) {
            cerr << "Repairing invalid contour (EasyEDA makes these sometimes)" << std::endl;
            bg::append(region, region.front());
          }
          draws.push_back(simplify_cutins(region));
          region.clear();
        }
@ gerberimporter.cpp:833 @ pair<multi_polygon_type_fp, map<coordinate_type_fp, multi_linestring_type_fp>> G
      contour = true;
    } else if (currentNet->interpolation == GERBV_INTERPOLATION_PAREA_END) {
      contour = false;
      if (region.front() != region.back()) {
        cerr << "Repairing invalid contour (EasyEDA makes these sometimes)" << std::endl;
        bg::append(region, region.front());
      }
      draws.push_back(simplify_cutins(region));
      region.clear();
    } else if (currentNet->interpolation == GERBV_INTERPOLATION_CW_CIRCULAR ||
```

skotopes, 2021-04-10:
> EasyEDA is making invalid contours. According to [the spec](https://www.bosco.co.za/DownloadDocs/The_Gerber_Format_Specification.pdf) section 4.6.1, contours must be closed. EasyEDA is making open contours. Maybe it assumes that they will be automatically closed by the tools?
> 
> Anyway, it easy enough to fix this in pcb2gcode. I already have some working code and I'll get a patch done soon. If you can't wait 1-2 days:
> 
> ```
> diff --git a/gerberimporter.cpp b/gerberimporter.cpp
> index f222791c..11a9eedd 100644
> --- a/gerberimporter.cpp
> +++ b/gerberimporter.cpp
> @@ -816,6 +816,10 @@ pair<multi_polygon_type_fp, map<coordinate_type_fp, multi_linestring_type_fp>> G
>        } else if (currentNet->aperture_state == GERBV_APERTURE_STATE_OFF) {
>          if (contour) {
>            bg::append(region, stop);
> +          if (region.front() != region.back()) {
> +            cerr << "Repairing invalid contour (EasyEDA makes these sometimes)" << std::endl;
> +            bg::append(region, region.front());
> +          }
>            draws.push_back(simplify_cutins(region));
>            region.clear();
>          }
> @@ -826,6 +830,10 @@ pair<multi_polygon_type_fp, map<coordinate_type_fp, multi_linestring_type_fp>> G
>        contour = true;
>      } else if (currentNet->interpolation == GERBV_INTERPOLATION_PAREA_END) {
>        contour = false;
> +      if (region.front() != region.back()) {
> +        cerr << "Repairing invalid contour (EasyEDA makes these sometimes)" << std::endl;
> +        bg::append(region, region.front());
> +      }
>        draws.push_back(simplify_cutins(region));
>        region.clear();
>      } else if (currentNet->interpolation == GERBV_INTERPOLATION_CW_CIRCULAR ||
> ```

Thanks, this patch works perfectly.

skotopes, 2021-04-10:
I've sent bug report to EasyEDA forums. Hope there they fix issue on their side too.
Thanks your amazing work.

eyal0, 2021-04-10:
Thanks for doing that. If you could, reply with a URL link to the issue that you raised on the EasyEDA forum. And also put a link to this bug on that forum. That way, if someone finds one, they will be able to find the other.

Enjoy!

skotopes, 2021-04-10:
> Thanks for doing that. If you could, reply with a URL link to the issue that you raised on the EasyEDA forum. And also put a link to this bug on that forum. That way, if someone finds one, they will be able to find the other.
> 
> Enjoy!

Here it goes:  https://easyeda.com/forum/topic/EasyEDA-generated-gerber-files-doesn-t-match-gerber-specification-71e36f539d4e4d5e837787ef1d280a4d

PR #578 (2021-04-11): fix: Close unclosed contours
According to the [Gerber Format
Specification](https://www.bosco.co.za/DownloadDocs/The_Gerber_Format_Specification.pdf)
section 4.6.1, a contour may only be ended by a D02 or G37 when the
contour is closed.  For example, a rectangle must have four line
segments, not three.

EasyEDA seems to be making shapes missing a line segment and assumes
that interpreters will close the contour.  `pcb2gcode` can support
this by checking that the region is closed.  If not, add a point at
the end which matches the first point so that the contour is closed.

This fixes #577
commit 5e8e6f80 (2021-04-11): fix: Close unclosed contours

According to the [Gerber Format Specification](https://www.bosco.co.za/DownloadDocs/The_Gerber_Format_Specification.pdf) section 4.6.1, a contour may only be ended by a D02 or G37 when the contour is closed.  For example, a rectangle must have four line segments, not three.

EasyEDA seems to be making shapes missing a line segment and assumes that interpreters will close the contour.  `pcb2gcode` can support this by checking that the region is closed.  If not, add a point at the end which matches the first point so that the contour is closed.

This fixes #577
closed 2021-04-11 by eyal0