pcb2gcode/pcb2gcode #579: component angular hole error

musarac, 2021-05-23:
Hello Friends
in all the pcb drawings I do, I sometimes have to use bottom components.The bord connection soldering legs of these components need to be drilled or cut in a quadrangular shape.But all the gcods I exported and bought don't show these holes.If this problem is not the fault of the program, I wonder if I am doing something wrong.
I install the example GBR dos and gcode example.

[image]
[image]
[image]
[zzzz - CADCAM.ZIP](https://github.com/pcb2gcode/pcb2gcode/files/6528612/zzzz.-.CADCAM.ZIP)

eyal0, 2021-05-23:
So, first up, I never use pcb2gcodeGUI and I don't support it.  It's just too much work for me.  I run pcb2gcode directly.

You didn't include your version of pcb2gcode so I don't know what you're running.  The best is if you can send me the command line that you ran.

I'll look at it some more tomorrow.

musarac, 2021-05-23:
[image]
pcb2gcodeGUI version

eyal0, 2021-05-24:
That's a super old version of pcb2gcode.  Definitely try again without the GUI.  I don't support the GUI, just the command-line version.

I think that the GUI will show you the command-line that was run.  Without the command-line, I don't know where to start.

musarac, 2021-05-24:
can you help me with how to use windows 7?

musarac, 2021-05-24:
Can I convert C to csharp to integrate your codes into cnc control software. can you help me on this?

musarac, 2021-05-24:
our job is really great and very simple. It would be great if you could update the GUI software.especially for Windows users.

eyal0, 2021-05-24:
Can you try to run the windows version?  It's here:

https://github.com/pcb2gcode/pcb2gcode/releases/download/v2.4.0/pcb2gcode-windows-2.4.0.zip

Press the "Windows" button, then type "cmd" and hit enter.  Then go to the correct directory to unzip and run pcb2gcode.  Let me know if you have trouble with it.