pcb2gcode/pcb2gcode #581: "Can't compare with units and without." exception when using milldrill-diameter and min-milldrill-hole-diameter

ndilieto, 2021-05-24:
Enabling milldrill-diameter and min-milldrill-hole-diameter
makes the program abort with the following exception:

>  Exporting milldrill... terminate called after throwing an instance of 'comparison_exception'
>    what(): Can't compare with units and without.

I traced it down and made a pull request to fix it: https://github.com/pcb2gcode/pcb2gcode/pull/580

eyal0, 2021-05-24:
Thanks.  Are you able to add a unit test for this or should I?  If you want to do it yourself, you can duplicate one of the directories in `testing/gerbv_examples` and modify the `millproject` file in there.  Then need to add that test to `integration_tests.py` and then run the test.

If it's too much for you then I'll do it.

ndilieto, 2021-05-24:
Thanks for the fast reply. I'd appreciate if you did it. Note, I added another commit to fix https://github.com/pcb2gcode/pcb2gcode/issues/582

eyal0, 2021-05-24:
Okay I'll do that.  Can you send me a sample command line that causes the failure?  Something like `pcb2gcode --milldrill-diameter=123 ...`  Or if you have a millproject file, that'll work, too.  Thanks.