pcb2gcode/pcb2gcode #587: Boost 1.76.0: /usr/include/c++/11.1.0/cstdlib:75:15: fatal error: stdlib.h: No such file or directory

synthead, 2021-06-13:
## Description of bug:

When building against boost 1.76.0, this error is raised during `make`:

```
make[2]: Entering directory '/home/max/aur/pcb2gcode-git/src/pcb2gcode'
  CXX      autoleveller.o
In file included from /usr/include/c++/11.1.0/ext/string_conversions.h:41,
                 from /usr/include/c++/11.1.0/bits/basic_string.h:6594,
                 from /usr/include/c++/11.1.0/string:55,
                 from autoleveller.hpp:29,
                 from autoleveller.cpp:20:
/usr/include/c++/11.1.0/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
   75 | #include_next <stdlib.h>
      |               ^~~~~~~~~~
```

This happens against the latest commit in the master branch at the time of this writing, 5459c2962e2fd9f059c629b143737053efa838cb.

Here is the full result of `autoreconf`, `./configure`, and `make`: https://gist.github.com/synthead/10b33c5d7828dabf6dce224cb1b7f730

This is on Arch Linux.

Packages:

- boost 1.76.0-1
- boost-libs 1.76.0-1
- gcc 11.1.0-1
- glibc 2.33-5

## What did you try (include command-line arguments):
1. Have an updated Arch Linux system available with boost to 1.76.0-1
1. Attempt to build pcb2gcode from [the AUR](https://aur.archlinux.org/packages/pcb2gcode-git) or [the common build steps](https://github.com/pcb2gcode/pcb2gcode#common-build-steps) (both reproduces the error).
2. Observe that the error above is displayed, and the build stops.

## What happened:

This error is raised during compilation:

```
/usr/include/c++/11.1.0/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
```

## What did you expect to happen:

No error is raised during compilation.

## pcb2gcode version (run pcb2gcode --version to see this):

_pcb2gcode does not build, and `--version` cannot be called._

eyal0, 2021-06-14:
stdlib is a pretty common thing to have on a computer. And the CI is running smoothly. This looks like a problem on your computer. Are you able to compile other things normally? Did you install build-essentials and other basics that are used for building code?

eyal0, 2021-06-15:
https://github.com/pcb2gcode/pcb2gcode/actions/runs/934578307

You can see there it compiled just fine on windows, ubuntu, and mac, multiple versions of each.

http://5.9.10.113/56397962/cmake-gtest-fatal-error-missing-stdlib-h-when-cross-compiling

Maybe it's related to that?  I *am* using -isystem for some headers.  If you look inside the configure.ac file, I think that you'll find mentions of `isystem` in there.

https://github.com/pcb2gcode/pcb2gcode/blob/master/configure.ac#L53

On that line where it says `-isystem `, carefully replace that with `-I`, making sure to remove the space at the end of `-isystem `.  You'll need to do it for every occurence in the configure.ac file.  Then try to build again.  Let me know if that helps.

fabalthazar, 2021-06-30:
Same issue on my Gentoo machine.

I succeeded in compiling by commenting out the whole following lines in configure.ac:
#AC_SUBST([BOOST_CPPFLAGS_SYSTEM], ['$(subst -I,-isystem ,$(BOOST_CPPFLAGS))'])
...
#AS_IF([test "x$HAVE_GEOS" = "xyes"],
#      [AC_SUBST(GEOS_CFLAGS_SYSTEM, ['$(subst -I,-isystem ,$(GEOS_CFLAGS))'])])
...
#AC_SUBST([gdkmm_CFLAGS_SYSTEM], ['$(subst -I,-isystem ,$(gdkmm_CFLAGS))'])

The first commenting was likely enough.

In case this can help understand the root cause.
I am not comfortable with autotools and libs used in this project so I honestly don't understand well what this hack implies.

eyal0, 2021-07-12:
I'm happy that it works for you now.  I'm not sure how to make a fix that will cause this to work for you and also to work for everyone else.  Hmm.

moeC137, 2021-09-19:
I have the same problem but under Arch.
I tryed the git version to same problem.

eyal0, 2021-09-19:
I'd like to explain what that line in the config is doing.

With the compilers, there are two ways to include a subdirectory for searching, `-I` and `-isystem`. The former is the common way and it tells the compiler where to search for header files that are included with `#include "foo.h"`. The latter is where to search for system files that are included with `#include <foo.h>`.

The other thing that isystem does is that you can ignore warnings that it generates. So if you have `-Werror` in your compile, which we have for this project, then the warnings that come from problems in libraries do not trigger the compile to fail.

That line in the config is changing boost's include type to be system, because boost has warnings.

moeC137, 2021-09-19:
i am not even sure if this is the same error anymore...its still generating error when compiling something with boost in it.

here is my output, maybe it helps even if its partly german lol

`yay -S pcb2gcode
:: There are 2 providers available for pcb2gcode:
:: Repository AUR
    1) pcb2gcode 2) pcb2gcode-git 

Enter a number (default=1): 1
:: Checking for conflicts...
:: Checking for inner conflicts...
[Aur:1]  pcb2gcode-2.4.0-1

  1 pcb2gcode                                (Build Files Exist)
==> Packages to cleanBuild?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> A
:: Deleting (1/1): /home/moe/.cache/yay/pcb2gcode
:: Downloaded PKGBUILD (1/1): pcb2gcode
  1 pcb2gcode                                (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> 
:: (1/1) Parsing SRCINFO: pcb2gcode
==> Erstelle Paket: pcb2gcode 2.4.0-1 (So 19 Sep 2021 11:58:38 CEST)
==> Empfange Quellen...
  -> Lade v2.4.0.tar.gz herunter...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   137  100   137    0     0    484      0 --:--:-- --:--:-- --:--:--   484
100 7282k    0 7282k    0     0   800k      0 --:--:--  0:00:09 --:--:--  859k
  -> patch gefunden
==> Überprüfe source Dateien mit sha256sums...
    v2.4.0.tar.gz ... Erfolg
    patch ... Erfolg
==> Erstelle Paket: pcb2gcode 2.4.0-1 (So 19 Sep 2021 11:58:50 CEST)
==> Prüfe Laufzeit-Abhängigkeiten...
==> Prüfe Buildtime-Abhängigkeiten...
==> Empfange Quellen...
  -> v2.4.0.tar.gz gefunden
  -> patch gefunden
==> Überprüfe source Dateien mit sha256sums...
    v2.4.0.tar.gz ... Erfolg
    patch ... Erfolg
==> Entferne existierendes $srcdir/ Verzeichnis...
==> Entpacke Quellen...
  -> Entpacke v2.4.0.tar.gz mit bsdtar
==> Beginne prepare()...
patching file configure.ac
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'm4'.
libtoolize: copying file 'm4/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
configure.ac:17: installing 'm4/compile'
configure.ac:18: installing 'm4/config.guess'
configure.ac:18: installing 'm4/config.sub'
configure.ac:13: installing 'm4/install-sh'
configure.ac:13: installing 'm4/missing'
Makefile.am: installing 'm4/depcomp'
parallel-tests: installing 'm4/test-driver'
==> Quellen sind fertig.
==> Erstelle Paket: pcb2gcode 2.4.0-1 (So 19 Sep 2021 11:59:00 CEST)
==> Prüfe Laufzeit-Abhängigkeiten...
==> Prüfe Buildtime-Abhängigkeiten...
==> WARNUNG: Verwende bestehenden $srcdir/ Baum
[... 270 lines cut ...]
In Datei, eingebunden von /usr/include/boost/optional.hpp:15,
                 von /usr/include/boost/format/internals.hpp:21,
                 von /usr/include/boost/format.hpp:38,
                 von surface_vectorial.cpp:58:
/usr/include/boost/optional/optional.hpp:1482:56: Anmerkung: Kandidat: »boost::optional<typename boost::decay<T>::type> boost::make_optional(T&&) [with T = boost::geometry::model::multi_polygon<boost::geometry::model::polygon<boost::geometry::model::d2::point_xy<double> > >&; typename boost::decay<T>::type = boost::geometry::model::multi_polygon<boost::geometry::model::polygon<boost::geometry::model::d2::point_xy<double> > >]«
 1482 | optional<BOOST_DEDUCED_TYPENAME boost::decay<T>::type> make_optional ( T && v  )
      |                                                        ^~~~~~~~~~~~~
In Datei, eingebunden von /usr/include/boost/container_hash/hash.hpp:111,
                 von /usr/include/boost/functional/hash/hash.hpp:6,
                 von bg_operators.hpp:6,
                 von path_finding.hpp:8,
                 von surface_vectorial.hpp:41,
                 von surface_vectorial.cpp:65:
/usr/include/c++/11.1.0/optional:1261:5: Anmerkung: Kandidat: »constexpr std::enable_if_t<is_constructible_v<typename std::decay<_Tp>::type, _Tp>, std::optional<typename std::decay<_Tp>::type> > std::make_optional(_Tp&&) [with _Tp = boost::geometry::model::multi_polygon<boost::geometry::model::polygon<boost::geometry::model::d2::point_xy<double> > >&; std::enable_if_t<is_constructible_v<typename std::decay<_Tp>::type, _Tp>, std::optional<typename std::decay<_Tp>::type> > = std::optional<boost::geometry::model::multi_polygon<boost::geometry::model::polygon<boost::geometry::model::d2::point_xy<double> > > >; typename std::decay<_Tp>::type = std::decay<boost::geometry::model::multi_polygon<boost::geometry::model::polygon<boost::geometry::model::d2::point_xy<double> > >&>::type]«
 1261 |     make_optional(_Tp&& __t)
      |     ^~~~~~~~~~~~~
make[2]: *** [Makefile:1144: surface_vectorial.o] Fehler 1
make[2]: Verzeichnis „/home/moe/.cache/yay/pcb2gcode/src/pcb2gcode-2.4.0“ wird verlassen
make[1]: *** [Makefile:1180: all-recursive] Fehler 1
make[1]: Verzeichnis „/home/moe/.cache/yay/pcb2gcode/src/pcb2gcode-2.4.0“ wird verlassen
make: *** [Makefile:900: all] Fehler 2
==> FEHLER: Ein Fehler geschah in build().
    Breche ab...
error making: pcb2gcode
`

schumar, 2022-01-22:
Seeing this error on Debian (bullseye) too:
```
make[2]: Entering directory '/usr/local/src/CNC/pcb2gcode'
  CXX      autoleveller.o
In file included from /usr/include/c++/10/ext/string_conversions.h:41,
                 from /usr/include/c++/10/bits/basic_string.h:6545,
                 from /usr/include/c++/10/string:55,
                 from autoleveller.hpp:29,
                 from autoleveller.cpp:20:
/usr/include/c++/10/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
   75 | #include_next <stdlib.h>
      |               ^~~~~~~~~~
compilation terminated.
```

`libstdc++-10-dev` (which provides `/usr/include/c++/10/stdlib.h`) is installed.

`./configure` says
```
checking for Boost's header version... 1_74
checking for the toolset name used by Boost for g++... configure: WARNING: could not figure out which toolset name to use for g++
```
not sure if this is related.

eyal0, 2022-01-22:
@moeC137 

Did you try this solution: https://github.com/pcb2gcode/pcb2gcode/issues/587#issuecomment-871668641

Let me know if it works.  If so, maybe I can figure out how to incorporate it.

3snowp7im, 2022-02-22:
> @moeC137
> 
> Did you try this solution: [#587 (comment)](https://github.com/pcb2gcode/pcb2gcode/issues/587#issuecomment-871668641)
> 
> Let me know if it works. If so, maybe I can figure out how to incorporate it.

I had the same issue and [#587 (comment)](https://github.com/pcb2gcode/pcb2gcode/issues/587#issuecomment-871668641) resolved it for me

eyal0, 2022-02-23:
> I had the same issue and [#587 (comment)](https://github.com/pcb2gcode/pcb2gcode/issues/587#issuecomment-871668641) resolved it for me

I'm building with errors enabled, though.  Does your solution work with this:

```
./configure pcb2gcode_CPPFLAGS_EXTRA="-Wall -Werror"
```

?

eyal0, 2022-02-24:
Maybe related: https://github.com/msys2/MINGW-packages/issues/10761

cinnaboot, 2025-07-26:
I was able to get this to compile on current arch linux by replacing -isystem with -idirafter, which I seems like it should have the save effect as marking them as system directories: https://gcc.gnu.org/onlinedocs/gcc/Directory-Options.html

```
diff --git a/configure.ac b/configure.ac
index 1210d41b..33b42b4f 100644
--- a/configure.ac
+++ b/configure.ac
@@ -50,7 +50,7 @@ BOOST_FORMAT
 BOOST_OPTIONAL
 BOOST_VARIANT

-AC_SUBST([BOOST_CPPFLAGS_SYSTEM], ['$(subst -I,-isystem ,$(BOOST_CPPFLAGS))'])
+AC_SUBST([BOOST_CPPFLAGS_SYSTEM], ['$(subst -I,-idirafter ,$(BOOST_CPPFLAGS))'])

 PKG_CHECK_MODULES([glibmm], [glibmm-2.4 >= 2.8])
 PKG_CHECK_MODULES([gdkmm], [gdkmm-2.4 >= 2.8])
```

found this suggestion here: https://github.com/Linuxbrew/brew/issues/724#issuecomment-397803958