pcb2gcode/pcb2gcode #600: Larger bit clearing leaving inexplicable small uncleared areas

c-, 2021-11-22:
Me again.

Designed a board where I wanted a lot of area cleared with a 0.8mm bit, with a 0.2mm to deal with some details. Everything is great *except* for this one little path the 0.2mm bit needed to take through the main area to clean up:

[image]

I had originally thought that was just the 0.2mm using an existing path to get to the other side, but examining processed_front_1.svg shows that there really are a bunch of tiny islands left by the 0.8mm bit:

[image]

It's pretty minor in the grand scheme of things, but also odd enough that I think it's worth the time to write this up.

Source:

[small_clearing.zip](https://github.com/pcb2gcode/pcb2gcode/files/7585068/small_clearing.zip)

eyal0, 2021-11-24:
Thanks for submitting this issue.  Yeah, it seems odd.

pcb2gcode is suppose to detect where the big bit was able to mill properly and where it wasn't.  And the it should clean up with the small one.  So that is what the small one is doing.  It's cleaning up in places where the big one wasn't able to tuck into a sharp corner.

The reason that they are all attached is due to path finding.  pc2gcode calculated that it would be faster to mill between them than to raise the bit, race over, and lower it again.  There are settings that you can adjust that will affect this.

But the real question is why all the cleanup is needed.  If we were milling between shapes then we'd expect to need some cleanup because if the big bit got too close, it would chew up the traces.  But this is happening in an area that has no traces!  pcb2gcode should be smart enough to know about this.  Hmm.  I'll look into it.

c-, 2021-11-24:
> But this is happening in an area that has no traces! pcb2gcode should be smart enough to know about this.

It's almost like it's not overlapping the different clearing "zones" sufficiently. As an extra data point, here's a shot of the milled board:

[image]

The obvious track from the 0.2mm bit is where I started paying attention to this, but another interesting artifact is those two little "dots" of copper left just below the center of the board. They're still very solidly attached even after a pass with sandpaper, so it's not just burring. Or maybe it's something like #554. where we have something that sits on the voronoi dividing line.

FWIW, those little islands will get cleared by the larger bit if you bump up the overlap to 60%, which in itself seems really odd.

c-, 2021-11-25:
> those little islands will get cleared by the larger bit if you bump up the overlap to 60%

Turning *down* the overlap increases the size of the islands. It definitely seems like some individual clearing paths are refusing to overlap into neighbour areas that are also being cleared.

Adjusting overlap in either direction doesn't show anything around those little leftover pieces of copper, so that might be a red herring; I've been running that bit for a while, could just be dull and deflecting, but I bet they'd be gone if overlap between areas was more aggressive.