pcb2gcode/pcb2gcode #604: v2.40 crashes with gerber file

rmisko11, 2022-01-06:
When processing Fritzing myLED_Fx_R_copperBottom.gbl pcb2gcode v2.40 processes for around a minute and  appearing to consume all my desktop ram (32GB) before reporting that the export was killed.  pcb2gcode seems to process the contour and drill files okay but not the '*.gbl' files.  I tried different '*.gbl' files and all crash.

Any help fixing this problem would be greatly appreciated,
  Roger

This is my edited version of my terminal output:
~/Projects/LED_Strips/WolfgangLED/gerber$ time ./pcb2gcode_V2_4.sh
2.4.0
Git commit: v2.4.0
Boost: 106600
Gerbv: 2.7A
Geos: 3.9.0
Importing front side... not specified.
Importing back side... DONE.
Importing outline... DONE.
Processing input files... DONE.
**Exporting back... ./pcb2gcode_V2_4.sh: line 70: 2700631 Killed**                  

/usr/local/bin/pcb2gcode --noconfigfile --outline=myLED_Fx_R_contour.gm1 --back=myLED_Fx_R_copperBottom.gbl --drill=myLED_Fx_R_drill.txt --output-dir=/home/roger/Projects/LED_Strips/WolfgangLED/gerber --metric=true --metricoutput=true --mirror-absolute=true --optimise=1 --tile-x=1 --tile-y=1 --tolerance=0.0100 --zchange=10.0000 --zchange-absolute=false --zero-start=true --zsafe=5.0000 --extra-passes=0 --mill-feed=600 --mill-speed=10000 --milldrill-diameter=2.000 --offset=0.1000 --voronoi=false --zwork=-0.0500 --drill-feed=1000 --drill-side=back --drill-speed=10000 --milldrill=false --nog81=true --nog91-1=true --onedrill=true --zdrill=-3.0000 --bridges=0.5000 --bridgesnum=2 --cut-feed=600 --cut-infeed=10.0000 --cut-side=auto --cut-speed=10000 --cut-vertfeed=60 --cutter-diameter=1.0000 --fill-outline=true --zbridges=-0.6000 --zcut=-2.5000

real	0m58.203s

[Gerber.zip](https://github.com/pcb2gcode/pcb2gcode/files/7818891/Gerber.zip)

eyal0, 2022-01-06:
That's a pretty difficult-to-process gerber file.  The really big fill is actually many many many small contours, which is a lot of work for pcb2gcode to process.

Hmm, I tried it with different options and it worked really well!  So I think that one of the options in your command-line is giving pcb2gcode a hard time.  Let's see if I can figure out which one with trial and error...

Ah, it was the `--optimise`.  The default is 0.000001 or something like that, really small.  You made it too large and things got weird.  Removing just that one line got me a good result:

[image]

That's the path that the CNC will take.  Not bad.

Fritzing is not doing a good job of making round thermals.  Look at this one:

[image]

Not quite round!  The CNC is doing the same.

Anyway, remove the `optimise` and you're good to go.  Enjoy!

closed 2022-01-06 by eyal0

rmisko11, 2022-01-06:
Wow, thank you very much for looking at it so quickly and finding the
solution.
I misunderstood the 'optimise' option, I thought it was true/false or
enable/disable.
I noticed the fritzing gerber were not very nice, the pdf files look much
better.  Maybe I have an incorrect setting there too.  I will look to see
if I can improve the output.

Again, thank you very much for your prompt response,
  Roger

eyal0, 2022-01-06:
optimise used to be a true/false thing but then I changed it to work differently.  That's why I didn't notice what you did immediately.  Best of luck!