pcb2gcode/pcb2gcode #608: custom g-code after M6

AlexeyGusev, 2022-01-21:
Please consider adding an option to insert custom g-code after each M6 command, similar to e.g. existing preamble:
```
 --preamble filename.ngc
              gcode preamble file, inserted at the very beginning
```
so proposed feature could look like this:
```
 --m6postamble filename.ngc
              gcode file, inserted after each M6 command
```

Rationale: low-end mills that require manual tool change need Z-zero adjustment after each tool change. Despite the fact that LinuxCNC allows complete remapping of M6 command, [implementing it properly](https://linuxcnc.org/docs/html/remap/remap.html#_remapping_tool_change_related_codes_t_m6_m61) is cumbersome, requires complex setup and is excessive for given task. Running a custom O-word procedure (dynamic tool offset and friends) after M6 command could be near–perfect solution, and automating code injection in proper places by `pcb2gcode` itself will simplify workflow by obviating generated g-code post-processing.

eyal0, 2022-01-22:
Oh man, how annoying is LinuxCNC!  I used bCNC in the past and it had a reasonable tool change procedure: https://github.com/vlachoudis/bCNC/wiki/Tool-Change

It did the change and then a tool probe to find the tool length offset (TLO).  You mean that LinuxCNC isn't doing that by default?

I'm wary of adding tons and tons of options because it makes the code more complex each time.  I dunno.

AlexeyGusev, 2022-01-23:
LinuxCNC, AFAIK, is meant to drive high-end(-ish) mills that have tool changers etc. In such use case, you can specify TLO for each tool in tool table, and that value stays, because with tool changer TLO is always the same for the same tool. We hobbyists are, of course, working with poor man's machines and change tools with a wrench. 

LinuxCNC also has an option to remap M6 to custom routine, but this is an awkward solution because one must implement all the M6 stuff that LinuxCNC does behind the scenes. Post-processing looks like a better idea.

I understand the concept of keeping the code clean, orthogonal and as simple as possible, and appreciate the approach. How about a more generic form then that can be useful is potentially any use case?

```
--command-postamble COMMAND filename.ngc
              insert contents of filename.ngc after each occurrence of COMMAND
```

So you could specify `--command-postamble M6 filename.ngc` or `--command-postamble M0 filename.ngc` or `--command-postamble G21 filename.ngc` etc.

Otherwise, I will just stick to my `sed '/^M6.*/a O<probe_tool> CALL' back.ngc > back-processed.ngc` and shut up :)