pcb2gcode/pcb2gcode #610: should drill sizes always be ascending order by size?

c-, 2022-02-02:
I output a combined PTH and NPTH drill file from Kicad, and ended up with a gcode file containing:

    ( This file uses 6 drill bit sizes. )
    ( Bit sizes: [0.35mm] [0.6mm] [0.7mm] [0.8mm] [1mm] [0.9mm])

I'm pretty sure I know where that ordering came from; the 0.9mm was only used for NPTH while all the other drills were used for PTH holes, and presumably pcb2gcode processed NPTH second and just appended the drills it needed instead of sorting them into a common list.

Let me know if you think this is a real bug and I'll add a trivial test case.

eyal0, 2022-08-30:
Yeah, the drills aren't sorted.  Does it matter?

I'm not sure if there is a sort between the different drills.  It probably wouldn't make much difference in which order the different drills are used because you need to move the drill to the toolchange position each time anyway.

I could sort but it doesn't really matter, I think.

c-, 2022-09-01:
> Does it matter?

The more I think about it... yes.

I guess my concern is that you almost have to go out of your way in order for Kicad+pcb2gcode to generate a drill gcode file where the drills *aren't* all in ascending order (I assume drill ordering is mostly driven by Kicad if you're not explicitly sorting). So a user can get used to that ordering, and expect it, and when laying out drills for any job might get into a habit of laying out drills in ascending order. And one day, instead something trivial like mixing up the 0.9mm and 1.0mm drills, it's the 1.2mm and 0.6mm bits that get switched. It's not just a matter of keeping track of which tool, but if someone is using a machine where they have to set the RPM's outside of the gcode (i.e. a Dremel) then consistency becomes even more important.

I suppose an alternative to sorting might be to always output drills in the order of drills-available, which might be even more intuitive.

closed 2023-02-09 by eyal0