pcb2gcode/pcb2gcode #622: Wrong gcode generate

bbernhard, 2022-04-29:
Hello,

When I try to translate this gerber file 

[image]

with `pcb2gcode` into gcode, I end up with the following gcode (the below screenshot comes from bCNC):

[image]

My `millproject` file looks like this:

```
back=attiny85-breakout-F_Cu.gbr
#outline=attiny85-breakout-B_Silkscreen.gbr
drill=attiny85-breakout-PTH.drl
metric=true
metricoutput=true
#software=custom
nog64=true

# milling
mill-feed=2500
mill-speed=0
mill-diameters=0.05
extra-passes=2
zsafe=2.0
mill-vertfeed=400
zwork=0

# drilling
zdrill=-1.8
zchange=2.0
drill-feed=300.0
drill-speed=10000
nog81=true

# outline cutting
cutter-diameter=3.175
cut-feed=60
cut-infeed=0.6
cut-vertfeed=120
cut-speed=3600
zcut=-1.8
```

Not sure if it is related, but when I invoke `pcb2gcode` with the above `millproject`, I get the following warning:

`Warning: pcb2gcode hasn't been able to fulfill all clearance requirements.  Check the contentions output and consider using a smaller milling bit`

Is there something I am doing wrong here? The `mill-diameters` is already set to `0.05` (which is pretty tiny). So not sure what I am doing wrong here :thinking:. I've used more or less the same settings in flatcam and already successfully milled a PCB that way. So I think the settings in the `millproject` file shouldn't be completely off. 

On a related note: Is the man page more update to date or the output of `pcb2gcode --help`? I've noticed that `pcb2gcode --help` shows the `mill-diameters` parameter in its output whereas the man page doesn't show it.

I've attached a zip with the gerber/drill files and the millproject file I am using.

I am using the latest pcb2gcode (freshly checkout yesterday from the master branch) on my Ubuntu 20.04 system. The attached gerbers were created with Kicad 6.

[pcb2gcode-example.zip](https://github.com/pcb2gcode/pcb2gcode/files/8592180/pcb2gcode-example.zip)

Any help is greatly appreciated!

Many thanks,
Bernhard

bbernhard, 2022-04-29:
Ok, so I manually deleted some traces and labels from the gerber file to see whether I can get rid of the warning and if that fixes the problem.

My gerber now looks like this:

[image]

And although the above warning is now gone the generated gerber still looks strange:

[image]

bbernhard, 2022-04-29:
Ok, looks like the problem was the `zwork` parameter which was set to `0` (I misunderstood what the parameter actually does).

Not sure if the current behavior is a bug or WAD, but in case it is a bug, I think it could help to output a warning message in case it is set to 0 :) 

[image]

bbernhard, 2022-04-30:
Upon closer inspection of the gcode, I still found some issues (marked in red):

[image]

My millproject file now looks like this:

```
back=attiny85-breakout-F_Cu.gbr
#outline=attiny85-breakout-Edge_Cuts.gbr
drill=attiny85-breakout-PTH.drl
metric=true
metricoutput=true
software=custom
nog64=true

# milling
mill-feed=2500
mill-speed=0
offset=0.025
extra-passes=2
zsafe=2.0
mill-vertfeed=400
zwork=-0.08

# drilling
zdrill=-1.8
zchange=2.0
drill-feed=300.0
drill-speed=10000
nog81=true

# outline cutting
cutter-diameter=3.175
cut-feed=60
cut-infeed=0.6
cut-vertfeed=120
cut-speed=3600
zcut=-1.8
```

eyal0, 2022-04-30:
The lines in those red circles are optimizations. Pcb2gcode figured out that it would be faster to mill from one to the other than it would be to lift, move, and plunge.

The optimization section of the wiki covers it. You can set path finding to zero if you prefer. Or modify your preference on milling avoided vs time lost, which is infinity by default.

bbernhard, 2022-05-03:
Completely missed that in the wiki. Many thanks for your help - everything works fine now!

Also thanks a lot for developing & maintaining this great piece of software - very much appreciated!

closed 2022-05-03 by bbernhard