pcb2gcode/pcb2gcode #626: Install hanging on libgerbv version check

the-w-e-e-d-i-a-n, 2022-06-09:
Using package manager in Manjaro. Attempting install of pcb2gcode, this occurs:

,......
checking pkg-config is at least version 0.9.0... yes
checking for glibmm... yes
checking for gdkmm... yes
checking for gerbv... no
configure: error: Package requirements (libgerbv >= 2.1.0) were not met:

Package dependency requirement 'libgerbv >= 2.1.0' could not be satisfied.
Package 'libgerbv' has version '', required version is '>= 2.1.0'

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables gerbv_CFLAGS
and gerbv_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
==> ERROR: A failure occurred in build().
    Aborting...
    
    
GerbV has been implicitly installed on my machine, version 2.8.x
I have no idea why this is occurring, given the GerbV package version.

Please advise.
Thank you.

eyal0, 2022-06-13:
The search for gerbv is done by using `pkg-config`, which is a pretty standard install tool.  You can do the test yourself by typing:

`pkg-config --cflags "libgerbv >= 2.1.0"`

Could you type that in on your computer and see that it works?

Some other ones are tested the same way, for example:

`pkg-config --cflags "gdkmm-2.4 >= 2.8"`

Usually the location of the gerbv library will be near the location of the gerbv executable, which you can find by typing this:

`which gerbv`

Please provide the outputs of those commands and let's see if we can figure it out.  Thanks!  Worst case, I can teach you how to install gerbv in a way that will surely work.

arnief3, 2022-07-17:
I am running into the same issue.

pkg-config --cflags "libgerbv >= 2.1.0"
returns: "Requested 'libgerbv >= 2.1.0' but version of libgerbv is"

which gerbv
returns: /usr/bin/gerbv

What is the recommended procedure to solve this? I am running Pop!_OS 22.04 LTS.

Thank you for maintaining / upgrading this amazing tool!

arnief3, 2022-07-17:
I found /usr/lib/x86_64-linux-gnu/pkgconfig/libgerbv.pc and added
export PKG_CONFIG_PATH=/usr/lib/x86_64-linux-gnu/pkgconfig
but am still getting the same error.

libgerbv.pc contains:
prefix=/usr
exec_prefix=${prefix}
libdir=${prefix}/lib/x86_64-linux-gnu
includedir=${prefix}/include
pkgincludedir=${prefix}/include/gerbv-

Name: libgerbv
Description: Core library for gerbv
Requires: glib-2.0 gtk+-2.0
Version: 
Libs: -L${libdir} -lgerbv
Cflags: -I${pkgincludedir}

I notice that Version is blank. Is that the issue? I am a bit out of my depth here.

eyal0, 2022-07-21:
Let's see if gerbv can be found on your computer.  Try this command:

```bash
pkg-config --debug --exists --print-errors libgerbv |& less
```

It will output a lot.  Search that output for `gerbv`.  You might have to read through that output and look for errors or warnings.

arnief3, 2022-07-22:
I setup pcb2gcode on another machine and it works perfectly.

I will continue to investigate this issue as I have time. Thank you for taking the time to help me investigate.

eyal0, 2022-07-22:
In the CI process we build gerbv from sources and install it locally. You can read the GitHub workflows CI file and see the steps to do that. Then you'll have your own working version of gerbv independent from whatever your OS installed. That should work.

i-hayes, 2022-07-27:
Just edit the file /usr/lib/pkgconfig/libgerbv.pc.

prefix=/usr
exec_prefix=${prefix}
libdir=${exec_prefix}/lib
includedir=${prefix}/include
pkgincludedir=${prefix}/include/gerbv-

Name: libgerbv
Description: Core library for gerbv
Requires: glib-2.0 gtk+-2.0
Version:                                   <-- add to the version number installed on your system here
Libs: -L${libdir} -lgerbv
Cflags: -I${pkgincludedir}

This worked for me and allowed me to continue.

d-digiacomo, 2022-08-24:
On a fresh install of Lubuntu 22.04.1, I followed the install from git and common build steps, however ./configure doesn't pass. /usr/lib/pkgconfig/libgerbv.pc doesn't exist (pkgconfig directory empty). "... 2.1.0' but version of libgerbv is <blank>". Please advise.

eyal0, 2022-08-24:
What if you type this:

```bash
find . -name libgerbv.pc
```

Can you find the file?  I have just installed gerbv on a fresh build of MX Linux 21 and installing gerbv worked to solve the problem.

d-digiacomo, 2022-08-25:
Thanks for the reply. After running "find", I was able to locate the file in /usr/lib/x86_64-linux-gnu/pkgconfig. As with arnief3 above, running "which gerbv" (while in /usr/bin) also returned "/usr/bin/gerbv". Version number doesn't appear in About Gerbv when launching it, however compile date was 24 Dec 2021, so I presume it's v2.8.2. After revising libgerbv.pc with that version number, ./configure now passes.

eyal0, 2022-08-25:
I'm curious, what version did it say before the change and then what version did you change it too?

eyal0, 2022-08-25:
https://github.com/gerbv/gerbv/blob/213d066f26f78d446ce1d5cd0f3ed8b46f681073/src/libgerbv.pc.in

This file broke. When it generated text it didn't find the version. I will try to retrieve this problem and see when it got introduced. If I'm able.

ooxi, 2022-08-25:
@eyal0 my Makefilefoo is not that great, but my guess is on a binary missing inside the container doing the build generation.

What script handling those replacements?

```
prefix=@prefix@
exec_prefix=@exec_prefix@
libdir=@libdir@
includedir=@includedir@
pkgincludedir=@includedir@/@PACKAGE@-@VERSION@

Name: libgerbv
Description: Core library for gerbv
Requires: glib-2.0 gtk+-2.0
Version: @VERSION@
Libs: -L${libdir} -lgerbv
Cflags: -I${pkgincludedir}
```

eyal0, 2022-08-25:
No idea. I'll have to investigate.

Sounds like users are seeing this when installing the package with apt-get. And it's on version2.8.2, which is since after thia repo was created. It's not the old code.

So my guess is that someone checkouts out this repo, compiles it, and then packages it into a debian package (or rpm or arch, I dunno) and then it gets published somewhere. And one of those steps made a file that doesn't have a version in it.

@d-digiacomo @the-w-e-e-d-i-a-n are you installing with apt-get? Do you know from where it installed? I think that when you install a package, it'll print out where it grabbed package like a URL. Maybe we follow that URL and it leads us to someone who was responsible for packaging gerbv into an Ubuntu/Fedora/Redhat/Debian/whatever package.

In the meanwhile, I will build gerbv on my computer and see if that version field gets filled out correctly.

d-digiacomo, 2022-08-26:
Yes, as described in README.md. [https://packages.ubuntu.com/jammy/gerbv](https://packages.ubuntu.com/jammy/gerbv)

eyal0, 2022-08-30:
Bug report: https://bugs.launchpad.net/ubuntu/+source/gerbv/+bug/1988117

dizcza, 2023-01-29:
Still have this issue.

```
$ pkg-config --cflags "libgerbv >= 2.1.0"
Requested 'libgerbv >= 2.1.0' but version of libgerbv is
$ pkg-config --cflags "gdkmm-2.4 >= 2.8"
-pthread -I/usr/include/gdkmm-2.4 -I/usr/lib/x86_64-linux-gnu/gdkmm-2.4/include -I/usr/include/giomm-2.4 -I/usr/lib/x86_64-linux-gnu/giomm-2.4/include -I/usr/include/pangomm-1.4 -I/usr/lib/x86_64-linux-gnu/pangomm-1.4/include -I/usr/include/glibmm-2.4 -I/usr/lib/x86_64-linux-gnu/glibmm-2.4/include -I/usr/include/cairomm-1.0 -I/usr/lib/x86_64-linux-gnu/cairomm-1.0/include -I/usr/include/sigc++-2.0 -I/usr/lib/x86_64-linux-gnu/sigc++-2.0/include -I/usr/include/gtk-2.0 -I/usr/lib/x86_64-linux-gnu/gtk-2.0/include -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/x86_64-linux-gnu -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/fribidi -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/harfbuzz -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16
```

Any fix for this? Like downgrading the libgerbv...

eyal0, 2023-01-30:
@dizcza This is the workaround: https://github.com/pcb2gcode/pcb2gcode/issues/626#issuecomment-1196256414

eyal0, 2024-11-08:
It's probably installed and just that one file Failed.

You might be unable to edit it if it was installed as root.  Use `sudo` to edit it.

rrffggee, 2024-11-09:
Hello
This file was empty (size = 0 octets)
Thanks, I could edit it with sudo nano.
Tried to insert version number 2.1.0  --> configure failed
After installing another gerbv in another folder
found another version number : 2.8.2-1
Tried to insert version number 2.1.0  --> configure failed again.
How can I get this version number ?
Or how can I install gerbv with libgerbv in a reliable way ?
I am lost, thanks for your help ...

rrffggee, 2024-11-09:
My system is 64bits , Ubuntu 22.4

jonathan-kuhn, 2025-02-15:
> Just edit the file /usr/lib/pkgconfig/libgerbv.pc.
>
> prefix=/usr exec_prefix=${prefix} libdir=${exec_prefix}/lib includedir=${prefix}/include pkgincludedir=${prefix}/include/gerbv-
>
> Name: libgerbv [...]

I tried this, but it didn't work. The file didn't exist and creating it also didn't change anything. It seems to be that the file is stored in '/usr/lib/x86_64-linux-gnu/pkgconfig'. At least it was for me, and editing that helped. Hope this can help someone else too.

michaeljtbrooks, 2025-05-04:
I ran into this and ended up spending a lot of time hacking around before finding this thread and editing libgerbv.pc.

Rather than passing the buck to whoever put the libgerv into Ubuntu's repos without a version number, could the configure script be patched so that it actually accepts a blank version number as valid? Not ideal, but better than expecting users to find this thread and go hacking around entering version numbers into obscure places.

Alternatively, can you offer an unofficial PPA? Or ideally get the Ubuntu official PPA packages upgraded?

eyal0, 2025-05-05:
I guess that I could just delete this line: https://github.com/pcb2gcode/pcb2gcode/blob/master/configure.ac#L57

Then it wouldn't check for gerbv, it would just try to use it and potentially fail if gerbv is not found.  Thoughts?