pcb2gcode/pcb2gcode #627: Path cut line to pcb board

cacheflood, 2022-08-25:
Line segment cut through existing shapes and cut paths.

Dunno should that move be rapid or is the solver somehow broken?
[image]

version:
2.4.0
Git commit: latest-22-ge51ef9e5
Boost: 107600
Gerbv: 2.9.2
Geos: Not installed

Files to reproduce(?):

[tilulilu.zip](https://github.com/pcb2gcode/pcb2gcode/files/9428513/tilulilu.zip)

eyal0, 2022-08-29:
The problem seems to be because you don't have geos installed.  The boost polygon library, though I really like using it, has some flaws.  :-(  And it's not so fast, either.  One of the flaws is that is causes errors in the output.  Usually they are minor but in your case it's a big one!

Here's the version without geos:

```
2.4.0
Git commit: v2.4.0-66-gd9f5a6f6
Boost: 106600
Gerbv: 2.7.0
Geos: Not installed
```

[image]

To install geos++ and recompile, I issued these commands:

```bash
sudo apt-get install libgeos++-dev
autoreconf -fvi && ./configure BOOST_ROOT${HOME}/private/boost/boost1.66 pcb2gcode_CPPFLAGS_EXTRA="-Wall -Werror" --disable-silent-rules --enable-static --disable-shared && make clean && make -j 20
```

Now the version is:

```
2.4.0
Git commit: v2.4.0-66-gd9f5a6f6
Boost: 106600
Gerbv: 2.7.0
Geos: 3.9.0
```

And now my output is:

[image]

If this doesn't solve your problem, please re-open the issue.  Thanks!

closed 2022-08-29 by eyal0

cacheflood, 2022-08-29:
Thank you for the information, you guys are doing a great job  ! :)

-Jani