pcb2gcode/pcb2gcode #630: linker error when building pcb2gcode:  undefined reference to `geos::geom::GeometryFactory::getCoordinateSequenceFactory() const

codehero, 2022-09-01:
## Description of bug:

## What did you try (include command-line arguments):
1. First step
Installed all dependencies on Linux Mint Vanessa, including geos 3.8.0

2. Second step

./configure
make -j4

## What happened:

  CXXLD    pcb2gcode
/usr/bin/ld: geos_helpers.o: in function `to_geos(boost::geometry::model::linestring<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, std::vector, std::allocator> const&)':
/home/dbender/extsrc/pcb2gcode/geos_helpers.cpp:89: undefined reference to `geos::geom::GeometryFactory::getCoordinateSequenceFactory() const'
/usr/bin/ld: /home/dbender/extsrc/pcb2gcode/geos_helpers.cpp:92: undefined reference to `geos::geom::Coordinate::Coordinate(double, double, double)'
/usr/bin/ld: geos_helpers.o: in function `to_geos(boost::geometry::model::ring<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, true, true, std::vector, std::allocator> const&)':
/home/dbender/extsrc/pcb2gcode/geos_helpers.cpp:99: undefined reference to `geos::geom::GeometryFactory::getCoordinateSequenceFactory() const'
/usr/bin/ld: /home/dbender/extsrc/pcb2gcode/geos_helpers.cpp:103: undefined reference to `geos::geom::Coordinate::Coordinate(double, double, double)'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1091: pcb2gcode] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/dbender/extsrc/pcb2gcode'
make[1]: *** [Makefile:1348: all-recursive] Error 1
make[1]: Leaving directory '/home/dbender/extsrc/pcb2gcode'
make: *** [Makefile:934: all] Error 2

## What did you expect to happen:

Successful link

## pcb2gcode version (run pcb2gcode --version to see this):

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**  You might need to zip them.  If you don't provide a millproject file, make sure that you copy-paste the command-line that you used.

eyal0, 2022-09-01:
I'm usually using geos 3.11 but I will try it with 3.8 and see what happens.

In the meanwhile, could you try to run this so that we can get more output?

```bash
autoreconf -fvi && ./configure --disable-silent-rules && make clean && make -j 4
```

eyal0, 2022-09-01:
Something strange is going on, it shouldn't be possible to use pcb2gcode with geos with geos version 3.8.  The minimum version is 3.8.1.  https://github.com/pcb2gcode/pcb2gcode/blob/master/configure.ac#L71

Are you sure that your version number for geos is precisely 3.8.0?

codehero, 2022-09-01:
Apologies, as per ./configure, the version number DETECTED is not 3.8.0 as I had incorrectly stated...

checking for geos-config... /usr/bin/geos-config
checking for GEOS version >= 3.8.1... yes
configure: Found geos, is it geos++?
configure: Found geos++, we'll use it instead of some of the boost geometry for greater accuracy and speed.

However, to add to the confusion, the version installed by apt-get is actually 3.10.2

dbender@qesemet:~$ sudo apt-get install libgeos++-dev
[sudo] password for dbender:
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libgeos-c1v5 libgeos-dev libgeos3.10.2
Suggested packages:
  libgdal-doc
The following NEW packages will be installed:
  libgeos++-dev libgeos-c1v5 libgeos-dev libgeos3.10.2
0 upgraded, 4 newly installed, 0 to remove and 203 not upgraded.
Need to get 1,216 kB of archives.
After this operation, 5,946 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libgeos3.10.2 amd64 3.10.2-1 [713 kB]
Get:2 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libgeos-c1v5 amd64 3.10.2-1 [82.5 kB]
Get:3 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libgeos-dev amd64 3.10.2-1 [46.6 kB]
Get:4 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libgeos++-dev amd64 3.10.2-1 [374 kB]
Fetched 1,216 kB in 1s (1,336 kB/s)
Selecting previously unselected package libgeos3.10.2:amd64.
(Reading database ... 333246 files and directories currently installed.)
Preparing to unpack .../libgeos3.10.2_3.10.2-1_amd64.deb ...
Unpacking libgeos3.10.2:amd64 (3.10.2-1) ...
Selecting previously unselected package libgeos-c1v5:amd64.
Preparing to unpack .../libgeos-c1v5_3.10.2-1_amd64.deb ...
Unpacking libgeos-c1v5:amd64 (3.10.2-1) ...
Selecting previously unselected package libgeos-dev.
Preparing to unpack .../libgeos-dev_3.10.2-1_amd64.deb ...
Unpacking libgeos-dev (3.10.2-1) ...
Selecting previously unselected package libgeos++-dev.
Preparing to unpack .../libgeos++-dev_3.10.2-1_amd64.deb ...
Unpacking libgeos++-dev (3.10.2-1) ...
Setting up libgeos3.10.2:amd64 (3.10.2-1) ...
Setting up libgeos-c1v5:amd64 (3.10.2-1) ...
Setting up libgeos-dev (3.10.2-1) ...
Setting up libgeos++-dev (3.10.2-1) ...
Processing triggers for man-db (2.10.2-1) ...
Processing triggers for libc-bin (2.35-0ubuntu3) ...
dbender@qesemet:~$ cd extsrc/
dbender@qesemet:~/extsrc$ ls

I will try the step you suggested tomorrow morning, thanks!

codehero, 2022-09-02:
Configure output:

dbender@qesemet:~/extsrc/pcb2gcode$ autoreconf -fvi && ./configure --disable-silent-rules && make clean && make -j 4
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'm4'.
libtoolize: copying file 'm4/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: configure.ac: not using Intltool
autoreconf: configure.ac: not using Gtkdoc
autoreconf: running: aclocal --force -I m4
autoreconf: running: /usr/bin/autoconf --force
configure.ac:21: warning: The macro `AC_PROG_LIBTOOL' is obsolete.
configure.ac:21: You should run autoupdate.
m4/libtool.m4:99: AC_PROG_LIBTOOL is expanded from...
configure.ac:21: the top level
configure.ac:77: warning: The macro `AC_LANG_CPLUSPLUS' is obsolete.
configure.ac:77: You should run autoupdate.
./lib/autoconf/c.m4:262: AC_LANG_CPLUSPLUS is expanded from...
configure.ac:77: the top level
configure.ac:111: warning: The macro `AC_TRY_COMPILE' is obsolete.
configure.ac:111: You should run autoupdate.
./lib/autoconf/general.m4:2847: AC_TRY_COMPILE is expanded from...
lib/m4sugar/m4sh.m4:692: _AS_IF_ELSE is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
./lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
./lib/autoconf/general.m4:2270: AC_CACHE_CHECK is expanded from...
m4/ax_c_compile_value.m4:91: AX_C_COMPILE_VALUE is expanded from...
configure.ac:111: the top level
configure.ac:120: warning: The macro `AC_HEADER_STDC' is obsolete.
configure.ac:120: You should run autoupdate.
./lib/autoconf/headers.m4:704: AC_HEADER_STDC is expanded from...
configure.ac:120: the top level
configure.ac:129: warning: AC_OUTPUT should be used without arguments.
configure.ac:129: You should run autoupdate.
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:20: installing 'm4/compile'
configure.ac:16: installing 'm4/missing'
Makefile.am: installing 'm4/depcomp'
autoreconf: Leaving directory '.'
checking whether make supports nested variables... yes
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... none needed
checking whether make supports the include directive... yes (GNU style)
checking dependency style of g++... gcc3
checking for gcc... gcc
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
[... 103 lines cut ...]
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating man/Makefile
config.status: creating config.h
config.status: executing depfiles commands
config.status: executing libtool commands

Linker error remains the same:

/bin/bash ./libtool  --tag=CXX   --mode=link g++  -DGIT_VERSION=\"`git describe --dirty --always --tags`\" -DGERBV_VERSION=\"`pkg-config --modversion libgerbv`\" -g -O2 -Wall    -o pcb2gcode autoleveller.o backtrack.o board.o bg_helpers.o bg_operators.o common.o drill.o eulerian_paths.o geos_helpers.o gerberimporter.o layer.o merge_near_points.o ngc_exporter.o path_finding.o segment_tree.o segmentize.o surface_vectorial.o tile.o trim_paths.o options.o outline_bridges.o svg_writer.o voronoi.o main.o  -lgerbv -lgtk-x11-2.0 -lgdk-x11-2.0 -lpangocairo-1.0 -latk-1.0 -lcairo -lgdk_pixbuf-2.0 -lgio-2.0 -lpangoft2-1.0 -lpango-1.0 -lgobject-2.0 -lglib-2.0 -lharfbuzz -lfontconfig -lfreetype -lboost_program_options  -L/usr/lib/x86_64-linux-gnu -lgeos
libtool: link: g++ -DGIT_VERSION=\"latest-27-g0898bdd3\" -DGERBV_VERSION=\"2.8.2-1\" -g -O2 -Wall -o pcb2gcode autoleveller.o backtrack.o board.o bg_helpers.o bg_operators.o common.o drill.o eulerian_paths.o geos_helpers.o gerberimporter.o layer.o merge_near_points.o ngc_exporter.o path_finding.o segment_tree.o segmentize.o surface_vectorial.o tile.o trim_paths.o options.o outline_bridges.o svg_writer.o voronoi.o main.o  -lgerbv -lgtk-x11-2.0 -lgdk-x11-2.0 -lpangocairo-1.0 -latk-1.0 -lcairo -lgdk_pixbuf-2.0 -lgio-2.0 -lpangoft2-1.0 -lpango-1.0 -lgobject-2.0 -lglib-2.0 -lharfbuzz -lfontconfig -lfreetype -lboost_program_options -L/usr/lib/x86_64-linux-gnu -lgeos
/usr/bin/ld: geos_helpers.o: in function `to_geos(boost::geometry::model::linestring<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, std::vector, std::allocator> const&)':
/home/dbender/extsrc/pcb2gcode/geos_helpers.cpp:89: undefined reference to `geos::geom::GeometryFactory::getCoordinateSequenceFactory() const'
[... cut ...]
/usr/bin/ld: /home/dbender/extsrc/pcb2gcode/geos_helpers.cpp:103: undefined reference to `geos::geom::Coordinate::Coordinate(double, double, double)'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1091: pcb2gcode] Error 1
make[2]: *** Waiting for unfinished jobs....
mv -f .deps/wkt_to_svg.Tpo .deps/wkt_to_svg.Po
make[2]: Leaving directory '/home/dbender/extsrc/pcb2gcode'
make[1]: *** [Makefile:1348: all-recursive] Error 1
make[1]: Leaving directory '/home/dbender/extsrc/pcb2gcode'

eyal0, 2022-09-10:
So weird.  It seems that you have geos but it can't find it?

Perhaps send also `config.log`, which will show how it tested for geos.

eyal0, 2022-09-10:
You could also install geos yourself locally.  It's not that hard to do and it will solve your problem.  That's how I do it so that I can best control my version of geos.

```bash
          pushd ~
          git clone --depth 1 --branch main https://github.com/libgeos/geos.git
          pushd geos
          mkdir build
          pushd build
          cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Release .. -DCMAKE_INSTALL_PREFIX=$HOME/.local
          make -j ${NUM_CPUS}
          make install
          popd
          popd
          popd
```

That will install it into `~/.local`.  Then when you run `configure`, you add this flag:

```bash
./configure --with-geos=$HOME/.local
```

Again, this shouldn't be necessary but it's another option for how to solve your issue.

Dromantor, 2022-09-21:
I investigated this further because I wanted to build in WSL with Ubuntu 22.x with the geos you get when using apt install, which is version 3.10.2. Naturally I faced the same issue, but I'm pretty sure now the issue is with geos and not a configuration issue.

Findings:

- the symbol `geos::geom::GeometryFactory::getCoordinateSequenceFactory()` is present in the included header file `geos/geom/GeometryFactory.h` or else it would fail before the linking stage
- the symbol `geos::geom::GeometryFactory::getCoordinateSequenceFactory()` and some others are NOT present in the dump of libgeos.so, created with `nm -gDC libgeos.so > ~/geos.txt`, so it fails when linking

Then I cloned the geos repo, checked out the 3.10.2 tag, and checked the sources. Findings:

- the header file is of course exactly the same as in the apt version, providing the declaration of said symbol, but no implementation
```    
/// \brief
/// Returns the CoordinateSequenceFactory associated
/// with this GeometryFactory
const CoordinateSequenceFactory* getCoordinateSequenceFactory() const;
```
- the only implementation of said function is done in `geos/geom/GeometryFactory.inl`
 ```
INLINE const CoordinateSequenceFactory*
GeometryFactory::getCoordinateSequenceFactory() const
{
    return coordinateListFactory;
}
```
I usually don't do C++ and I never saw a .inl file before, so sorry if I'm wrong, but from my understanding the function body of getCoordinateSequenceFactory() is never provided to the user as a standalone function because it is neither in the .h file provided via apt, nor is it present in the .so library because all uses are inlined because it's in the .inl file and tagged with INLINE.

This would explain why linking fails for us, using the lib, but not for the creators of geos, compiling the lib, because they have said .inl file

The current version of geos, the head of their git repo, has said implementation in the header file itself, no .inl anymore, so it compiles and links correctly.
```
/// \brief
/// Returns the CoordinateSequenceFactory associated
/// with this GeometryFactory
const CoordinateSequenceFactory* getCoordinateSequenceFactory() const
{
    return coordinateListFactory;
};
```

TLDR:
-
The geos version provided by apt, 3.10.2, is broken.

eyal0, 2022-12-07:
Looks like getCoordinateSequenceFactory() was removed in the latest version of geos!  How annoying that they keep changing their API!  Oy.

This was the breaking change: https://github.com/libgeos/geos/commit/60edf0e5f040e065193bd5beb79f7fa8febb1cd5

It may be possible for me to write pcb2gcode to support both the old and new geos.

codehero, 2022-12-08:
Let me know if you need me to test a branch pull

eyal0, 2022-12-08:
Should be ready, want to try it now?

codehero, 2022-12-08:
Still have the same error when I use a fresh checkout from master

From configure:

checking for geos-config... /usr/bin/geos-config
checking for GEOS version >= 3.8.1... yes
configure: Found geos, is it geos++?
configure: Found geos++, we'll use it instead of some of the boost geometry for greater accuracy and speed.

From make:

/usr/bin/ld: geos_helpers.o: in function `to_geos(boost::geometry::model::linestring<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, std::vector, std::allocator> const&)':
/home/dbender/extsrc/pcb2gcode/geos_helpers.cpp:100: undefined reference to `geos::geom::Coordinate::Coordinate(double, double, double)'
/usr/bin/ld: geos_helpers.o: in function `to_geos(boost::geometry::model::ring<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, true, true, std::vector, std::allocator> const&)':
/home/dbender/extsrc/pcb2gcode/geos_helpers.cpp:110: undefined reference to `geos::geom::Coordinate::Coordinate(double, double, double)'
collect2: error: ld returned 1 exit status

sudo apt-get install libgeos++-dev     
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libgeos++-dev is already the newest version (3.10.2-1).

eyal0, 2022-12-08:
Sounds like it's compiling fine but failing to link.  Which means that the header file is there but the library is the wrong version?

Maybe run find and see if you have too many Geos libraries installed?

`find / -name 'libgeos*' -ls`

We expect just one libgeos to be found.

codehero, 2022-12-13:
My /usr/include/geos/version.h indicates 3.10.2, and so does my /usr/lib. I only have one instance of libgeos installed.

eyal0, 2022-12-14:
It's a problem with the linker and not the compiler.  I'm pretty sure that Ubuntu built this image incorrectly.

https://github.com/libgeos/geos/issues/645

https://bugs.launchpad.net/ubuntu/+source/geos/+bug/1980147

Not much for me to do here.  Ubuntu package maintainer made a bad build.

closed 2022-12-14 by eyal0