pcb2gcode/pcb2gcode #631: Holes above min-milldrill-hole-diameter are still drilled

imihajlow, 2022-09-10:
When using a set of available drills and a min-milldrill-hole-diameter, drills above the min-milldrill-hole-diameter are rounded to the best matching drill from the set and drilled alongside with milldrilling. This leads to redundant drilling.

Version:
```
2.4.0
Git commit:
Boost: 107900
Gerbv: 2.9.2
Geos: Not installed
```

[bug.zip](https://github.com/pcb2gcode/pcb2gcode/files/9539764/bug.zip)

eyal0, 2022-09-10:
Seems like a bug!  I'll take a quick look right now.

eyal0, 2022-09-10:
Okay, so I understand what is happening better now.  First, `pcb2gcode` is gathering the list of holes that you want to drill.  Then, it needs to figure out how to milldrill each hole.  It will only milldrill the holes that are bigger than the minimum milldrilling size.

Then it does the drilling phase.  It goes through the holes looking for holes that need to be drilled.  For each one, it picks the most appropriate drill.  Then it adjusts the size of each one to match the drill that will be used.  And then it drills them.

For the milldrilling, it's working fine.

For the drilling, it has a problem.  You list of drills is like this:

```
0.0004 m
0.0005 m
0.0006 m
0.0007 m
0.0008 m
0.0009 m
0.001 m
0.0011 m
0.0012 m
0.0017 m:-0.0001 m:+0.0001 m
```

The "m" is for meters.  A little math will show you that this list matches the list that you put in your `millproject` file.  All those drills except the last one have no tolerance, which means to use the best match.  On the last one, it will only use it if it's within 0.1mm of 1.7mm.

So the codes is taking the big holes in your project and saying that the 1.2mm is the best match for those big holes because the 1.7mm is only for holes 1.6-1.8mm.  So it rounded the size down to 1.2mm.  And since that size is less than the the 2.0mm minimum milldrilling size, it needs to be drilled!

This logic is totally nuts!  I'll fix it.

eyal0, 2022-09-10:
BTW, that explains these messages: 

```
Info: bit 3 (0.85mm) is rounded to 0.8mm
Info: bit 6 (2.5mm) is rounded to 1.2mm
Info: bit 7 (3.2mm) is rounded to 1.2mm
```

imihajlow, 2022-09-10:
Hi @eyal0,
thanks for your quick reply. I set the tolerance for the 1.7mm drill trying to fix this behavior. Without the tolerance pcb2gcode selected the 1.7mm bit for larger holes. I hoped that with the tolerance it won't select it because it won't match, but it simply picked the next one from the list.

PR #632 (2022-09-10): Don't drill holes that are going to be milldrilled.
This fixes #631
commit 7eaebcab (2022-09-10): fix: Don't round bits that will be milldrilled.

This fixes #631
closed 2022-09-10 by eyal0