pcb2gcode/pcb2gcode #636: Some vias get cut in half despite build with geos

JoniSt, 2022-09-19:
Hi,

I've hit an optimization(?) bug in pcb2gcode that causes it to cut through previously milled traces.
[Gerbers, millproject, and pcb2gcode output](https://github.com/pcb2gcode/pcb2gcode/files/9602418/gerber.zip)

I've read some old issues and tried the suggestions I found (and other stuff), but none of it worked:
- Build with and without geos (I had it built with geos originally)
- `backtrack=0`
- `optimise=0`
- `path-finding-limit=0`
- `tsp-2opt=0`

It appears that pcb2gcode believes that a G0 fast move between two cuts can be removed and replaced with a G1 at cutting depth, but by doing that, it runs the endmill through a couple of vias.

The result is this (`processed_back_final_0.svg`, 1mm endmill, towards the bottom right):
[image]

I would've expected it to go up to zsafe, perform a G0, and then go back down to zwork instead.

pcb2gcode version:
```
2.5.0
Git commit: 
Boost: 107400
Gerbv: 2.9.2-1
Geos: 3.11.0
```

Thank you very much for your support and for making this awesome tool! :smiley:

JoniSt, 2022-09-20:
My head hurts, but I think I found it...

This is a spike. The green and purple traces intersect the convex hull of the light-blue trace's polygons. `add_spikes` generates a very long spike that runs entirely within that convex hull, thereby intersecting the other two traces. It's not stopped by `spikes_keep_out` because the light-blue trace's polygons are concave and thereby "evade" the spike until it's too late.

So `spikes_keep_out` alone does not work for concave polygons. Maybe spikes should simply get checked against other traces?

eyal0, 2022-09-20:
I believe that spikes used to get checked against other traces but it didn't work as well.  I haven't looked at that bit of the code in a while.  I can take another look.

eyal0, 2022-09-20:
I have a plan for debugging it, I haven't forgotten you!  You're help in determining that it was the spikes was very useful!

In case you didn't know, spikes are for sharp corners.  There is a picture of a spike on this page: https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Milling#--isolation-width-prefer-this-instead-of-the-old---extra-passes

Right in the middle here:

[image]

See that little bit where it juts out?  If that didn't exist, the milling would miss a little patch in there.

You sound like a coder so I'll explain: The spikes are done on line 842 of surface_vectorial.cpp.  Only milling passes *after* the first one gets spikes because the first path by design is as close as possible to the trace.  A spike would dig into a trace!  So the offset there is 0 and 0 disables the spikes.

The length of a spike depends on how acute the angle is and it looks like yours is very acute!  It *should* have been trimmed to size by the "keep_out", which is basically taking all the little spikes and subtracting the area around the previous milling.  Somehow, one got through!

I have tools to print out areas and view them as SVG files so I will do that when I return to the compute.  Basically, I do something like `std::cout << bg::wkt(shape) << std::endl` and then I can use the wkt_to_svg binary tool that is part of pcb2gcode to view it.

This is also a tool that I use for comparing the svg outputs before and after my changes: https://github.com/eyal0/image-diff-gui

Anyway, if you want to dig in further, you may.  If you want to let me do it, no problem.  I'll put updates here as I figure it out.

JoniSt, 2022-09-20:
Thanks a lot! Yeah, tight corners wouldn't work out too well without the spikes!

I've dug into the code a bit already; as far as I can tell, the spike keep-out was working as intended but it's unfortunately not quite sufficient. Here's an awful illustration:

[image]

Orange is the trace being processed, red is the spike keep-out, blue is (part of) the ring currently being processed, purple is the "victim trace", green is the spike being generated (the dotted outline indicates what area it'll mill, similar to the SVG visualizations).

The spike gets generated at an odd angle because the geometry just happens to work out in a wonky way (because of the fork in the trace, most likely). This in itself shouldn't be too much of a problem. The red spike keep-out works as intended too, it stops the spike from going into the orange trace. However, the code only considers the last polygon ring of the current trace (orange) when generating the spike keep-out (red), it does not take into account the surrounding traces (in this case purple). If the spike angle works out just right, the spike will run almost parallel to the trace for a really long distance until it finally runs into the spike keep-out. And if there's another trace nearby, the spike might go straight through it - which results in purple getting hurt in this case.

I think the correct fix would be to consider the surrounding traces *in addition to* the spike keep-out that's already in place. That way the spike stops either when it hits the previous polygon ring (spike keep-out), or another surrounding trace.

So TL;DR the spike keep-out did in fact trim the spike to size, but that was still too much in this pathological case!

eyal0, 2022-09-21:
You are correct in all of your assessment.  I'll see what I can do.

eyal0, 2022-09-21:
The problem is that we do one trace at a time.  It's good for the final path when we join them all together.  But this means that the keepout for the trace you are trying to protect is not even ready by the time that we are calculating spikes.

So I could put the spike creation later but I feel like it'll be such a big change in the code that I will suffer greatly to do it.  What I *can* try is make a keep in for the spikes based on the voronoi region of the trace.  Generally I only expected spikes to point at the trace, not away.  Like in the image above.  And I also have the voronoi region calculated for all polygons by this point in the code.  So I'll give that a shot, we'll take a diff, and see what happens, yeah?

JoniSt, 2022-09-21:
Masking the spikes against the trace's voronoi sounds good to me! :+1: 

I'll also have to get some sleep now, so I'll come back to you tomorrow. Thank you very much for tackling this! :smile:

eyal0, 2022-09-21:
Here's the diff in image-diff-gui (also I get to show off my program image-diff-gui):

[image]

We can see that the diff is pretty much just that one spot.  I'll zoom in.

[image]

Looks perfect!  That spike stops at the right spot for me.

Another thing that I noticed is that not a single one of my regression tests was affected.  Oh no!  Would you be okay with me using your gerber files and millproject inside the code base as a test in the `testing/gerbv_examples` directory?  I will try to trim away as many things in the design as possible so that the test is smaller and you can keep your design more secret.  Is this okay with you?

JoniSt, 2022-09-21:
I just gave it a quick test too, and yeah, it works absolutely flawlessly. The spike perfectly stops at the voronoi boundary. Nice work! :+1: 

You can of course include my gerbers in your regression tests (under whatever open license you like) - it's just a part of a small university project and not secret at all.

Thank you so much for fixing this so quickly!

eyal0, 2022-09-21:
You bet, I'm glad that it was easy to fix!  Your hunting down of the spikes saved me a bunch of time.

I tried to remove a few traces from your design to see if the bug still appears and it does not!  Even if I delete traces on the far side of the design!  The code is complicated and everything interacts with everything so it might be that when I resave the gerber file after removing some elements, they get resorted or rounded differently and the effect doesn't happen.

So I guess that I'm just adding your design as-is to the code.  Adding the test helps ensure that the bug doesn't come back.  Thanks for your contribution and good luck in your studies!

JoniSt, 2022-09-21:
Hah, typical Heisenbug! Funny that I even hit it in the first place.

It's totally fine with me if you just add the entire gerber file as it is. :+1:

PR #637 (2022-09-22): fix: Use spikes_keep_in for regular spikes, too.
The spike math might lead a spike to point away from the trace instead of toward it.  This is unlikely but when it happens, it might aim at a neighboring trace!  And we don't yet have the neighboring trace's keep out so we can't use it.

Rather than defer the computation of spikes, which would be difficult, we will just keep the spikes within the voronoi region of the trace.

This only applies to regular traces, not to thermal reliefs nor voronoi milling.

This fixes #636
commit 2453cbd3 (2022-09-22): fix: Use spikes_keep_in for regular spikes, too.

The spike math might lead a spike to point away from the trace instead
of toward it.  This is unlikely but when it happens, it might aim at a
neighboring trace!  And we don't yet have the neighboring trace's keep
out so we can't use it.

Rather than defer the computation of spikes, which would be difficult,
we will just keep the spikes within the voronoi region of the trace.

This only applies to regular traces, not to thermal reliefs nor
voronoi milling.

This fixes #636
closed 2022-09-22 by eyal0