pcb2gcode/pcb2gcode #646: Spindle Speeds Are Integers Error?

rebeltaz, 2022-11-26:
I installed pcb2cgode and pcb2gcodeGUI. I passed pcb2gcode a gbr file from kicad and it is telling me:

```
Error while processing input files (error code 101)

pcb2gcode output:
Error: You've supplied an invalid parameter.
Note that spindle speeds are integers!
Details: unrecognised option '--output-dir=/tmp/pcb2gcode-5935'
...

I can't find anything about this. My OS is Ubuntu 18.04. pcb2gcode is the version in the repository - 1.1.4. pcb2gcodeGUI is 1.3.2-1. kicad is 6.0.4

Any help or point me in the right direction would be greatly appreciated. Thanks.

eyal0, 2022-11-26:
I don't support pcb2gcodeGUI and you version of pcb2gcode is very old. Are you able to use a newer one without GUI?

rebeltaz, 2022-11-26:
Yeah... I knew the GUI wasn't yours, but it looked like the errors were being passed by pcb2gcode itself. I looked at the github code, but it looked like only 20.0 and up was supported? I can try to compile it and see what happen.