pcb2gcode/pcb2gcode #650: I get error when I try to build with "geos"

ivanbraga, 2022-12-14:
/pcb2gcode$ ./configure
checking whether make supports nested variables... yes
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... none needed
checking whether make supports the include directive... yes (GNU style)
checking dependency style of g++... gcc3
checking for gcc... gcc
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... /usr/bin/ld -m elf_x86_64
checking if the linker (/usr/bin/ld -m elf_x86_64) is GNU ld... yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking for g++ option to produce PIC... -fPIC -DPIC
checking if g++ PIC flag -fPIC -DPIC works... yes
checking if g++ static flag -static works... yes
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether g++ supports C++14 features by default... yes
checking whether C compiler accepts -fext-numeric-literals... yes
checking CXXFLAGS for maximum warnings... -Wall
checking whether to build with code coverage support... no
checking for valgrind... valgrind
checking for Valgrind tool memcheck... yes
checking for Boost headers version >= 1.60.0... yes
checking for Boost's header version... 1_74
checking for the toolset name used by Boost for g++... configure: WARNING: could not figure out which toolset name to use for g++

checking for boost/program_options.hpp... yes
checking for the Boost program_options library... yes
checking for boost/geometry.hpp... yes
checking for boost/algorithm/string.hpp... yes
checking for boost/format.hpp... yes
checking for boost/optional.hpp... yes
checking for boost/variant/variant_fwd.hpp... yes
checking for boost/variant.hpp... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for glibmm-2.4 >= 2.8... yes
checking for gdkmm-2.4 >= 2.8... yes
checking for libgerbv >= 2.1.0... yes
checking for librsvg-2.0 >= 2.0... yes
checking for librsvg-2.0 >= 2.46... yes
checking for librsvg-2.0 >= 2.52... yes
checking for geos-config... /usr/bin/geos-config
checking for GEOS version >= 3.8.1... yes
configure: Found geos, is it geos++?
configure: Found geos++, we'll use it instead of some of the boost geometry for greater accuracy and speed.
checking value of GERBV_INTERPOLATION_LINEARx10 by analyzing object code... 1
checking for egrep... (cached) /usr/bin/grep -E
checking for _Bool... no
checking for stdbool.h that conforms to C99... yes
checking for inline... inline
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating man/Makefile
config.status: creating config.h
config.status: executing depfiles commands
config.status: executing libtool commands

/usr/bin/ld: geos_helpers.o: in function `to_geos(boost::geometry::model::linestring<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, std::vector, std::allocator> const&)':
/pcb2gcode/geos_helpers.cpp:100: undefined reference to `geos::geom::Coordinate::Coordinate(double, double, double)'
/usr/bin/ld: geos_helpers.o: in function `to_geos(boost::geometry::model::ring<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, true, true, std::vector, std::allocator> const&)':
/pcb2gcode/geos_helpers.cpp:110: undefined reference to `geos::geom::Coordinate::Coordinate(double, double, double)'
collect2: error: ld returned 1 exit status

eyal0, 2022-12-14:
This is a duplicate of #630, which I will re-open.

closed 2022-12-14 by eyal0

ivanbraga, 2022-12-16:
Thanks for answering, it was to investigate the solution based on your information.