pcb2gcode/pcb2gcode #651: pcb2gcode really doesn't like optimise=0.01mm on this board

c-, 2023-01-03:
For both of these cases, commenting out the optimise=0.01mm line makes the problem go away.

[optivector_range_check_error.zip](https://github.com/pcb2gcode/pcb2gcode/files/10334237/optivector_range_check_error.zip)

This first gerber causes pcb2gcode to fail with a:

> Exporting back... vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)

error.

This second gerber, which is just the first with the "Plot Edge.Cuts on all layers" option turned off, causes pcb2gcode to go into a spin and exhaust all system memory. Which is particularly humorous when you start the job and head out to the workshop expecting it to be ready to run when you get there... ;)

[optifinity.zip](https://github.com/pcb2gcode/pcb2gcode/files/10334230/optifinity.zip)

eyal0, 2023-02-07:
Optimize tries to make your file smaller by removing some of the points of the milling, so that you don't have as big a file.

I'm sure that this was some sort of weird rounding error in some algorithm deep in the boost geometry code.  I can't do anything about it.  The default optimize is 0.00254mm and unless you're having a file size problem, I would just leave it be.

Best of luck!

closed 2023-02-07 by eyal0

c-, 2023-02-08:
> I'm sure that this was some sort of weird rounding error in some algorithm deep in the boost geometry code.

That's what I suspected. Wasn't sure if it would be straightforward to trace it down to a particular bit of geometry that would make a useful bug report for the boost folks.

It's not critical; I set and forgot that optimize level back when the network connection to my workshop was bad enough that file size actually mattered.