pcb2gcode/pcb2gcode #652: Expand isolation beyond outline to avoid burr

nerdunio, 2023-01-20:
pcb2gcode is an amazing tool! However, here's a feature request :)

[image]

Refer to the image above. I am using a 0.8mm and 0.1mm mill for isolation. pcb2gcode uses the 0.8mm cutter to isolate the outermost line. A 1mm cutter cuts the outline. The router bit forms a copper burr because it cuts through the copper. Once I turn the board over, the burr prevents the board from lying flat.

[image]

I suggest adding a --outline-isolation-expansion (or something) option so that the 0.8mm mill (in my case) cuts beyond the outline routing area to avoid the burr I mentioned.

The final result would be the following (made with FlatCAM).

[image]

I have attached a .zip file with the script I used. The version of pcb2gcode is 2.4.0.

## What did you try (include command-line arguments):
1. Go through all available options in pcb2gcode --help
[outline_expansion.zip](https://github.com/pcb2gcode/pcb2gcode/files/10466400/outline_expansion.zip)

eyal0, 2023-01-24:
Here's my first try but I worry that it still isn't enough. 
https://github.com/eyal0/pcb2gcode/archive/refs/heads/mask_outline_with_diameter.zip

I'll sleep on it.

eyal0, 2023-01-24:
I made it bigger now that I've thought about it.  Hmm.  I dunno, give it a shot, let me know how it goes, yea?

eyal0, 2023-01-31:
I suppose that for deburring what I really want is to run the mill along the edge of the outline.  It would be great if I could figure out how to do this within the existing functionality.  I could treat the outline as a trace, for example, but that might not do milling far enough away from the trace in order to deburr.  I could make the outline of the cutter's path a trace but that wouldn't do it, either.  I could also just add milling right along the cutter's edge.  I guess that I would just have a deburr flag and add in the path directly.  But there are different milling bits, how would I know which one to use?  Thickest is best but if it doesn't fit then what?

nerdunio, 2023-01-31:
I set --isolation-width to 100mm so that all areas that are not copper are cut off. pcb2gcode extends the isolation to the contour. IMHO, adding an option to extend the isolation beyond the contour by a user-definable amount is sufficient.

A --deburr flag would be very elegant. I assume that the isolation contour is extended by the value of --cutter-diameter.

If the thickest cutter doesn't fit, we take the second thickest; as pcb2gcode always does. Or do I understand this wrong?

Thanks a lot for your effort!

eyal0, 2023-01-31:
The existing code stops milling at the edge of the outline.  I modified it so that it will stop at the outer edge of the cutter's path.  That should solve your problem so long as you keep with the big `isolation-width`.

It would be even nicer if I could have it mill the edge of the cutter's path even when the `isolation-width` is small.  I just don't know how to do that yet.

eyal0, 2023-01-31:
Here's that code: https://github.com/eyal0/pcb2gcode/archive/refs/heads/mask_outline_with_diameter.zip

eyal0, 2023-01-31:
> If the thickest cutter doesn't fit, we take the second thickest; as pcb2gcode always does. Or do I understand this wrong?

This is true for milling traces but not for cutting outlines.