pcb2gcode/pcb2gcode #655: Windows build for 2.5.0?

sybrenstuvel, 2023-01-28:
Hey folks,

Could someone make a Windows build of [version 2.5.0](https://github.com/pcb2gcode/pcb2gcode/releases/tag/v2.5.0)? 

The README.md file reads:
> Windows prebuilt binaries (with all the required DLLs) are available in the [release](https://github.com/pcb2gcode/pcb2gcode/releases) page.

If Windows binaries will no longer be released, then this text should be adjusted.

eyal0, 2023-01-31:
I used to have a Windows build in the CI but it got too difficult to maintain.  Windows is a really difficult architecture.

sybrenstuvel, 2023-01-31:
Yeah, I know -- that's why I was hoping for a CI instead of having to set things up myself ;-)

eyal0, 2023-01-31:
The difficulty was that the github CI keeps changing in ways that break the Windows build.  Even the Linux one breaks sometimes but the Windows one was worse.

You could perhaps follow the directions in the Windows CI to build it yourself?

sybrenstuvel, 2023-02-11:
I'll give it a try, but I do feel that the README is a bit misleading:

> You can easily build pcb2gcode for Windows with MSYS2

Uhu, "easily". Uhu uhu, suuuure 😆

sybrenstuvel, 2023-02-11:
Here's some feedback on the current README:

> You can easily build pcb2gcode for Windows with MSYS2 (http://sourceforge.net/projects/msys2/).

I'd change the link to https://www.msys2.org as that's a much friendlier website, and its' a lot clearer where to go and what to do.

> then run "MinGW-w64 Win32 Shell"

That's called "MSYS MINGW32" (for i686) or "MSYS MINGW64" (for x86) nowadays. The README could also use a warning that it's important which shell you start, as it seems to influence which compilers etc. are available to you.

> The following commands are for the i686 binary

I think nowadays it's better to list the commands for a 64-bit binary, and leave adjustments for a 32-bit as a side-note.

> Now let's download, build and install gerbv (version 2.6.1 is broken, don't use it)

Well, 2.6.0 is also broken:
```
Making all in src
make[2]: Entering directory '/home/Sybren/workspace/pcb2gcode/gerbv-2.6.0/src'
/bin/sh ../libtool --tag=CC   --mode=link gcc -DBACKEND_DIR='"/mingw64/share/gerbv/scheme/"' -DSCMSUBDIR='"scheme"' -mms-bitfields -mwindows -Wall  -I/mingw64/include/gtk-2.0 -I/mingw64/lib/gtk-2.0/include -I/mingw64/include/atk-1.0 -I/mingw64/include/pango-1.0 -mms-bitfields -I/mingw64/include/gdk-pixbuf-2.0 -I/mingw64/include/cairo -I/mingw64/include/harfbuzz -I/mingw64/include/freetype2 -I/mingw64/include/glib-2.0 -I/mingw64/lib/glib-2.0/include -DLIBDEFLATE_DLL -I/mingw64/include/webp -I/mingw64/include/pixman-1 -I/mingw64/include/fribidi -I/mingw64/include/libpng16 -I/mingw64/include/cairo -I/mingw64/include/freetype2 -I/mingw64/include/pixman-1 -I/mingw64/include/glib-2.0 -I/mingw64/lib/glib-2.0/include -I/mingw64/include/harfbuzz -I/mingw64/include/libpng16   -o gerbv.exe dynload.o attribute.o lrealpath.o main.o project.o scheme.o callbacks.o interface.o render.o libgerbv.la gerbv_icon.o -lm   -lmsimg32 -lgdk-win32-2.0 -lgdi32 -limm32 -lshell32 -lole32 -Wl,-luuid -latk-1.0 -lharfbuzz -lpangocairo-1.0 -lgtk-win32-2.0 -lpangowin32-1.0 -lgio-2.0 -lpango-1.0 -lgdk_pixbuf-2.0 -lcairo -lglib-2.0 -lintl -lgobject-2.0 -lcairo
libtool: link: gcc -DBACKEND_DIR=\"/mingw64/share/gerbv/scheme/\" -DSCMSUBDIR=\"scheme\" -mms-bitfields -mwindows -Wall -I/mingw64/include/gtk-2.0 -I/mingw64/lib/gtk-2.0/include -I/mingw64/include/atk-1.0 -I/mingw64/include/pango-1.0 -mms-bitfields -I/mingw64/include/gdk-pixbuf-2.0 -I/mingw64/include/cairo -I/mingw64/include/harfbuzz -I/mingw64/include/freetype2 -I/mingw64/include/glib-2.0 -I/mingw64/lib/glib-2.0/include -DLIBDEFLATE_DLL -I/mingw64/include/webp -I/mingw64/include/pixman-1 -I/mingw64/include/fribidi -I/mingw64/include/libpng16 -I/mingw64/include/cairo -I/mingw64/include/freetype2 -I/mingw64/include/pixman-1 -I/mingw64/include/glib-2.0 -I/mingw64/lib/glib-2.0/include -I/mingw64/include/harfbuzz -I/mingw64/include/libpng16 -o .libs/gerbv.exe dynload.o attribute.o lrealpath.o main.o project.o scheme.o callbacks.o interface.o render.o gerbv_icon.o -Wl,-luuid  ./.libs/libgerbv.dll.a -lmsimg32 -lgdk-win32-2.0 -lgdi32 -limm32 -lshell32 -lole32 -latk-1.0 -lharfbuzz -lpangocairo-1.0 -lgtk-win32-2.0 -lpangowin32-1.0 -lgio-2.0 -lpango-1.0 -lgdk_pixbuf-2.0 -lglib-2.0 -lintl -lgobject-2.0 -lcairo -L/mingw64/lib
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: project.o:project.c:(.bss+0x0): multiple definition of `screenRenderInfo'; mai
n.o:main.c:(.bss+0x20): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: callbacks.o:callbacks.c:(.bss+0x0): multiple definition of `CALLBACKS_SAVE_FIL
E_TYPE'; main.o:main.c:(.bss+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: callbacks.o:callbacks.c:(.bss+0x20): multiple definition of `screenRenderInfo'
; main.o:main.c:(.bss+0x20): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: interface.o:interface.c:(.bss+0x0): multiple definition of `CALLBACKS_SAVE_FIL
E_TYPE'; main.o:main.c:(.bss+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: interface.o:interface.c:(.bss+0x20): multiple definition of `screenRenderInfo'
; main.o:main.c:(.bss+0x20): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: render.o:render.c:(.bss+0x0): multiple definition of `CALLBACKS_SAVE_FILE_TYPE
'; main.o:main.c:(.bss+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: render.o:render.c:(.bss+0x20): multiple definition of `screenRenderInfo'; main
.o:main.c:(.bss+0x20): first defined here
collect2.exe: error: ld returned 1 exit status
make[2]: *** [Makefile:437: gerbv.exe] Error 1
make[2]: Leaving directory '/home/Sybren/workspace/pcb2gcode/gerbv-2.6.0/src'
make[1]: *** [Makefile:319: all-recursive] Error 1
make[1]: Leaving directory '/home/Sybren/workspace/pcb2gcode/gerbv-2.6.0'
make: *** [Makefile:248: all] Error 2
```
Gerb 2.6.0 is also outdated, it's so old that it predates the releases on their [releases page](https://github.com/gerbv/gerbv/releases). Now that I read more, it turns out that that's a fork because the original project stalled. From their README:

[... cut ...]
You can automate this, so that people trying to follow the steps don't have to manually go over each library:

```
cp $(ldd pcb2gcode.exe  | cut -d'>' -f2 | cut -d'(' -f1 | grep mingw64 ) libs -v
```

And now I have an executable that works, at least on my machine:

```
C:\msys64\opt\pcb2gcode\bin>pcb2gcode.exe --version
2.5.0
Git commit: latest
Boost: 108100
Gerbv: 2.9.6~77211d
Geos: Not installed
```

So yeah, "You can easily build pcb2gcode for Windows with MSYS2" was not exactly true. But with some creativity, it was doable.

eyal0, 2023-02-11:
You can use the latest gerbv. I'm a developer of the latest gerbv, too.

PR #658 (2023-02-11): Update Windows build instructions in README
This PR updates the `README.md` file with instructions that actually work on my machine, as listed in https://github.com/pcb2gcode/pcb2gcode/issues/655

The following changes are made:

- Change "Windows prebuilt binaries **are** available" to "... **may be** available". Also I've added a little explanation as to why the Windows builds may be missing. It's intentionally vague so that it doesn't need updating for every release.
- Use the currently used names for the 32-bit and 64-bit MSYS2 shells.
- Write the instructions for 64-bit builds, leaving it [... cut]
closed 2023-02-11 by eyal0