pcb2gcode/pcb2gcode #656: ALignment problem on back side

Terak8510, 2023-02-05:
Git commit: latest-40-g5459c296
Boost: 107400
Gerbv: 2.7.0
Geos: Not installed

Using modified pcb2gcodeGUI with 'scribe-etch' method, scribe back side first, flip the board, scribe front side, drill, cut out.

config:
[pcb2gcodeGUI-current_scratch-etch.txt](https://github.com/pcb2gcode/pcb2gcode/files/10610083/pcb2gcodeGUI-current_scratch-etch.txt)

pictures front: 
[image]
[image]

pictures back:
[image]
[image]

The is a little bit wander on the front side from the smaller drills flexing drilling a non piloted hole.
The back side is a bit of a dogs breakfast.

I havent upgraded git-src in a while, mabye thats it?

Terak8510, 2023-02-05:
2 mods to the pcb2gcodeGUI source added i use:
[image]
[image]

eyal0, 2023-02-05:
When you flip it over, how do you make it line up?

Terak8510, 2023-02-05:
At X=0 I have 2 index pins on oppisite ends of Y at edges of clad board, I tried  X and Y offsets but that does is offset ALL
 cnc jobs at once, front back drill cut etc.

Terak8510, 2023-02-05:
might try the gcode on the 3d printer to see if it is consistant..

Terak8510, 2023-02-05:
It could be candle v2.2, i have asked over there as well.
https://github.com/Schildkroet/Candle2/issues/41

eyal0, 2023-02-05:
I don't use pcb2gcodeGUI, just pcb2gcode. Do you have test files that I can try?

Terak8510, 2023-02-05:
k.
[the-zip-file.zip](https://github.com/pcb2gcode/pcb2gcode/files/10610291/the-zip-file.zip)
its in 7z format.

eyal0, 2023-02-05:
I decompressed your files.  I also added a file called `millproject` to the directory and I put in the contents from [pcb2gcodeGUI-current_scratch-etch.txt](https://github.com/pcb2gcode/pcb2gcode/files/10610083/pcb2gcodeGUI-current_scratch-etch.txt) which you linked above.  Then I ran ~/pcb2gcode and it completed in 17 seconds on my computer.

I use `bCNC` to open the generated gcode and I got these two files from front and back:

[image]

What you are describing should work.  Let me make sure that I understand:

First, you are placing your PCB on the CNC and you have two pins sticking out, one is at X=0,Y=<something> and the other is at X=0 and Y=<something else>.  And you have put on a copper clad that has two holes it in that match the two pins?  Also, when the CNC goes to X=0, it is exactly between these two pins?

If so, it should work.  I will show in bCNC:

[image]

I have drawn a measurement from the center to the center of the closest pad.  We see that, on the front, delta x is 2.19138mm and delta y is 2.16399mm.  And on the back, the measurements are the same except that delta x is negative.  This is because the flip is along axis 0.  (The measurements are not precisely the same because I'm limited by my ability to find the exact center of the pad with my mouse!)

So it seems like it should work.  Maybe your procedure is not what I described?  Can you describe in more details, perhaps with photos?

Also, I direct you to the wikipedia page on all of this:

https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Alignment-for-two-sided-PCB-and-offsets

Terak8510, 2023-02-05:
sounds like you have possibly found the issue, the other piece of software is
Kicad, mabye the gerbers are not at the same position  with the exellon drl files?
here is a picture of my laseraxe 2417 cnc engraver with pins at X=0:
[image]
left area of spoilboard is back, right area is front when i flip it over.

The board is purposely skewed to see if that mattered, it does not as long as the index pins are well fastened.

lower pin of copper board is at X=0, so is upper pin, just verified, powered down steppers and carved a straight line from upper index to lower index hole by manually turning Y knob and it drew a line right to the other index hole.
do you use kicad?

here is a screenshot of my plot and drill settings for that program:
[image]
[image]

Terak8510, 2023-02-05:
currently googleing "kicad drill files not aligned with gerbers"

Terak8510, 2023-02-05:
I turned up the stepper current on all 3 axis earlier today, it made no difference.
(thought it was skipping steps at high speed)
besides the inscribed pattern looks perfect.

Terak8510, 2023-02-05:
I don't think it is a Kicad .grb/.drl issue, As the drill output is in perfect alignment
with the same 'side' of  the other gerber layer output.

it's just when i flip the board over.

eyal0, 2023-02-05:
The drill file appears aligned to me.  In your images, looks like the back side is wrong in both x and y.  This seems strange.  Are you confident that your procedure for flipping over the board is maintaining the (0,0) correctly?  How could you test this...?

Well, you would jog your machine to x=0 and check that the pins that you are using for alignment when you flip are directly below the milling bit.  And it should be like this for both sides.

BTW, you are flipping only left to right, along the x axis, right?  You can't flip vertically, only horizontal.

I see the drilling coordinates in bCNC and they look perfect so I don't understand the issue.

Terak8510, 2023-02-05:
Uh, I should have sent you the 'output' folder:
[cncfiles.zip](https://github.com/pcb2gcode/pcb2gcode/files/10610469/cncfiles.zip)

And currently running the .ngc files online through https://ncviewer.com/

Terak8510, 2023-02-05:
meh, that online viewer only does 2 dimensional draws and cannot use multiple files.

I also noticed in Kicad, that the copper fill is larger than the outline.

Terak8510, 2023-02-05:
I have found it,
The cnc engraver is not square on all 3 axis. (most likely)
I used a piece of photocard as a straight edge.

Unplugged and set X for middle -top of waste-board, 
Layed down a photocard, put an index pin there.
Hand turned Y from 0 to other limit, put an index pin there.

Drew a line from lower left edge of photocard on spoilboard.
Flipped over card on index holes and drew another line with pencil left to right.

I am measuring an error length of 3.21mm of the line ends of X.
Z axis is also not square as there is a little Y error of previous attempts.

This CNC has seen quite a few nasty bit crashes over the years.
closing.

Terak8510, 2023-02-06:
I tightened every screw and checked for square on whole CNC frame.
Really not that far off, the photocard was probably not lined up right.
Even if the frame was out of alignment, Wouldn't the freshly drilled index holes 
take care of any alignment problems? 

Running one last scribe job to see if anything has changed.
[image]
At least now the holes are mostly inside the pads, the gantry still needs to be adjusted Y+/- ??? on one side.

Terak8510, 2023-02-10:
Well i have realigned the CNC for squareness and resurfaced the spoilboard.
Still getting alignment errors even with big pads and holes in X direction.
[image]

Does not matter where i put the job on the spoilboard, the hole/pad is always
off to one side on X:
[image]

I can get around this by: 
drilling and milling front first,
flip board then run back job setting feed rate slow,
pause the job on first hole and turn X knob to compensate, then continue.
All the machine errors should have been eliminated when i resurfaced the spoilboard.

How difficult would it be to add --offset-side mill/drill option in the code?

eyal0, 2023-02-15:
This must be a problem in your CNC, right?  I looked at the drilling and milling in bCNC and it's entirely accurate.  If you flip along the X axis, the holes should line up *precisely*.

Do you think that there remains an imprecision in your CNC or do you think that the error is in `pcb2gcode`.  If the latter, let's try to find definitive proof one way or the other.

Terak8510, 2023-02-15:
Can confirm that it's NOT your code, I am stumped about why X alignment is off.
[image]

Thanks for your patience.

closed 2023-02-15 by Terak8510

Terak8510, 2023-02-19:
This was solved by not setting --zero-start and enabling auxiliary axis in kicad.

Terak8510, 2023-02-25:
Just for reference, I found the problem with why X/Y was not lining up when flipping the board.
There was a misalignment on the CNC, being X was not perpendicular with Y.
X was off by 0.8mm from one end to the other.
So when then board flipped, the error was now 1.6mm.
I loosened one side of X and used a straight edge between the rails of the X axis and moved the edge of the Y table right against the strait edge and re-tightened X rails.
Problem solved.