pcb2gcode/pcb2gcode #660: Coordinates of alignment holes in two sided PCB's

manushawijekoon, 2023-02-18:
I am trying to make a two sided PCB with Kicad and pcb2gcode. I have been using flatcam before, but the GUI based process is very slow so want to migrate to pcb2gcode. I have previously used it to make 2 sided boards, but this time things did not go well and there are things that I do not understand so wanted open this issue to get them clarified.

This wiki (https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Alignment-for-two-sided-PCB-and-offsets) mentions to jog to alignment holes and drill them as the second step. How do I know the coordinates of alignment holes? Is it relative to the board origin/drill and placement origin or is it relative to the coordinate system that is internally used by pcb2gcode? Or is this just the offset that we specify in the millproject file under mirror-axis parameter (so we do not need to think about any coordinate system)? If this is the case and if we assume that mirror-axis is set to 150, all I need to do is to go to X=150 and make two holes at any two Y coordinates?

eyal0, 2023-02-18:
Yup! If mirror axis is 150 then you need to flip the board along x=150. The easiest way to do that might be to drill a couple holes along x=150 and then flip and align on those holes.

I use bCNC to view my .NGC files. If you concatenate the front and back files, you can view them both at the same time and see how it lines up.

Good luck!

manushawijekoon, 2023-02-18:
Thanks. This essentially means mirror-axis is relative to whatever coordinate system that is used inside pcb2gcode and has nothing to do with the coordinate system used in the board. Also, origin of this pcb2gcode coordinate system is same as machine origin. 

I will try bCNC. How do you concatenate two files?

eyal0, 2023-02-18:
> Thanks. This essentially means mirror-axis is relative to whatever coordinate system that is used inside pcb2gcode and has nothing to do with the coordinate system used in the board.

Yes, but there is an option to make the coordinate systems be the same and that is called zero-start.

> I will try bCNC. How do you concatenate two files?

In Unix you can type `cat front.ngc back.ngc > both.ngc`. or you can open an editor and just copy and paste the two files into a single file, with the contents of one after the contents of the other.

manushawijekoon, 2023-02-18:
ok, its just two text file concatenation. 

I guess even if I set zero-start to false, there won't be any change in procedure. In my above example, I will still need to move to X=150 and put the holes? 

The issue I face with zero-start set to true is that the origin used by the tool is bit offset from board edge so setting X=0, Y=0 in the machine is bit difficult without knowing this offset especially the copper board is approximately same size as the board.

eyal0, 2023-02-21:
The idea behind zero-start is that the 0,0 of the milling will be in the bottom-left corner of all the area needed for milling.  So, by default, if your 0,0 is lined up with the bottom left corner of the copper FR4 then you should be maximizing the area that you can mill.