pcb2gcode/pcb2gcode #661: Arcs seem to have 32 segments per 360 degrees, making them chunky as they get bigger

hachi, 2023-03-12:
## Description of bug:

## What did you try (include command-line arguments):
I'm using KiCAD to layout a board with many curved traces, and some of the larger radius traces are visibly becoming polygons with pcb2gcode

## What happened:
For any size of arc I'm getting a visibly 32 sided per 360 degrees interpolation of curves.

grepping through the code I found a '32' in bg_helpers.cpp rather quickly, but after changing that to 128 the behavior didn't visibly change.

## What did you expect to happen:

Some reasonably smooth interpolation of larger arcs.

$ pcb2gcode --version
2.5.0
Git commit: latest-dirty
Boost: 107400
Gerbv: 2.9.6
Geos: Not installed

$ git describe
warning: tag 'v1.2.0-alpha' is externally known as 'v1.2.0'
v1.2.0-1560-g7dd6252b

My working tree has a single change in it... the 32 -> 128 in bg_helpers.cpp, thus the dirty mark.

hachi, 2023-03-12:
[ArcTest-F_Cu.gbr.txt](https://github.com/pcb2gcode/pcb2gcode/files/10951338/ArcTest-F_Cu.gbr.txt)

Here is a reasonably small example case for input

eyal0, 2023-03-13:
Can you try modifying here instead?

https://github.com/pcb2gcode/pcb2gcode/blob/master/board.cpp#L130

It's 30 points per circle.  Try to change it to 128?

hachi, 2023-03-13:
That works, thanks.

I had optimise off, and so this increased the size of the output considerably, but then turning it back on helps a lot.

eyal0, 2023-03-13:
What if I changed it so that the points per circle depended on the size of the circle. Is that better? Like we the points so that they are never more than some known distance apart?

hachi, 2024-04-30:
I should have replied to this long long ago, but yes that would be good... I'm still using my slightly modified copy for building pcbs.

m-anish, 2024-12-24:
this would be a nice fix to have!

great project, btw. i'm a migratory bird from flatcam-land now :)

PR #716 (2025-06-06): feat: Make circles less "chunky" by using more segments in their polygonal approximation.
Rather than always use 32 segments per circle, use as many as necessary to achieve 0.0004 inch or less per segment.  (0.0004 inches is the default tolerance value.)

This fixes #661
closed 2025-06-06 by eyal0