pcb2gcode/pcb2gcode #663: Tool indexing starts at zero, typically a reserved value by machines

hachi, 2023-03-13:
## Description of bug:

I know there are open bug(s) about toolchange customization, but this one should be easy to consider out of step.

I'm using LinuxCNC and T0 specifically means 'no tool', aka 'unload tool, leave spindle empty'. pcb2gcode really needs to start indexing at 1 in the default case.

Anecdotally I've been told that many other machines take the same meaning, though I can't cite reference for them:

https://linuxcnc.org/docs/html/gcode/other-code.html#sec:select-tool specifically the first note in this section.

eyal0, 2023-03-13:
Doesn't the tool number come from the input Gerber? Or maybe that is only for drilling? I don't remember.

If it comes from pcb2gcode then I can fix this, yeah.

hachi, 2023-03-13:
Gerber has no tool concept in it. I do think you're talking about the drill table. Thanks for your work on these tools, they're great.

eyal0, 2023-03-14:
> Gerber has no tool concept in it. I do think you're talking about the drill table. Thanks for your work on these tools, they're great.

Yes, it seems like, for milling, `pcb2gcode` is inventing the numbers, start at 0.  For the drill file, it uses whatever the drill file has.  So I'll just fix the milling files, does that seem okay to you?

closed 2023-03-14 by eyal0

hachi, 2023-03-14:
Totally, thank you!

PR #664 (2023-03-14): fix: Start tool indicies at 1.
This fixes #663