pcb2gcode/pcb2gcode #665: configure program file is missing

wryan67, 2023-03-26:
## Description of bug:
The "configure" program file is missing, therefore compiling is impossible.  Would like to compile on Ubuntu 20.04.

## What did you try (include command-line arguments):
```
    $ git clone https://github.com/pcb2gcode/pcb2gcode.git
    $ git branch
    * master
    $ git rev-parse HEAD
    aaf5c83f51f4b21c82312cc69f61a7e385524ace
    $ ./configure
    bash: ./configure: No such file or directory
    
    $ find . -name configure
    {not found}
```

## What happened:

can't compile 

## What did you expect to happen:

would like to compile

eyal0, 2023-03-26:
Oops! Can you run `autoreconf -fvi` and then try again?

wryan67, 2023-03-26:
Opps indeed!  I must have been looking at old instructions.  Thank you!
That got me further down the road.  I ran auoreconf -fvt, got boost 70
installed, and ran configure as seen below, but now get this error:

$ ./configure  --with-boost=/usr/local --enable-static-boost
...
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for glibmm... yes
checking for gdkmm... no
configure: error: Package requirements (gdkmm-2.4 >= 2.8) were not met:

No package 'gdkmm-2.4' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables gdkmm_CFLAGS
and gdkmm_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

My options for libgddmm are:

$ sudo apt install libgdk
libgdk3.0-cil            libgdk-pixbuf2.0-0       libgdk-pixbuf2.0-common
 libgdk-pixbuf2.0-doc
libgdk3.0-cil-dev        libgdk-pixbuf2.0-bin     libgdk-pixbuf2.0-dev

Do you know if any of these will work, or do I need to download this and
compile manually?

_wade

*501-944-9849*

wryan67, 2023-03-26:
Hmm, it was hiding.  Again, thank you!

$ sudo apt install libgtkmm-2.4-dev
success!
$ sudo apt install librsvg2-dev
success!
$ ./configure  --with-boost=/usr/local --enable-static-boost
success!
$ make
success!

new error:

_wade

*501-944-9849*

eyal0, 2023-03-26:
What new error?

wryan67, 2023-03-26:
Thanks, I can now mirror y-axis!  That's what I needed!

$ pcb2gcode \
> --metric \
> --mirror-yaxis=on \
> --front   ${PREFIX}-F_Cu.gbr \
> --back    ${PREFIX}-B_Cu.gbr \
> --zsafe 1 \
> --zchange 10 \
> --zwork -0.02 \
> --offset 0.1 \
> --extra-passes 5 \
> --mill-feed 100 \
> --mill-speed 6000
Importing front side... DONE.
Importing back side... DONE.
Importing outline... not specified.
Processing input files... DONE.
Exporting back... DONE. (Height: 1.07417in Width: 1.19465in)
Exporting front... DONE. (Height: 1.07417in Width: 1.19465in)
Importing drill... not specified.
END.

_wade

*501-944-9849*

wryan67, 2023-03-26:
Thank you!  Ubuntu 20.04 confirmed working on version 2.5.0

closed 2023-03-26 by wryan67