pcb2gcode/pcb2gcode #666: Via’s need extra passes

wryan67, 2023-03-26:
When using brass rivets for via’s, the top (front) side needs extra passes so that the head of the rivet doesn’t touch copper which might be part of a polygon segment or ratsnest.  The via drill file may be needed to identify where the via’s are on the front side, and 1 or two parameters.

Example brass rivets:  https://www.amazon.com/dp/B015CV377O

New command line options:

```
--via-isolation-radius [unit]
--via-pad-radius [unit]
--via-drill-file [filename]
```
The idea is that the program will compute and etch the necessary passes to remove copper between the via-pad-radius and the via-isolation-radius.  I’m not sure if the via-pad-radius can be extracted from the drill file or not.  If it can, then that parameter wouldn’t be necessary.

eyal0, 2023-03-26:
Can you provide a sample file with vias?

Keep it simple, please!

wryan67, 2023-03-27:
Via's are in the Excellon drill file that is usually output with the Gerber files, or in addition to the Gerber files.  In order to locate the via's, you'll need to know the tool number that was used, so maybe these options make more sense:

```
--via-isolation-diameter [unit]
--via-drill-bit-diameter [unit]
--via-drill-file [filename]
--via-tool-num [values]
```

Where:
* The extra passes start at either the diameter of the drill bit, or where the previous etching stopped, and extend to the diameter supplied by --via-isolation-diameter.  
* The pcb2gcode feature should recognize wires attached to the via's and take care not to cut them.
* "unit" is floating point notation and the units are governed by the --metric option.
* "filename" is the relative or full path to the drill file that contains the via's.
* "values" is a comma separated list of tool numbers, and tool numbers can be optionally preceded with "T" or "t".  I've also observed that some programs zero-fill the tool numbers, like T01 instead of just T1.  

In my example, the file named "blinky-PTH.drl", contains all the drill locations, including the via's, and the via tool is labeled "T1".  I'm not sure how to upload a file to the "issues" tracking section of git hub, so here's a public link to my test project:  https://1drv.ms/u/s!AgXEVbmX6KyIgdAzPi9q7nrF6nTRhg?e=gLTbUt

The 3 lines from "blinky-PTH.drl" that are directly associated with the Via's are:

```
T1
X92.71Y24.13
X107.95Y21.59
```

The test project zip file contains these specific files which may be useful for testing this feature request:

* Specific drill file:  ./blinky/gerber/blinky-PTH.drl
* gerber files: ./blinky/gerber/*
* kicad project:  ./blinky/blinky.pro

Note:  The drill diameter can also be obtained from the drill file by parsing the line:
```
T1C0.400
```
So that means that the option --via-drill-diameter should be optional and if supplied, use the command line value for the via drill diameter, and if not supplied, parse it from the drill file.  

BTW, my email address is wryan67@gmail.com.  Feel free to contact me via email if that is more convenient than the issue tracking system.

c-, 2023-03-27:
This isn't really a pcb2gcode issue.

Your project design rules aren't right for the rivets you're using. In the design rules for those netclasses (or just make it default) you need to specify the correct via drill size (usually 0.1mm larger than the OD of the via) and a via size (AKA the OD of the annular ring around the hole) such that it's no smaller than the OD of the rivet once pressed. In practice that's usually twice the rivet diameter, but it depends entirely on the thickness of the PCB, the length of the rivet, and any errors in the assembly (rivets don't always press out to an even diameter)

You have a 0.4mm drill size and a 0.8mm via size defined for those vias, and that's not even close. For those 0.9mm rivets, I use a 1mm drill and 1.8mm annular ring, which is fine with a 1.4mm PCB and gives you some room for error if/when the via crushes sideways a bit.

This is going to be the case with all your design rules; you need to adjust them for the milling tools you plan to use with the board, because it's way too easy to lay out a board that you simply can't mill with the tools you have on hand or within the constraints of your machine (i.e. you're going to struggle to mill 0.5mm pitch parts if you have 0.1mm of total runout).

eyal0, 2023-03-27:
I've never done vias.

When you do this, the rivet is to connect the front and the back of the board, yeah? Do you then also solder or wire to connect the rivet to the traces?

Usually I prefer to paint before I drill so that the little drill holes don't fill with the UV paint. So I guess that I would first etch, then drill vias, then rivet and solder those to the traces, then paint, and then drill the through-hole components?

Like that?

c-, 2023-03-28:
> the rivet is to connect the front and the back of the board, yeah?

Yes. They're quite convenient for joining sides together. The main downside is good quality/smaller size rivets aren't cheap.

> Do you then also solder or wire to connect the rivet to the traces?

Wire, no, the via is pressed solidly to the copper and it's conductive (copper or brass), so nothing extra is needed. Soldering rivets isn't *supposed* to be necessary, but it's a good idea (corrosion could get under them, and I've had vias lose connectivity after conformal coating). I usually solder with hot air, so I just dab with a smear of solder paste before placing components.

Uh... here's a section of a board I had sitting on my desk to give a sense of what it looks like... this is the front:

[image]

And here's the back of that section:

[image]

Those vias are in 0.7mm holes with 1.4mm vias. IIRC the manufacturer recommends 1.2mm, and you can probably tell that wouldn't leave much room for errors.

>Usually I prefer to paint before I drill so that the little drill holes don't fill with the UV paint. 

If you're using paint/solder mask (I don't bother with it), the best approach would be to keep the via "pad" area (the annular ring) free of solder mask and then drill all the various holes together after applying the mask. This keeps the area the via gets pressed into free of the mask. In Kicad, you avoid plotting solder mask over the vias by turning on "Do not tent vias" when plotting the Gerbers.

eyal0, 2023-03-28:
Wouldn't I *prefer* to tent the vias? I want to paint over them so that I don't have all these exposed traces ony board that can short.

That's why I would drill and put in rivets before the paint. But I would drill the through-hole components after the paint so that I don't get paint in the little holes.

All this requires having a good jog that will let you remove the board from the CNC and put it back with accurate positioning.

c-, 2023-03-28:
That's why I like conformal coating. I can just paint a clear coating on the entire board after it's assembled and tested. Easy to do, no extra steps during milling, and protects it from almost everything. It also looks *lovely* with black FR4.

I don't use a jig for alignment. bCNC and a cheap USB endoscope does optical alignment way more accurately (<0.1mm error) than I've ever achieved with jigs.

wryan67, 2023-03-28:
> This isn't really a pcb2gcode issue.

Agree that this is a feature request as the first comment suggested, not really an issue.  I don't think github distinguishes the two categories yet, one of their few shortcomings.

Well, unless I missed a feature in the pcb software, I don't think there's a way to change all the via's at one time, which some of my larger boards have upwards of 75-100 via's.  If I mill the board myself, I use a 1.0 mm bit and 0.9 mm rivets.  This allows the rivets to drop into place quick and easy, then I solder them, top and bottom.  But if I send the board off for manufacturing, I want them drilled with the smallest size possible, mostly for aesthetics, but I can also then shrink the board size down too, if need be.

So then, what I'm doing is designing the boards with the smaller size via's and just making sure I leave room if I'm milling it myself, then when it comes time to drill and the machine asks for a 0.4 mm bit, I stick in my 1.0 mm bit.  mostly problem solved.   right now, I just have the board etched with massive isolation tolerances everywhere so that the heads don't touch unwanted copper.  I'd like to shorten the etching time by using a smaller isolation gap, but use these new parameters to give extra isolation at the via points.

wryan67, 2023-03-28:
> The main downside is good quality/smaller size rivets aren't cheap.

I bought 500 rivets on aliexpress for $4.13 delivered.  I thought that was pretty cheap.  Only down side is that it took 2 weeks for delivery.  Sometimes I buy small quantities of stuff on Amazon to tide me over until aliexpress can cross the ocean.  

https://www.aliexpress.us/item/3256803948375815.html

wryan67, 2023-03-28:
> "c-": referring to your images

Mine look similar and I can see that your isolation gab is quite massive, like mine.  That is exactly what I'd like to see shrink, except around the via's.  It appears that you're also pounding your via's in using a hammer and a punch or something.  I was doing that as well, but then the heads tend to get mangled and touch unwanted copper, plus, that is time consuming.  I found that it was much faster to drill the hole larger than the via.  That way the via's simply drop into place or just need a slight push using a fingernail.  You can drop in 10-20 quickly, then solder the tops down, then do another 10-20, and so on until you're done with the top.  Then flip the board over and solder the bottoms.  Next comes continuity checks...

eyal0, 2023-03-28:
How do you do optical alignment? Is the camera on the CNC head with cross hairs and you jog over to the 0,0 or something? Do you do it on multiple points and then have a potential not-square milling?

wryan67, 2023-03-28:
> How do you do optical alignment? Is the camera on the CNC head with cross hairs and you jog over to the 0,0 or something? Do you do it on multiple points and then have a potential not-square milling?

It's not optical at all.  There's two limit switches, which are used to zero the machines x & y axis.    From then on, using machine coordinates, I know exactly where the holes in my MDF are, and where the holes in my copper-clad are, and where to move the head to in order to zero out the work coordinates before etching/drilling/etc. 

I made a spread sheet to keep track of all the different coordinates.  When I replace the MDF, I'll probably have to take care to cut a new board the same size as my current one and clamp it down in nearly the same spot.  Then I can use my spreadsheet to re-drill the copper-clad positioning holes.  

BTW, the pre-drilling is done 10 mm to the left of the positioning holes so as to not disturb my precious alignment holes.  

The z-axis is zeroed using the machines probe feature.

[image]

c-, 2023-03-28:
> Well, unless I missed a feature in the pcb software, I don't think there's a way to change all the via's at one time

You missed a feature ;)

Under the Edit menu, there's a "Edit Track & Via Properties" dialog that allows you to do global size changes in one operation. You need to add the via sizes in the design rules first, though.

> Mine look similar and I can see that your isolation gab is quite massive

Absolutely. I keep the gap around my ground pours large because I normally use a 0.8mm endmill to do the bulk of the clearing to save time and wear on my finer cutters, and larger isolation makes solder bridging less of a problem. Between traces I'll allow clearances down to maybe 0.2mm when necessary. When I'm working with finer pitch components I try to clear away as much unnecessary copper as possible so I'm not fighting with solder in places I don't need it, which in more extreme cases ends up looking like:

[image]

or

[image]

However, things like the isolation gaps, drills sizes, via sizes, etc are things that I set up in Kicad, and pcb2gcode gets configured to render those pretty accurately.

The ability to use multiple bit sizes for clearing copper is probably one of my favorite features of pcb2gcode.

> I bought 500 rivets on aliexpress for $4.13 delivered.

0.9mm brass rivets are cheap. 0.6mm copper rivets are pricey and the smallest I can find. I gave up on those. I've switched to just pressing copper EDM electrode wire into my vias, which I find faster and allows for even smaller sizes (I've tested down to 0.35mm holes).

c-, 2023-03-28:
> How do you do optical alignment? Is the camera on the CNC head with cross hairs and you jog over to the 0,0 or something? Do you do it on multiple points and then have a potential not-square milling?

Yes. You drill some holes on one side of the board, flip it, load the other side, and find them with the (calibrated) camera, tell bCNC which holes in the g-code each one corresponds to, and it calculates a correction that can be used to nail the alignment.

The makers of The Ant PCB mill have a [really detailed video](https://www.youtube.com/watch?v=LN1jj2Rn3CA) about how to do it in PCB milling, and that's pretty much what I do.

wryan67, 2023-03-28:
@c-  -- quite impressive.  I guess I got suck on v-bits.  I'm using a 4-flute V-Grove, 60 degree, which has taken a beating and is still working fine, even for the TSSOP chip.  https://www.amazon.com/gp/product/B099DGL66T

What bit exactly are you using?  Does it ever break?  How many boards can you etch before needing to use a new bit?

With a 0.8 mm mill bit, I guess that means TSSOP is out, right?  TSSOP requires 0.65 mm between pins and pins are 0.3 mm.  The suggested pads are supposed to be 0.45, which leaves a gap of 0.2, but I think my gaps are approximately 0.3.

eyal0, 2023-03-28:
It's fun to see all these cool uses of pcb2gcode. Thanks for the photos.

For the alignment with screws, did you consider screwing in bolts from below, counter sunk? And then you can you can use nuts on top. Nuts on bolts will hole threads a lot longer than screws in wasteboard wood.

Anyway, cool work you two!

wryan67, 2023-03-28:
Ya'll are making me jealous with your cameras.  lol     

Technically, I've not use pcb2gcode yet.  I'm wanting to switch to it very soon.   I've been using eagle-cad (free), which has size limitations.  I want very much to use kicad and pcb2gcode.  

I get what you're saying with the nuts and bolts, but I elected to use the wood screws because the heads are v-shaped.  The v-shape forces the copper-clad board into the exact same position no matter which side you're screwing down.  

My method is pretty accurate.  For some reason, I have to adjust the vertical 0.4 mm when I flip the board.  Other than that one adjustment, it's quite accurate, even for boards 80x80 mm.  I wish I could figure out the reason for the 0.4 mm error.  IDK if it's something in my machine setup or the eagle-cad software plugin that I'm currently using (pcb-gcode v3.6.3).    Just can't pin the reason down yet.

wryan67, 2023-03-28:
> Under the Edit menu, there's a "Edit Track & Via Properties" dialog that allows you to do global size changes in one operation. You need to add the via sizes in the design rules first, though.

Thanks C- !   I will definitely be checking into that this week!

wryan67, 2023-03-28:
Back on topic.  Below you can see the vias that I'm making now come awfully close to the copper rats nest, despite a 0.5 mm tolerance on the wire tracings.   This board takes 40 minutes to etch as is.   Increasing the tolerance make the time go up significantly.  I would actually like to reduce the tracing tolerance so that the time will come down a bit.  I guess using the different size bits as you've suggested would improve the tolerance radius around the via's, but then you have to change bits.   For me that's a small problem since I have to change bits manually, but it makes me sigh heavily thinking about yet another drill bit change, lol.  Also, I'm not sure what the time difference would be for 1 bit vs. 2.  

So, in the end, I'd still like to see if Eyal, or someone, could code the 2nd set of options suggested above and then let's test to see which method is faster/easier overall.  

[image]

More photos:  https://photos.app.goo.gl/xL2JEkzpdf1aaetP8

Hackaday:  https://hackaday.io/project/189794-rotor

"C-", now you also have me curious about that 0.35 mm via.  If you have a video or some more instructions on how to do that, I would appreciate it.

c-, 2023-03-28:
>For me that's a small problem since I have to change bits manually

Again the secret sauce is bCNC, and specifically its assisted tool change support with tool length offset probing. [On my machines I use tool length switches](https://www.printables.com/model/158994-cnc-1610-tlo-switch-holder), but if you have your probe permanently wired up then you can just probe against a fixed position on the copper.

But you'll also find that larger bits clear faster, which makes the tool change time a wash. I've done plenty of manual tool changes, it really is worth it.

FWIW, 0.8mm is kinda arbitrary. You could go larger, or smaller. 0.8mm just happens to hit a sweet spot where it can still clear between 0.1" pads.

>If you have a video or some more instructions on how to do that, I would appreciate it.

I'm starting to think I should just write up a tutorial of my entire workflow.

The simple explanation is that I jam an EDM rod into a via hole until it bottoms out, cut it flush-ish with good cutters (bad cutters distort the rod such that it won't fit the next hole), then press it with a modified [bead press](https://www.amazon.com/Setting-Machine-Colorful-Clothes-Accessories/dp/B0B4H35HV7) (the slight distortion from the cutter is enough to hold the via in by friction while getting it into position for pressing). The modification is a finer head and adjusting it for a smaller gap (which required tapping an existing M12(?) thread a bit deeper). Here's a photo of some 0.6mm vias between cutting and pressing:

[image]

Because EDM rod is hollow and distorts very flat in even a weak press, you can allow a surprising amount of excess than you could get away with using solid wire. EDM rod used like this effectively becomes an nearly-headless hollow rivet.

0.3mm rod is more finicky, and I think I prefer brass over copper just for the stiffness, but it works, at least in testing.

The process doesn't require flipping the board, so it's a lot faster than conventional rivets, and you can get away with being sloppy. The last board I did had something like 200 vias and took maybe 30 minutes, using almost an entire 400mm electrode.

Price-wise, 400mm copper EDM electrodes are about $0.10 each.

c-, 2023-03-28:
I picked up [these ones last time] (https://aliexpress.com/item/32633816123.html). I have a preference for copper in that size as it presses better. For the smaller sizes I think brass is a bit stiffer, and at 0.3mm it's going to flatten like butter anyways.

wryan67, 2023-04-06:
Still waiting for my rods to come in, but in the meantime, I've done a full end-to-end test using pcb2gcode, and very much like it!  But still, I can't help but think that anyone using rivets would want these extra control parameters to speed up the etching process.  As you can see in my test below, In order to get a full 1.5 mm isolation diameter around my via's, the software had to make two passes around everything, including the long straight wire tracks, which don't need 2 passes.  FYI, I was using a 0.8 mm milling bit.

If you look at the 3rd photo below, you can see that I only have 5 via's.  So those are the only spots that need 2 passes (1.5 mm).  The entire rest of the etching would have been just fine with only 1 pass (0.8 mm).   I think the suggested parameters would have cut the etching time in half, not that this board took that long, it was only 29x33mm, but I plan on making boards in the 100x150mm range.  

More photos here:  [Relay 101](https://photos.app.goo.gl/Jw3wBWY27KajZzZWA)

[image]

[image]

[image]

c-, 2023-04-07:
My gut feeling is the way to get more space around the vias is to just make them larger (say, via width 2mm for a 1mm hole). It won't give more *isolation*, but it'll give breathing room for soldering and riveting errors.

The *best* place for this kind of clearance control would be in Kicad itself, using copper pours, because it already knows about vias and clearances and stuff, but it doesn't have anything specifically for vias like it does with pad clearance overrides. You can fake it using [the teardrop plugin](https://github.com/NilujePerchut/kicad_scripts) by adding extra clearance on the teardrop areas, and there's probably a more direct way to script it.