pcb2gcode/pcb2gcode #667: Unwanted cut if backtrace is used

Tobias4git, 2023-04-30:
I'm trying to get rid of an unwanted cut for a very simple test board.
The layout should look like this:
[image]

(screen shot of gerbv)

The gerber file looks very simple:
G75*
%MOIN*%
%OFA0B0*%
%FSLAX24Y24*%
%IPPOS*%
%LPD*%
%AMOC8*
5,1,8,0,0,1.08239X$1,22.5*
%
%ADD10C,0.0520*%
%ADD11C,0.0157*%
D10*
X001500Y001500D03*
X002500Y001500D03*
X002500Y002500D03*
X001500Y002500D03*
D11*
X002500Y002500D01*
M02*

With pcb2gcode V 2.5.0 and the following parameters
pcb2gcode --zsafe 5 --zchange 100  --zwork -0.05 --zcut -1.8 --mill-diameters 0.2mm --cutter-diameter 2.4mm --mill-feed 200 --cut-feed 800 --mill-speed 20000 --cut-speed 16000  --front millTest2-2bot-traces-GERBER_RS274X-1.gerber  --basename millTest2-2bot-traces --metricoutput  --vectorial 1 --software mach3 --optimise 0 --eulerian-paths 0

I got these cuts:
[image]

On the left hand side is the unwanted cut. Why is it there?
If I disable backtrack (using --backtrack 0) , it works fine:

[image]

Is it a bug or a feature? For bigger designs, backtracking might be very useful. I don't like to disable it entirely.

Thank you
Tobias

c-, 2023-04-30:
> Is it a bug or a feature?

Feature. It surprised me too the first time I saw it happen, and I still have to do a double-take when I see it.

pcb2gcode optimizes moves between segments by not lifting the tool if there's "nothing" in between them and it's quicker than a z-lift+rapid+z-lower (as motion in z tends to be the slowest operation). If a section of copper isn't tied to any net (not a ground pour, for example), it's considered "nothing" and is fair game for this optimization.

A simple workaround is to use a pour/fill to tie that "nothing" copper area to a net (i.e. a ground pour), in which case pcb2gcode should leave it intact.

eyal0, 2023-05-01:
You can set it to zero.

https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Optimizations#--path-finding-limit

Tobias4git, 2023-05-01:
Thank you, you two!
It would have been much quicker, to ask. I got distracted by the first short explanation on the page
https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Optimizations
`"--backtrack arg (=inf) allow **retracing** a milled path if it's faster than retract-move-lower."`

closed 2023-05-01 by Tobias4git

eyal0, 2023-05-01:
Yeah, that was not a great choice of words!  I don't remember why I even called it backtracking.  Hmm.  In some cases, that little backtrack *does* a retrace but it's correct that it will sometimes just zip across a region that you don't care about.

c-, 2023-05-01:
I think my original suggestion really was backtracking, in the sense of following an already milled path. I can't remember how the constraint became "you can go anywhere we don't need to leave copper".

eyal0, 2023-05-01:
Probably I just got excited and went overboard!

Tobias4git, 2023-05-01:
I just stumbled across another unexpected behaviour:
(Should I open another ticket? Probably not, it's again probably work as designed and just surprising for me)
Tracks outside the outline are not milled. If you dont specify a gerber for the outline, everything is there.
If you add the outline gerber, some tracks are missing.
Background:
To mill a two sided board, you have to flip it over. To ease the adjustment I added two holes.
To make sure that I see the holes on any layer for verification, I added them to the front, back and also the outline.
Why does it disappear on the front.ngc.... because it's then outside the outline.
May I adjust the wiki page https://github.com/pcb2gcode/pcb2gcode/wiki/Manual?

eyal0, 2023-05-01:
Oh man I just added this feature at someone's request! Right? I think so. Let me check.

Tobias4git, 2023-05-01:
Don't get me wrong! If you cut the outline before milling (that's what I like to do, to limit the number of tool changes),
that's exactly what you need. It would be nice to add in the doc. I can help you there.
You did probably add code to achieve this, so the program knows exactly that it discards traces. If there had been a warning, it would have saved me an hour....

eyal0, 2023-05-01:
Ah, I hadn't done it yet: https://github.com/pcb2gcode/pcb2gcode/pull/653

But it sounds like what you want can't be done at the moment.  I'll have to think about it.