pcb2gcode/pcb2gcode #669: Having trouble with an outline file from easy-sdr project

Terak8510, 2023-05-06:
command output:
pcb2gcode--back="/home/user/Downloads/Easy-SDR-master/HF Upconverter/SMD/Gerbers/Gerber_BottomLayer.GBL" --drill="/home/user/Downloads/Easy-SDR-master/HF Upconverter/SMD/Gerbers/Gerber_Drill_PTH.DRL" --front="/home/user/Downloads/Easy-SDR-master/HF Upconverter/SMD/Gerbers/Gerber_TopLayer.GTL" --outline="/home/user/Downloads/Easy-SDR-master/HF Upconverter/SMD/Gerbers/Gerber_BoardOutline.GKO" --output-dir="/home/user/Downloads/Easy-SDR-master/HF Upconverter/SMD/Gerbers/kicad_import-out/pcb2gcode-out"  --metric=true --metricoutput=true --mirror-axis=0.0000 --mirror-yaxis=false --nog64=true --optimise=0.0100 --tile-x=1 --tile-y=1 --tolerance=0.0060 --zchange=25.0000 --zchange-absolute=false --zero-start=true --zsafe=1.0000 --extra-passes=2 --mill-feed=200 --mill-speed=7000 --offset=0.0200 --voronoi=false --zwork=-0.0250 --drill-feed=50 --drill-side=auto --drill-speed=7000 --milldrill=false --nog81=true --nog91-1=true --onedrill=false --zdrill=-1.5000 --bridges=0.5000 --bridgesnum=6 --cut-feed=12 --cut-infeed=0.6000 --cut-side=auto --cut-speed=7000 --cut-vertfeed=50 --cutter-diameter=1.1500 --fill-outline=true --zbridges=-0.5000 --zcut=-1.2000

The files: 
[thefiles.zip](https://github.com/pcb2gcode/pcb2gcode/files/11411622/thefiles.zip)

Running debian bullseye, pcb2gcode 2.4.0, pcb2gcodeGUI 1.3.2-1

Trying to get this to grind out on the CNC, after i add the outline file, i get no processed files, just empty headers in output dir.
This is from https://github.com/IgrikXD/Easy-SDR (there is no issues tab there)

Even when loading (importing into kicad from gerbview) files i see nothing wrong with the outline file.

Importing front side... DONE.
Importing back side... DONE.
Importing outline... DONE.
Processing input files... Some nearly-connected lines in the gerber input have been adjusted to properly connect
Found an unconnected loop while parsing a gerber file while expecting only loops
DONE.
Exporting back... DONE. (Height: 49.544mm Width: 82.056mm)
Exporting front... DONE. (Height: 49.544mm Width: 82.056mm)
Exporting outline... DONE. (Height: 49.544mm Width: 82.056mm) The board should be cut from the FRONT side. 
Importing drill... DONE.
Exporting milldrill... Exporting drill... DONE. The board should be drilled from the FRONT side.
END.

eyal0, 2023-05-06:
Wow that's really weird.  I'll take a look at it!

eyal0, 2023-05-06:
Let me open the outline in gerbv.  It looks like this:

[image]

Now I will display the properties of one piece of it, in white.

[image]

You see that is has y coordinate equal to 12.445mm.  The x goes from 1.905mm on the right to 0 on the left. 

Now look at the one that it is "connected" to:

[image]

It is connecting at the coordinate of 0, 12.446.  But 12.445 is not equal to 12.446!

If look in the file `Gerber_BoardOutline.GKO` on lines 19 and 20, I see this:

```
G01X0Y12445D01*
G01X0Y12446D02*
```

Those ought to be the same number!  If I change the second one to be "12445" then I get this result for milling:

[image]

That looks pretty good!  So you can make that change.  There is a bug in the gerber files.  Maybe a bug in the source of those files?

What tool made easy-sdr?

eyal0, 2023-05-06:
I have a function in the code to detect these kinds of errors.  It's specifically for easy eda, which is making these errors often.  That's why it says "Some nearly-connected lines in the gerber input have been adjusted to properly connect"  

0458bbaa7a0f788f90ac66e1e8d6f4b4f912c3ea

But it seems not to be working here!  Hmm!

Terak8510, 2023-05-06:
Thanks Eyal, I was not certian what was going on there.
That was so suttle, no wonder i missed it.
Next time i will look through the other files as you did here.

> What tool made easy-sdr?
https://easyeda.com/

eyal0, 2023-05-06:
Ah!  I figured it out.  I'm going to add that gerber that you just sent me to my test cases in this code.  Is that okay with you?

Terak8510, 2023-05-06:
It's fine with me thanks.
It's not my project, it belongs to https://github.com/IgrikXD/Easy-SDR
I could not complain there as there is no "issues" tab on /his?/ page.

I was just going to grind this out using Candle when i noticed i had empty output from pcb2gcodeGUI.

PR #670 (2023-05-06): fix: Use the merged paths for making eulerian paths.
This fixes #669
commit c74468e4 (2023-05-06): fix: Use the merged paths for making eulerian paths.

I did the merging but forgot to use the merged result!

This fixes #669 .
closed 2023-05-06 by eyal0