pcb2gcode/pcb2gcode #672: add laser engraver in tools

hosterweb1, 2023-06-03:
Hi! i use this app to generate gcode to engrave pcb on laser engrave, and i need to edit gcode on text editor. can you add laser support?

eyal0, 2023-06-07:
What edits do you need to make in order to work with laser?  There are some options for it already, I think, but maybe I need more...?

hosterweb1, 2023-06-08:
Maybe I'm doing something wrong?  Could you point out where in the gui to look for settings so that instead of milling there is laser engraving?  Ideally, it would be to add a laser point, and instead of lowering the cutter, the machine would turn on the laser, and vice versa, turn it off when lifting.  Thank you

eyal0, 2023-06-09:
I don't use the GUI, I use the CLI.  The GUI is very old.

I think that there are some options for modifying how pcb2gcode works that are compatible with laser cutters.

https://github.com/pcb2gcode/pcb2gcode/issues/72

What changes would you need?  Can you be specific?  I do not own neither laser cutter nor CNC.  Perhaps you could make a CNC output using pcb2gcode and then show me how it needs to be modified in order to work for laser?

Also, read the wiki and see if any of the options there can help you.